iberianpig/fusuma #133: Reverse directions with no alternatives

l0f4r0, 2019-09-14:
**Description**
I've followed instructions at https://github.com/iberianpig/fusuma/blob/master/README.md
Here is my ~/.config/fusuma/config.yml file:

```
swipe:
	3:
		left:
			command: 'xdotool key alt+Left'
		right:
			command: 'xdotool key alt+Right'
pinch:
	2:
		in:
			command: 'xdotool key ctrl+plus'
			threshold: 0.1
		out:
			command: 'xdotool key ctrl+minus'
			threshold: 0.1
threshold:
	swipe: 1
	pinch: 1

interval:
	swipe: 1
	pinch: 1
```
Directions are reversed. When I swipe left, fusuma thinks I go right, and vice-versa.
I know this is expected as some tutorials on the internet suggest to reverse the commands to balance the behavior. However, **setting alt+Right instead of alt+Left, and vice-versa doesn't produce any change**. So I can't understand...

**To Reproduce**
Steps to reproduce the behavior:
1. Swipe left
2. Swipe right

**Expected behavior**
Swipe left should go back in a browser, swipe right should go forward.

**Versions**
```
fusuma --version
I, [2019-09-14T18:00:50.011804 #30572]  INFO -- : ---------------------------------------------
I, [2019-09-14T18:00:50.011838 #30572]  INFO -- : Fusuma: 0.11.1
I, [2019-09-14T18:00:50.013439 #30572]  INFO -- : libinput: 1.6.3
I, [2019-09-14T18:00:50.013993 #30572]  INFO -- : OS: Linux 4.19.0-0.bpo.5-amd64 #1 SMP Debian 4.19.37-5+deb10u2~bpo9+1 (2019-08-16)
I, [2019-09-14T18:00:50.014521 #30572]  INFO -- : Distribution: Debian GNU/Linux 9 \n \l
I, [2019-09-14T18:00:50.015042 #30572]  INFO -- : Desktop session:
I, [2019-09-14T18:00:50.015104 #30572]  INFO -- : ---------------------------------------------
```
Many thanks in advance :)

l0f4r0, 2019-10-13:
Any suggestion please? ;)

iberianpig, 2019-10-14:
Please paste output of fusuma -v.

If fusuma doesn't  read ~/.config/fusuma/config.yml, it doesn't work correctly you expected.

l0f4r0, 2019-10-20:
Thank you for your answer.

First of all, I've just noticed you released newer versions so I've upgraded from 0.11.1 to 1.2.1 (via gem install). Thx ;)

Then, it's not indicated in your README.md which permissions should be associated to file ~/.config/fusuma/config.yml please... From my side, I've chosen 644.

It seems default config file is located at /var/lib/gems/2.3.0/gems/fusuma-1.2.1/lib/fusuma/config.yml. I've tried to force my config file ~/.config/fusuma/config.yml but the latter is not taken into account, see below:

```
sudo fusuma --config=~/.config/fusuma/config.yml
I, [2019-10-20T18:53:18.765233 #29360]  INFO -- : reload config : /var/lib/gems/2.3.0/gems/fusuma-1.2.1/lib/fusuma/config.yml
I, [2019-10-20T18:53:18.765568 #29360]  INFO -- : reload config : /var/lib/gems/2.3.0/gems/fusuma-1.2.1/lib/fusuma/config.yml
I, [2019-10-20T18:53:18.779142 #29360]  INFO -- : ---------------------------------------------
I, [2019-10-20T18:53:18.779172 #29360]  INFO -- : Fusuma: 1.2.1
I, [2019-10-20T18:53:18.780868 #29360]  INFO -- : libinput: 1.6.3
I, [2019-10-20T18:53:18.781450 #29360]  INFO -- : OS: Linux 4.19.0-0.bpo.5-amd64 #1 SMP Debian 4.19.37-5+deb10u2~bpo9+1 (2019-08-16)
I, [2019-10-20T18:53:18.782025 #29360]  INFO -- : Distribution: Debian GNU/Linux 9 \n \l
I, [2019-10-20T18:53:18.782559 #29360]  INFO -- : Desktop session:
I, [2019-10-20T18:53:18.782582 #29360]  INFO -- : ---------------------------------------------
```

Finally, see below as requested (as you can see, I've swipped left 3 times but action "xdotool key alt+Right" was triggered as per config file /var/lib/gems/2.3.0/gems/fusuma-1.2.1/lib/fusuma/config.yml):
```
sudo fusuma -v
I, [2019-10-20T18:55:43.436505 #29847]  INFO -- : reload config : /var/lib/gems/2.3.0/gems/fusuma-1.2.1/lib/fusuma/config.yml
I, [2019-10-20T18:55:43.436554 #29847]  INFO -- : ---------------------------------------------
I, [2019-10-20T18:55:43.436567 #29847]  INFO -- : Fusuma: 1.2.1
I, [2019-10-20T18:55:43.438189 #29847]  INFO -- : libinput: 1.6.3
I, [2019-10-20T18:55:43.438758 #29847]  INFO -- : OS: Linux 4.19.0-0.bpo.5-amd64 #1 SMP Debian 4.19.37-5+deb10u2~bpo9+1 (2019-08-16)
I, [2019-10-20T18:55:43.439315 #29847]  INFO -- : Distribution: Debian GNU/Linux 9 \n \l
I, [2019-10-20T18:55:43.439897 #29847]  INFO -- : Desktop session:
I, [2019-10-20T18:55:43.439920 #29847]  INFO -- : ---------------------------------------------
D, [2019-10-20T18:55:43.439940 #29847] DEBUG -- : ---------------------------------------------
D, [2019-10-20T18:55:43.439949 #29847] DEBUG -- : Enabled Plugins: 
D, [2019-10-20T18:55:43.440015 #29847] DEBUG -- :   Fusuma::Plugin::Buffers::GestureBuffer
D, [2019-10-20T18:55:43.440027 #29847] DEBUG -- :   Fusuma::Plugin::Detectors::PinchDetector
D, [2019-10-20T18:55:43.440035 #29847] DEBUG -- :   Fusuma::Plugin::Detectors::RotateDetector
D, [2019-10-20T18:55:43.440043 #29847] DEBUG -- :   Fusuma::Plugin::Detectors::SwipeDetector
D, [2019-10-20T18:55:43.440050 #29847] DEBUG -- :   Fusuma::Plugin::Events::Records::GestureRecord
D, [2019-10-20T18:55:43.440060 #29847] DEBUG -- :   Fusuma::Plugin::Events::Records::IndexRecord
D, [2019-10-20T18:55:43.440067 #29847] DEBUG -- :   Fusuma::Plugin::Events::Records::TextRecord
D, [2019-10-20T18:55:43.440074 #29847] DEBUG -- :   Fusuma::Plugin::Executors::CommandExecutor
D, [2019-10-20T18:55:43.440081 #29847] DEBUG -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
D, [2019-10-20T18:55:43.440088 #29847] DEBUG -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
D, [2019-10-20T18:55:43.440095 #29847] DEBUG -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
D, [2019-10-20T18:55:43.440102 #29847] DEBUG -- : ---------------------------------------------
D, [2019-10-20T18:55:43.441631 #29847] DEBUG -- : {:debug_events=>"stdbuf -oL -- libinput-debug-events"}
D, [2019-10-20T18:55:43.825396 #29847] DEBUG -- : {:input_event=>time: 2019-10-20 18:55:43 +0200, tag: libinput_command_input, record: -event3   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k}
D, [2019-10-20T18:55:43.826319 #29847] DEBUG -- : {:filtered=>time: 2019-10-20 18:55:43 +0200, tag: libinput_command_input, record: -event3   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k}
D, [2019-10-20T18:55:43.827058 #29847] DEBUG -- : {:input_event=>time: 2019-10-20 18:55:43 +0200, tag: libinput_command_input, record: -event11  DEVICE_ADDED     Video Bus                         seat0 default group2  cap:k}
D, [2019-10-20T18:55:43.827268 #29847] DEBUG -- : {:filtered=>time: 2019-10-20 18:55:43 +0200, tag: libinput_command_input, record: -event11  DEVICE_ADDED     Video Bus                         seat0 default group2  cap:k}
D, [2019-10-20T18:55:43.827776 #29847] DEBUG -- : {:input_event=>time: 2019-10-20 18:55:43 +0200, tag: libinput_command_input, record: -event1   DEVICE_ADDED     Sleep Button                      seat0 default group3  cap:k}
D, [2019-10-20T18:55:43.827970 #29847] DEBUG -- : {:filtered=>time: 2019-10-20 18:55:43 +0200, tag: libinput_command_input, record: -event1   DEVICE_ADDED     Sleep Button                      seat0 default group3  cap:k}
D, [2019-10-20T18:55:43.828772 #29847] DEBUG -- : {:input_event=>time: 2019-10-20 18:55:43 +0200, tag: libinput_command_input, record: -event9   DEVICE_ADDED     Raydium Corporation Raydium Touch System seat0 default group4  cap:t  size 274x154mm calib}
[... cut ...]
D, [2019-10-20T18:55:54.870918 #29847] DEBUG -- : {:filtered=>time: 2019-10-20 18:55:54 +0200, tag: libinput_command_input, record: -event0   KEYBOARD_KEY     +11.43s	*** (-1) pressed}
D, [2019-10-20T18:55:55.315892 #29847] DEBUG -- : {:input_event=>time: 2019-10-20 18:55:55 +0200, tag: libinput_command_input, record:  event0   KEYBOARD_KEY     +11.87s	*** (-1) pressed}
D, [2019-10-20T18:55:55.316549 #29847] DEBUG -- : {:filtered=>time: 2019-10-20 18:55:55 +0200, tag: libinput_command_input, record:  event0   KEYBOARD_KEY     +11.87s	*** (-1) pressed}
```

iberianpig, 2019-10-20:
Please use `$ fusuma` without `sudo`.

l0f4r0, 2019-10-21:
> Please use `$ fusuma` without `sudo`.

I can't, I don't have enough persmission:
```
ls -l /usr/local/bin/fusuma 
-rwx------ 1 root staff 472 oct.  20 18:21 /usr/local/bin/fusuma*
```

iberianpig, 2019-10-22:
It seems to be not found the configuration file  (~/.config/) because you are running fusuma as root. Please try sudo -H .

l0f4r0, 2019-10-22:
I think sudo -H is the default behavior on my computer so it doesn't change anything, see below:

```
sudo -H fusuma -v
I, [2019-10-22T21:05:00.791211 #3012]  INFO -- : reload config : /var/lib/gems/2.3.0/gems/fusuma-1.2.1/lib/fusuma/config.yml
I, [2019-10-22T21:05:00.791260 #3012]  INFO -- : ---------------------------------------------
I, [2019-10-22T21:05:00.791273 #3012]  INFO -- : Fusuma: 1.2.1
I, [2019-10-22T21:05:00.792887 #3012]  INFO -- : libinput: 1.6.3
I, [2019-10-22T21:05:00.793490 #3012]  INFO -- : OS: Linux 4.19.0-0.bpo.5-amd64 #1 SMP Debian 4.19.37-5+deb10u2~bpo9+1 (2019-08-16)
I, [2019-10-22T21:05:00.794036 #3012]  INFO -- : Distribution: Debian GNU/Linux 9 \n \l
I, [2019-10-22T21:05:00.794543 #3012]  INFO -- : Desktop session:
I, [2019-10-22T21:05:00.794566 #3012]  INFO -- : ---------------------------------------------
D, [2019-10-22T21:05:00.794585 #3012] DEBUG -- : ---------------------------------------------
D, [2019-10-22T21:05:00.794594 #3012] DEBUG -- : Enabled Plugins: 
D, [2019-10-22T21:05:00.794722 #3012] DEBUG -- :   Fusuma::Plugin::Buffers::GestureBuffer
D, [2019-10-22T21:05:00.794735 #3012] DEBUG -- :   Fusuma::Plugin::Detectors::PinchDetector
D, [2019-10-22T21:05:00.794743 #3012] DEBUG -- :   Fusuma::Plugin::Detectors::RotateDetector
D, [2019-10-22T21:05:00.794750 #3012] DEBUG -- :   Fusuma::Plugin::Detectors::SwipeDetector
D, [2019-10-22T21:05:00.794757 #3012] DEBUG -- :   Fusuma::Plugin::Events::Records::GestureRecord
D, [2019-10-22T21:05:00.794764 #3012] DEBUG -- :   Fusuma::Plugin::Events::Records::IndexRecord
D, [2019-10-22T21:05:00.794772 #3012] DEBUG -- :   Fusuma::Plugin::Events::Records::TextRecord
D, [2019-10-22T21:05:00.794779 #3012] DEBUG -- :   Fusuma::Plugin::Executors::CommandExecutor
D, [2019-10-22T21:05:00.794786 #3012] DEBUG -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
D, [2019-10-22T21:05:00.794793 #3012] DEBUG -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
D, [2019-10-22T21:05:00.794799 #3012] DEBUG -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
D, [2019-10-22T21:05:00.794806 #3012] DEBUG -- : ---------------------------------------------
```

iberianpig, 2019-10-23:
> I think sudo -H is the default behavior on my computer so it doesn't change anything, see below:

So is your `$HOME` is `/root` directory?

Please try set config.yml to `/root/.config/fusuma/config.yml`.

l0f4r0, 2019-10-26:
Unfortunately, I can't understand why but $HOME is not changed:
```
sudo -H echo $HOME
/home/l0f4r0
```

iberianpig, 2019-10-27:
Could you try `fusuma --config=/home/l0f4r0/.config/fusuma/config.yml` ?

l0f4r0, 2019-10-27:
> Could you try `fusuma --config=/home/l0f4r0/.config/fusuma/config.yml` ?

It doesn't work:
```
fusuma --config=/home/l0f4r0/.config/fusuma/config.yml
bash: /usr/local/bin/fusuma: Permission denied
```
I can't run fusuma as a simple user despite the fact I'm in _input_ group as specified/required in your documentation:
```
id
uid=1000(l0f4r0) gid=100(users) groupes=100(users),24(cdrom),25(floppy),27(sudo),29(audio),30(dip),44(video),46(plugdev),101(systemd-journal),106(input),108(netdev)
```
Please tell me which permissions should be associated to file ~/.config/fusuma/config.yml.
From my side, I have 644:
`-rw-r--r-- 1 l0f4r0 users 294 oct.  20 18:45 /home/l0f4r0/.config/fusuma/config.yml`

iberianpig, 2019-10-27:
Fusuma need only read permission for reading config file.

I don't understand why fusuma cannot read config.
it seems your environment is a little weird.

How did you install ruby?

l0f4r0, 2019-10-27:
As far as I can remember, I just followed your instructions (README.md)

l0f4r0, 2019-12-29:
Please don't close as my issue has not been resolved... Thank you

iberianpig, 2020-02-27:
@l0f4r0 Do you have this problem?
I think it is problem that you don't have a permission of fusuma to execute.

> ```
> ls -l /usr/local/bin/fusuma 
> -rwx------ 1 root staff 472 oct.  20 18:21 /usr/local/bin/fusuma*
> ```

l0f4r0, 2020-03-09:
> @l0f4r0 Do you have this problem?
> I think it is problem that you don't have a permission of fusuma to execute.
> 
> > ```
> > ls -l /usr/local/bin/fusuma 
> > -rwx------ 1 root staff 472 oct.  20 18:21 /usr/local/bin/fusuma*
> > ```

Since my post, I've made a fresh reinstall + I'm using Debian 10 instead of Debian 9.
The directions are OK now with fusuma.
I'm afraid we won't find out what it's all about...

Thanks for your help though :)

**PS** : What do you want to do with the ticket then? Closing it even if there has been no real solution (maybe some people will have the same issue)? Applying a specific tag maybe like "cantreproduce" ?

iberianpig, 2020-03-09:
I'm glad to know that you still use fusuma.
Ok, I'll make tag of cantreproduce.

closed 2020-04-12 by iberianpig