iberianpig/fusuma #135: Starting KDE-plasmoids using fusuma?

keiichiiownsu12, 2019-09-25:
Apologies that this is not directly fusuma-related.

I am trying to make a specific gesture activate one of the fullscreen application launchers for kde. It appears no specific command activates a plasmoid's interface: one first needs to attach a plasmoid to a panel, assign it a shortcut, then figure out where in one of the plasma-rc files the specific shortcut is stored, which then ultimately gives one a command they can then assign to something like fusuma. Only thing is. . .that command only works as long as the widget is docked to the panel. 

Curious if anyone knows of a command one can use to activate a plasma widget/plasmoid, without the need for it to be in a panel? Something you can pass to fusuma?

crespyl, 2019-09-29:
I recently started shifting over to kwin_wayland, and have encountered a similar issue with directly activating KWin features without being able to use xdotool.

Fortunately, many KWin and Plasma features are exposed via DBUS, and you can use the `qdbusviewer` and `qdbus` tools to explore and activate specific methods.

KGlobalAccel has a list of components, including KWin and Plasmashell, and each component can give you a list of which shortcuts it has via `shortcutNames`.

If you're using one launcher, you can use `command: 'qdbus org.kde.plasmashell /PlasmaShell org.kde.PlasmaShell.activateLauncherMenu'`, which should trigger whatever your default launcher is.  If you're using several different widgets, it should still be possible but might take some more setup.  I was able to add the fullscreen launcher widget to my desktop and assign it a shortcut.  After doing that, I was able to see that it was registered in DBUS under `org.kde.kglobalaccel /component/plasmashell`, as "activate widget 36", and it can be triggered by `qdbus org.kde.kglobalaccel /component/plasmashell org.kde.kglobalaccel.Component.invokeShortcut "activate widget 36"`.

Hope that helps!

keiichiiownsu12, 2019-09-29:
The top snippet was helpful. I already knew you could add a widget to the desktop/panels/latte-dock and activate using the qdbus commands "activate widget ###".

What I was trying to do is activate a widget WITHOUT adding it to the desktop/panels/latte-dock. Mostly because I want to keep these uncluttered, lol

keiichiiownsu12, 2019-09-29:
I can share my full config.yml 
I use qdbus commands to switch desktops and do the expose commands.

Using xdotool i kind of meh honestly. You have to keep those shortcuts assigned. What if you want to re-assign them lol?

crespyl, 2019-09-29:
Ah, yeah I see, glad that was helpful.

I'm not sure if there's a general solution to having "hidden" widgets that are present but invisible until activated (outside of native/standard features like the launcher/color picker/screenshots/etc).  It might be possible to directly manipulate the `~/.config/plasma-org.kde.plasma.desktop-appletsrc` file to set up a containment/panel for your widgets that is forced to always be offscreen, but I'm not very familiar with exactly how that piece works.

closed 2019-12-05 by stale[bot]