iberianpig/fusuma #139: pinch out gesture laggy

4Frisby, 2019-11-01:
When I pinch out in every application the gesture becomes laggy makes the os halt for several seconds. Pinch in works perfectly but pinch out is the problem.
fusuma --version output: 
I, [2019-11-01T13:54:15.859654 #19481]  INFO -- : reload config : /home/tnural/.config/fusuma/config.yml
I, [2019-11-01T13:54:15.859712 #19481]  INFO -- : ---------------------------------------------
I, [2019-11-01T13:54:15.859728 #19481]  INFO -- : Fusuma: 1.2.1
I, [2019-11-01T13:54:15.861962 #19481]  INFO -- : libinput: 1.10.4
I, [2019-11-01T13:54:15.862978 #19481]  INFO -- : OS: Linux 5.0.0-32-generic #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019
I, [2019-11-01T13:54:15.864055 #19481]  INFO -- : Distribution: Ubuntu 18.04.3 LTS \n \l
I, [2019-11-01T13:54:15.865009 #19481]  INFO -- : Desktop session:
I, [2019-11-01T13:54:15.865049 #19481]  INFO -- : ---------------------------------------------

this is my config file.
swipe:
  3:
    down: 
      command: 'xdotool key super+d'
    up: 
      command: 'xdotool key super'
    left: 
      command: 'xdotool key alt+Left'
    right: 
      command: 'xdotool key alt+Right'
  4:
    right:
      command: 'xdotool key ctrl+alt+Down'
    left:
      command: 'xdotool key ctrl+alt+Up'
pinch:
  in:
    command: 'xdotool key ctrl+plus'
  out:
     command: 'xdotool key ctrl+minus'

threshold:
  swipe: 0.4
  pinch: 0.8

interval:
  swipe: 0.8
  pinch: 0.1

iberianpig, 2019-11-15:
Could you try the following config?

```yml
pinch:
  2:
    in:
      command: 'xdotool keydown ctrl click 4 keyup ctrl'
    out:
      command: 'xdotool keydown ctrl click 5 keyup ctrl'
```

4Frisby, 2019-11-19:
Thanks it worked !

closed 2019-11-19 by 4Frisby