iberianpig/fusuma #141: Pinch in and out (3 fingers) gestures are reversed

homanwang, 2019-11-10:
### Description:

- It does the pinch in shortcut when I pinch out (3 fingers). So does the reverse.

- The problem only happens with 3 fingers. 2 fingers pinch in/out gestures are working properly.

- I've fix the problem temporarily by reversing the shortcuts in the configuration file.

### My entire configuration file:
(Though I think It's just a minor bug :)
```
swipe:
  3:
    left:
      command: 'xdotool key super+Left'
    right:
      command: 'xdotool key super+Right'
    up:
      command: 'xdotool key super+Up'
    down:
      command: 'xdotool key super+Down'
  4:
    left:
      command: 'xdotool key alt+Shift+Tab'
      threshold: 0.5
      interval: 0.5
    right:
      command: 'xdotool key alt+Tab'
      threshold: 0.5
      interval: 0.5
    up:
      command: 'xdotool key ctrl+super+s'
    down:
      command: 'xdotool key ctrl+super+w'
pinch:
  2:
    in:
      command: 'xdotool key ctrl+plus'
      threshold: 0.1
    out:
      command: 'xdotool key ctrl+minus'
      threshold: 0.1
  3:
    in:
      command: 'xdotool key super+h'
      threshold: 0.7
    out:
      command: 'xdotool key super+Up'
      threshold: 0.7
  4:
    in:
      command: 'xdotool key super+d'
      threshold: 0.7
      interval: 1.5
    out:
      command: 'xdotool key super+d'
      threshold: 0.7
      interval: 1.5

threshold:
  swipe: 0.9
  pinch: 0.9	

interval:
  swipe: 0.9
  pinch: 0.9
```

iberianpig, 2020-01-06:
Hi, @homanw104 

Could you try following config.yml?

If it works correctly, print `PINCH-IN` when you 2 or 3 fingers pinch-in gestures.
I tried this config.yml, but it has no problem.

```yaml
pinch:
  2:
    in:
      # command: 'xdotool keydown ctrl click 4 keyup ctrl'
      command: 'echo ----PINCH--IN------'
    out:
      # command: 'xdotool keydown ctrl click 5 keyup ctrl'
      command: 'echo ----PINCH--OUT-----'
  3:
    in:
      command: 'echo ----PINCH--IN------'
    out:
      command: 'echo ----PINCH--OUT-----'
```

iberianpig, 2021-05-01:
https://github.com/iberianpig/fusuma/issues/232#issuecomment-830691885

NikhilSaini38, 2021-05-02:
the issue has been there for quite some time, I personally just used to swap the commands according to the behaviour I wanted
 here's the relevant portion of my debugging config.

```
pinch:
  2:
    in:
      command: "echo 2\ finger\ pinch\ in"
    out:
      command: "echo 2\ finger\ pinch\ out"
  3:
    in:
      command: "echo 3\ finger\ pinch\ out"
    out:
      command: "echo 3\ finger\ pinch\ out"
  4:
    in:
      command: "echo 4\ finger\ pinch\ in"
    out:
      command: "echo 4\ finger\ pinch\ out"
```

the output from fusuma

```
I, [2021-05-02T20:37:32.181430 #179119]  INFO -- : reload config: /home/nikhil/.config/fusuma/config.yml
I, [2021-05-02T20:37:32.341959 #179119]  INFO -- : ---------------------------------------------
I, [2021-05-02T20:37:32.342329 #179119]  INFO -- : Fusuma: 2.0.4
I, [2021-05-02T20:37:32.345451 #179119]  INFO -- : libinput: 1.17.1
I, [2021-05-02T20:37:32.345502 #179119]  INFO -- : ruby 3.0.1p64
I, [2021-05-02T20:37:32.346403 #179119]  INFO -- : OS: Linux 5.10.32-1-MANJARO #1 SMP Wed Apr 21 14:54:10 UTC 2021
I, [2021-05-02T20:37:32.347131 #179119]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2021-05-02T20:37:32.351624 #179119]  INFO -- : Desktop session: plasma x11
I, [2021-05-02T20:37:32.351709 #179119]  INFO -- : ---------------------------------------------
I, [2021-05-02T20:37:32.351747 #179119]  INFO -- : Enabled Plugins: 
I, [2021-05-02T20:37:32.351970 #179119]  INFO -- :   Fusuma::Plugin::Buffers::AppmatcherBuffer
I, [2021-05-02T20:37:32.351993 #179119]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-05-02T20:37:32.352007 #179119]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2021-05-02T20:37:32.352020 #179119]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2021-05-02T20:37:32.352033 #179119]  INFO -- :   Fusuma::Plugin::Buffers::TimerBuffer
I, [2021-05-02T20:37:32.352046 #179119]  INFO -- :   Fusuma::Plugin::Detectors::AppmatcherDetector
I, [2021-05-02T20:37:32.352059 #179119]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2021-05-02T20:37:32.352072 #179119]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-05-02T20:37:32.352084 #179119]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-05-02T20:37:32.352097 #179119]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-05-02T20:37:32.352110 #179119]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2021-05-02T20:37:32.352123 #179119]  INFO -- :   Fusuma::Plugin::Events::Records::AppmatcherRecord
I, [2021-05-02T20:37:32.352135 #179119]  INFO -- :   Fusuma::Plugin::Events::Records::ContextRecord
I, [2021-05-02T20:37:32.352148 #179119]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-05-02T20:37:32.352160 #179119]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-05-02T20:37:32.352173 #179119]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2021-05-02T20:37:32.352185 #179119]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-05-02T20:37:32.352198 #179119]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-05-02T20:37:32.352210 #179119]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2021-05-02T20:37:32.352223 #179119]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2021-05-02T20:37:32.352235 #179119]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2021-05-02T20:37:32.352248 #179119]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-05-02T20:37:32.352260 #179119]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-05-02T20:37:32.352272 #179119]  INFO -- :   Fusuma::Plugin::Inputs::AppmatcherInput
I, [2021-05-02T20:37:32.352285 #179119]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-05-02T20:37:32.352298 #179119]  INFO -- :   Fusuma::Plugin::Inputs::TimerInput
I, [2021-05-02T20:37:32.352310 #179119]  INFO -- :   Fusuma::Plugin::Parsers::AppmatcherParser
I, [2021-05-02T20:37:32.352323 #179119]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2021-05-02T20:37:32.352335 #179119]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-05-02T20:37:32.352347 #179119]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2021-05-02T20:37:32.352358 #179119]  INFO -- : ---------------------------------------------
I, [2021-05-02T20:37:40.592402 #179119]  INFO -- : {:command=>"echo 2 finger pinch in", :args=>{:move_x=>0.0, :move_y=>0.0, :unaccelerated_x=>0.0, :unaccelerated_y=>0.0, :zoom=>1.6, :rotate=>0.11}}
2 finger pinch in
I, [2021-05-02T20:37:52.114804 #179119]  INFO -- : {:command=>"echo 2 finger pinch out", :args=>{:move_x=>0.44, :move_y=>3.08, :unaccelerated_x=>1.64, :unaccelerated_y=>11.48, :zoom=>0.54, :rotate=>-0.83}}
2 finger pinch out
I, [2021-05-02T20:38:01.603572 #179119]  INFO -- : {:command=>"echo 3 finger pinch in", :args=>{:move_x=>0.0, :move_y=>-0.88, :unaccelerated_x=>0.0, :unaccelerated_y=>-3.28, :zoom=>1.71, :rotate=>0.82}}
3 finger pinch in
I, [2021-05-02T20:38:07.559418 #179119]  INFO -- : {:command=>"echo 3 finger pinch out", :args=>{:move_x=>2.2, :move_y=>0.88, :unaccelerated_x=>8.2, :unaccelerated_y=>3.28, :zoom=>0.52, :rotate=>-0.6}}
3 finger pinch out
I, [2021-05-02T20:38:14.641485 #179119]  INFO -- : {:command=>"echo 4 finger pinch in", :args=>{:move_x=>-1.32, :move_y=>2.64, :unaccelerated_x=>-4.92, :unaccelerated_y=>9.84, :zoom=>1.62, :rotate=>0.21}}
4 finger pinch in
I, [2021-05-02T20:38:17.215778 #179119]  INFO -- : {:command=>"echo 4 finger pinch out", :args=>{:move_x=>1.76, :move_y=>3.52, :unaccelerated_x=>6.56, :unaccelerated_y=>13.12, :zoom=>0.5, :rotate=>-0.67}}
4 finger pinch out
I, [2021-05-02T20:38:17.540320 #179119]  INFO -- : {:command=>"echo 4 finger pinch out", :args=>{:move_x=>0.43, :move_y=>0.43, :unaccelerated_x=>1.64, :unaccelerated_y=>1.64, :zoom=>0.33, :rotate=>-0.1}}
4 finger pinch out
```

closed 2022-04-28 by stale[bot]