iberianpig/fusuma #143: Two-finger zoom direction not reversing

dsr373, 2019-12-04:
I love this program! Thanks a ton, I've been using it for years and it makes my workflow a charm! Here's the bug:

**Describe the bug**
When using two-finger pinch gestures, if you zoom in, then immediately (without lifting fingers) zoom out, the direction reversal is not detected. That is, fusuma keeps zooming out. This is also the case for the opposite directions (out, then in). This was not an issue before the 1.3.0 update.

**To Reproduce**
Steps to reproduce the behavior:
1. Start fusuma 1.3.0
2. Go to a window that supports zoom and pinch out with two fingers (to zoom in). The window zooms in.
3. **Without lifting fingers off touchpad**, pinch in (to zoom out). Window keeps zooming in.

**Expected behavior**
On step 3, the window should zoom out.

**Versions**
```
I, [2019-12-04T12:12:18.859438 #17088]  INFO -- : reload config : /home/xxx/.config/fusuma/config.yml
I, [2019-12-04T12:12:18.859469 #17088]  INFO -- : ---------------------------------------------
I, [2019-12-04T12:12:18.859479 #17088]  INFO -- : Fusuma: 1.3.0
I, [2019-12-04T12:12:18.860097 #17088]  INFO -- : libinput: 1.14.3
I, [2019-12-04T12:12:18.860575 #17088]  INFO -- : OS: Linux 5.3.12-1-MANJARO #1 SMP PREEMPT Thu Nov 21 10:55:53 UTC 2019
I, [2019-12-04T12:12:18.861049 #17088]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2019-12-04T12:12:18.863203 #17088]  INFO -- : Desktop session: /usr/share/xsessions/plasma
I, [2019-12-04T12:12:18.863220 #17088]  INFO -- : ---------------------------------------------
I, [2019-12-04T12:12:18.863228 #17088]  INFO -- : ---------------------------------------------
I, [2019-12-04T12:12:18.863233 #17088]  INFO -- : Enabled Plugins: 
I, [2019-12-04T12:12:18.863261 #17088]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2019-12-04T12:12:18.863268 #17088]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2019-12-04T12:12:18.863274 #17088]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2019-12-04T12:12:18.863278 #17088]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2019-12-04T12:12:18.863284 #17088]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2019-12-04T12:12:18.863289 #17088]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2019-12-04T12:12:18.863293 #17088]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2019-12-04T12:12:18.863298 #17088]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2019-12-04T12:12:18.863303 #17088]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2019-12-04T12:12:18.863308 #17088]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2019-12-04T12:12:18.863313 #17088]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2019-12-04T12:12:18.863318 #17088]  INFO -- : ---------------------------------------------
```

**Additional context**
My config file is attached. The `desktop --next` commands are just calling my own scripts.

[kde.txt](https://github.com/iberianpig/fusuma/files/3921725/kde.txt)

iberianpig, 2019-12-04:
Thank you for reporting!
I confirmed that this problem was reproduced.

mehanoid, 2021-05-24:
It looks like it was broken again in some of these commits: cad89357 97ce3d31 or 4d5e05a3

Drarig29, 2021-05-24:
I confirm I have the same problem.

iberianpig, 2021-05-24:
@mehanoid @Drarig29 Thanks for letting me know.

Please try to set `update:` property for pinch gestures. It will become "continuous gestures."

```yaml
pinch:
  2:
    in:
      update:
        command: 'xdotool key ctrl+equal' # Zoom out
        interval: 3.0
    out:
      update:
        command: 'xdotool key ctrl+minus' # Zoom in
        interval: 3.0
```

In Fusuma v2, there are now two types of gestures: "one-shot gestures" and real-time "continuous gestures." Previous pinch gestures are now treated as "one-shot gestures."

However, Fusuma v2 has reproduced the bug of one-shot gestures. I do not intend this behavior, so that I will fix the one-shot gesture.

 I will fix the behavior of the one-shot gestures.

Drarig29, 2021-05-24:
Nice! And it doesn't work well with an `interval` property set to `1.0`. But `2.0` is fine (I found `3.0` to be too sparse.

commit f564ea15 (2020-01-07): Merge pull request #150 from iberianpig/feature/143/direction_not_reversing

Fix the issue that it detects incorrectly with pinch-in/out in continued events

close #143
closed 2022-04-28 by stale[bot]