iberianpig/fusuma #144: Each time I do a gesture a key is being typed

WardenScotland, 2019-12-16:
I was writing some code in visual studios jumping from application to application, when I noticed random letters being typed (mainly s and a keys) 
After looking a little closer I figured out that these keys are entered every time I make a gesture. Is this a way to stop this happening?

iberianpig, 2019-12-16:
If it caused by xdotool, please try https://github.com/iberianpig/fusuma-plugin-wmctrl .

utkarsh92, 2020-01-30:
faced a similar issue. i was trying to add a gesture for going to desktop with 'super+d' but when i make the gesture it just opens the search bar and types the letter 'd'

iberianpig, 2020-01-31:
Please execute same keys combination with xdotool from terminal.
Could you get same issue?

utkarsh92, 2020-01-31:
> Please execute same keys combination with xdotool from terminal.
> Could you get same issue?

something's weird happening here. just tried 'xdotool key super+d' in terminal and it led to me desktop, so its working fine. then i went back to fusuma's config file and added 'super+d' once again just to give it a shot. now, it works sometimes and other times it just opens search and type d. i noticed if i'm very careful with my 3 finger swipe down (which is the gesture i was editing) it works most of the time, but if i'm just swiping down casually like doing something and then swipe down, it leads to typing. my guess, there's probably some bug in the code to detect swipe? but i'll leave that to you...

iberianpig, 2020-02-03:
It may be caused by sending keys at the same time.
Could you try the following command?

```sh
$ xdotool keydown Super key d  keyup Super
```

closed 2020-05-22 by stale[bot]