iberianpig/fusuma #146: When xdotool acts crazy. Workaround should be clearly informed.

epertinez, 2019-12-21:
I had a had time with xdotool. It seems it refuses to work properly or do it more that once in a while.
I solved it jumping to other libraries that do similar. In this case I used xte and wmctrl. It would be nice if this alternative solution enters in config.yml (commented out). I did not mess around with config.yml and I don't know if you can comment things out. If you can, a commented header explaining what config.yml is would do the trick so we know we can comment out other things.

Anyway, I write down here my config.yml and a code I found on the internet that helped me sot it our.
Hope it helps someone.

swipe:
  3:
    left:
      command: '~/bin/desktop_switcher.sh 1'
    right:
      command: '~/bin/desktop_switcher.sh -1'
    up:
      command: 'xdotool key ctrl+t'
    down:
      command: 'xdotool key ctrl+w'
  4:
    left:
      command: '~/bin/desktop_switcher.sh 1'
    right:
      command: '~/bin/desktop_switcher.sh -1'
    up:
      command: "xte 'keydown Control_L' 'key F9' 'keyup Control_L'"
    down:
      command: "xte 'keydown Alt_L' 'key F1' 'keyup Alt_L'"

pinch:
  2:
    in:
      command: "xte 'keydown Control_L' 'key plus' 'keyup Control_L'"
    out:
      command: "xte 'keydown Control_L' 'key minus' 'keyup Control_L'"
  4:
    in:
      command: "xte 'keydown Control_L' 'key F8' 'keyup Control_L'"
    out:
      command: 'xdotool key super+s'

threshold:
  swipe: 0.3
  pinch: 0.1

interval:
  swipe: 1
  pinch: 1

desktop_switcher.sh copied from
https://github.com/iberianpig/fusuma/issues/113

#!/usr/bin/env bash

if [ $# -gt 1 ]; then
  echo "expect 1 argument" 1>&2
  exit 1
fi

switch_number=1
if [ $# -eq 1 ]; then
  switch_number=$1
fi

current_workspace=$(wmctrl -d | grep '\*' | cut -d' ' -f1)

cmd="wmctrl -s"
to=$((current_workspace + switch_number))
if [ $to -lt 0 ]; then
  exit 0
fi
$cmd $to

epertinez, 2019-12-21:
About commenting out things. I have those rules I am not using... swipe 3 down for example. I would like to coment them out but... a) don't know if it breaks something or what is the expected way to comment them (',//,/**/,#). b) don't know if letting them undefined in my user configuration makes the general configuration of the system run. c) Don't want to delete them because I might need them in the future.

iberianpig, 2020-02-19:
OK, I will update README.md or Wiki

* Workaround of xdotool
    * xte
        * http://manpages.ubuntu.com/manpages/bionic/man1/xte.1.html
    * https://github.com/iberianpig/fusuma-plugin-wmctrl
        * This plugin is instead of desktop_switcher.sh
* Example of config.yml
    * https://github.com/iberianpig/fusuma/wiki
        * Everyone can edit :)
* How to comment on YAML
    * https://www.tutorialspoint.com/yaml/yaml_comments.htm

iberianpig, 2020-03-02:
I create new plugin for fusuma
 https://github.com/iberianpig/fusuma-plugin-sendkey

This plugin is an alternative to xdotool.

closed 2020-03-25 by iberianpig