iberianpig/fusuma #148: Some gestures do not work anymore after resume

marcociara379, 2019-12-29:
Hello,
I am experimenting with Fusuma, and it seems to work fine on my machine, a Dell XPS 15-9530 with Ubuntu 18.04 with kernel `4.15.0-72-generic`. Fusuma version: `1.3.0`. Libinput version `1.10.4`
It works flawlessly until I suspend the system and resume it. At this point, the three finger swipe do not work anymore (sometimes the up/down swipes are recognized as scrolls).

Any hint?

my `~/.config/fusuma/config.yml` looks like the following:

```yaml
swipe:
  3:
    left:
      command: 'xdotool key alt+Shift+Tab'
    right:
      command: 'xdotool key alt+Tab'
    up:
      command: 'xdotool key super+s'
    down:
      command: 'xdotool key super+s'
  4:
    up:
      command: 'xdotool key ctrl+alt+Down'
    down:
      command: 'xdotool key ctrl+alt+Up'
pinch:
  in:
    command: 'xdotool key ctrl+plus'
    threshold: 0.1
  out:
     command: 'xdotool key ctrl+minus'
     threshold: 0.1

threshold:
  swipe: 1
  pinch: 1

interval:
  swipe: 1
  pinch: 1
```

iberianpig, 2019-12-31:
@marcociara379 
First, launch `fusuma -v` in the terminal in advance.
Next, suspend and wake up your machine.
Next, is there an error in the terminal output when swiping?

marcociara379, 2019-12-31:
@iberianpig 
I have changed and disabled pinch in my config. Swipe was already working properly. After resume it seems to work from the output, although I see many "POINTER AXIS" events when I pinch. Sometimes it is pinch and sometimes is this POINTER AXIS the event

iberianpig, 2019-12-31:
@marcociara379 
This issue comes from libinput does not detect correctly, and it is hard to fix in Fusuma.

marcociara379, 2019-12-31:
ok, I will close the issue here

closed 2019-12-31 by marcociara379