iberianpig/fusuma #149: Can not use fusuma with a error: NoMethodError

HelloGithubC, 2019-12-30:
Recently when I use my touchpad in Ubuntu 19.04 I find it can not work when I use my 3-finger gesture. When I run the fusuma manually I get this error:

/var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:90:in `block in generate_devices': undefined method `assign_attributes' for nil:NilClass (NoMethodError)

The whole error is below:

Traceback (most recent call last):
	26: from /usr/local/bin/fusuma:23:in `<main>'
	25: from /usr/local/bin/fusuma:23:in `load'
	24: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/exe/fusuma:41:in `<top (required)>'
	23: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma.rb:18:in `run'
	22: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma.rb:89:in `run'
	21: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:21:in `run'
	20: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:56:in `debug_events'
	19: from /usr/lib/ruby/2.5.0/open3.rb:95:in `popen3'
	18: from /usr/lib/ruby/2.5.0/open3.rb:205:in `popen_run'
	17: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:57:in `block in debug_events'
	16: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:57:in `each'
	15: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:58:in `block (2 levels) in debug_events'
	14: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:22:in `block in run'
	13: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma.rb:90:in `block in run'
	12: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma.rb:100:in `filter'
	11: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma.rb:100:in `any?'
	10: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma.rb:100:in `block in filter'
	 9: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/filters/filter.rb:17:in `filter'
	 8: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/filters/libinput_device_filter.rb:29:in `keep?'
	 7: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/plugin/filters/libinput_device_filter.rb:54:in `all'
	 6: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:39:in `available'
	 5: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:33:in `all'
	 4: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:62:in `fetch_devices'
	 3: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:81:in `generate_devices'
	 2: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:81:in `each_with_object'
	 1: from /var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:81:in `each'
/var/lib/gems/2.5.0/gems/fusuma-1.3.0/lib/fusuma/device.rb:90:in `block in generate_devices': undefined method `assign_attributes' for nil:NilClass (NoMethodError)

The version of fusuma: 1.3.0
The version of libinput: 1.12.6

iberianpig, 2019-12-31:
@HelloGithubC 
Please paste the output of `libinput list-devices`.
And please wraps it with ` ``` ` (for pretty-printing code lines)

thejinx0r, 2020-02-09:
@iberianpig 
I'm not OP, but I have the same issue. I'm on debian 10.
I have the same stack trace minus the path/version.

Here's the version information:
```
I, [2020-02-09T09:47:46.189740 #4807]  INFO -- : Fusuma: 1.3.2
I, [2020-02-09T09:47:46.192277 #4807]  INFO -- : libinput: 1.12.6
```

Here's my output of libinput list-devices:
```
event7  - HTX USB HID Device HTX HID Device Touchpad: kernel bug: clickpad advertising right button
Device:           Power Button
Kernel:           /dev/input/event1
Group:            1
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           HTX USB HID Device HTX HID Device Keyboard
Kernel:           /dev/input/event4
Group:            2
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           HTX USB HID Device HTX HID Device Consumer Control
Kernel:           /dev/input/event5
Group:            2
Seat:             seat0, default
Capabilities:     keyboard pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    disabled
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           HTX USB HID Device HTX HID Device Touchpad
Kernel:           /dev/input/event7
Group:            2
[... 280 lines cut ...]
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           ImExPS/2 Generic Explorer Mouse
Kernel:           /dev/input/event3
Group:            8
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a

```

HelloGithubC, 2020-02-10:
@iberianpig 
Thank you for your reply. I'm very sorry that I haven't replied to you for a long time. But now I have found a solution so you can pay you attention to @thejinx0r but me. Thanks to you again.

thejinx0r, 2020-02-10:
@HelloGithubC 
What was your solution?

HelloGithubC, 2020-02-10:
@thejinx0r 
I found libinput-gesture and it works.
https://github.com/bulletmark/libinput-gestures

iberianpig, 2020-02-11:
@thejinx0r Thanks.
I fix the bug you found. 

Please update Fusuma with the following command.
`gem update fusuma`

thejinx0r, 2020-02-11:
Nice! It works!

HelloGithubC, 2020-02-11:
@thejinx0r OK, I'm going to close it.

closed 2020-02-11 by HelloGithubC