iberianpig/fusuma #154: Fusuma not working

goutham-nekkalapu, 2020-03-01:
**Describe the bug**
Thank you for building cool feature. Installed the latest version of Fusuma (on ubuntu 18.04 LTS, X11) through the instructions provided but it does not work.Through the issue 106 here : "https://github.com/iberianpig/fusuma/issues/106" found that 'GESTURE_SWIPE_BEGIN' is not there in my  'libinput-debug-events'. 

In the past, I was able to use your 'xswipe' for multi gesture successfully on Ubuntu 16.04 LTS on my laptop. Are there any settings or specific ways to get 'GESTURE_SWIPE_BEGIN'  on Ubuntu 18.04 LTS ?. 

Note: Tested 'xdotool" individually and it works fine for all the 3 figure gestures.

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Versions**
You can get this information from copy and paste the output of `fusuma --version` from the command line. Also, please include the OS and what version of the OS you're running.

**Additional context**
Any additional context, your `~/.config/fusuma/config.yml` or data that might be necessary to reproduce the issue.

iberianpig, 2020-03-01:
Have you restarted after adding $USER to input group?

goutham-nekkalapu, 2020-03-04:
@iberianpig yes, I did

iberianpig, 2020-03-06:
If you don't get GESTURE_SWIPE_xxxx line from outputs of  libinput debug-events, unfortunately Fusuma cannot recognise gestures.

goutham-nekkalapu, 2020-03-06:
Thank you for the reply. Do u have any insight on  what  the issue might be for  'GESTURE_SWIPE_xxxx' not being recognized ?. 
Also any pointers as to where I should look to fix this is much appreciated

iberianpig, 2020-03-07:
It may be the following problem

* Wrong priority setting of driver at  /etc/X11/xorg.conf.d/ or /usr/share/X11/xorg.conf.d/
   * ref: https://wiki.archlinux.org/index.php/Libinput#Via_Xorg_configuration_file

* Libinput doesn't support the device
    * Check if it works with the latest libinput. You need to build and install libinput from source. https://wayland.freedesktop.org/libinput/doc/latest/building.html

goutham-nekkalapu, 2020-03-26:
@iberianpig Thank you for sharing the links, will use them and try to fix

closed 2020-06-20 by stale[bot]