iberianpig/fusuma #155: 2 finger swipe

Sandvich, 2020-03-22:
**Is your feature request related to a problem? Please describe.**
Hi! I've been loving Fusuma so far, but the one thing I'm really missing is two-finger swipes to go back and forwards a page, like on MacOS. Is this something that's possible?

I've already put it in my config, but sadly it doesn't seem to recognise the gesture. I have it mapped to alt+Left/alt+Right, which is already a thing in KDE

**Additional context**
Just in case this is already enabled and I've just made a mistake, here's my config:
```yaml
swipe:
  2:
    left:
      command: 'xdotool key alt+Left'
    right:
      command: 'xdotool key alt+Right'
  3:
    left:
      workspace: 'next'
    right:
      workspace: 'prev'
    up:
      command: 'xdotool key Control_L+F10'
pinch:
  2:
    in:
      command: 'xdotool keydown ctrl click 4 keyup ctrl' # Zoom in
    out:
      command: 'xdotool keydown ctrl click 5 keyup ctrl' # Zoom out

threshold:
  swipe: 1
  pinch: 1

interval:
  swipe: 1
  pinch: 1
```

l0f4r0, 2020-03-29:
What is the ouput of fusuma while you're swiping?
Is each swipe recognized but action is not triggered? Or don't you see anything in output?

Sandvich, 2020-03-29:
No output with 2 finger swipe.
Pinching in and out both work, although neither is particularly responsive.
Swiping up and down are already handled by KDE - maybe whatever handles the two-finger swipe to scroll is consuming the left and right swipes before they get to fusuma?

l0f4r0, 2020-03-29:
To make sure you could temporarily start by assigning other shortcuts (not already managed by KDE or any other software) for your 2 finger swipes?

Sandvich, 2020-03-29:
What do you mean by other shortcuts? Like map two finger left to ctrl+c and two finger right to ctrl+v or something similar?
I tried turning off horizontal scrolling in KDE, but that didn't help.

iberianpig, 2020-03-30:
The libinput debug-events have 4-finger and 3-finger swipe events, such as the one below.

```
 -event17  GESTURE_SWIPE_UPDATE +333.28s 3 -11.36/ 4.06 (-30.62/10.94 unaccelerated)
 event17  GESTURE_SWIPE_UPDATE +333.28s 3 -10.55/ 3.65 (-28.43/ 9.84 unaccelerated)
 event17  GESTURE_SWIPE_UPDATE +333.29s 3 -8.11/ 3.25 (-21.87/ 8.75 unaccelerated)
 event17  GESTURE_SWIPE_UPDATE +333.30s 3 -6.90/ 2.43 (-18.59/ 6.56 unaccelerated)
 event17  GESTURE_SWIPE_UPDATE +333.31s 3 -4.87/ 1.62 (-13.12/ 4.37 unaccelerated)
 event17  GESTURE_SWIPE_UPDATE +333.31s 3 -3.65/ 1.22 (-9.84/ 3.28 unaccelerated)
 event17  GESTURE_SWIPE_END +333.35s    3
 event17  GESTURE_SWIPE_BEGIN +333.65s  4
 event17  GESTURE_SWIPE_UPDATE +333.65s 4 11.87/-2.43 (31.99/-6.56 unaccelerated)
 event17  GESTURE_SWIPE_UPDATE +333.66s 4 12.78/-3.65 (34.45/-9.84 unaccelerated)
 event17  GESTURE_SWIPE_UPDATE +333.67s 4 11.87/-3.65 (31.99/-9.84 unaccelerated)
 event17  GESTURE_SWIPE_UPDATE +333.68s 4 12.17/-4.26 (32.81/-11.48 unaccelerated)
```
However, it does not send a two-finger swipe event because it is recognized as a scroll. That's why Fusuma doesn't implement a two-finger swipe.

Sandvich, 2020-03-31:
Thanks @iberianpig. Does this mean that it's impossible to add with the current setup? Or is there a way to map horizontal scroll perhaps?

iberianpig, 2020-03-31:
It's mean that Fusuma cannot detect 2 fingers swipe.

If someone creates a plugin to parse the scroll, it will be recognizable as swipe.

closed 2020-04-12 by iberianpig