iberianpig/fusuma #156: Touchpad is filtered in listed devices

remynicol, 2020-03-28:
Hello,

I try to use Fusuma with a touchpad and I think it needs a very tiny change to work with. Unfortunately I am a developper but I never worked with Ruby.

Why I think it could be just a tiny update :

- fusuma -v
It well receives the inputs of the touchpad :
```r
D, [2020-03-28T14:29:36.600762 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_MOTION     +75.55s     1 (1) 39.68/54.88 (191.24/146.36mm)}
D, [2020-03-28T14:29:36.601544 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_FRAME      +75.55s     }
D, [2020-03-28T14:29:36.610758 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_MOTION     +75.56s     0 (0) 54.46/55.16 (262.47/147.11mm)}
D, [2020-03-28T14:29:36.611637 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_MOTION     +75.56s     1 (1) 39.65/55.43 (191.09/147.83mm)}
D, [2020-03-28T14:29:36.612464 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_FRAME      +75.56s     }
D, [2020-03-28T14:29:36.620750 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_MOTION     +75.57s     0 (0) 54.43/55.81 (262.32/148.83mm)}
D, [2020-03-28T14:29:36.621630 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_UP         +75.57s     }
D, [2020-03-28T14:29:36.622489 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_FRAME      +75.57s     }
D, [2020-03-28T14:29:36.634683 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_UP         +75.59s     }
D, [2020-03-28T14:29:36.635377 #1853] DEBUG -- : {:input_event=>time: 2020-03-28 14:29:36 +0100, tag: libinput_command_input, record:  event1   TOUCH_FRAME      +75.59s     }
```
- libinput list-devices
It well sees the two devices (this with touch capabilitie is the touchscreen)
```r
Device:           ILITEK ILITEK-TP
Kernel:           /dev/input/event1
Group:            1
Seat:             seat0, default
Size:             482x267mm
Capabilities:     touch
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           ILITEK ILITEK-TP Mouse
Kernel:           /dev/input/event0
Group:            1
Seat:             seat0, default
Size:             482x267mm
Capabilities:     pointer
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a
```

- fusuma -l
But the touchscreen is filtered :
```r
I, [2020-03-28T14:33:00.509900 #1874]  INFO -- : ---------------------------------------------
I, [2020-03-28T14:33:00.510044 #1874]  INFO -- : Fusuma: 1.7.0
I, [2020-03-28T14:33:00.519909 #1874]  INFO -- : libinput: 1.12.6
I, [2020-03-28T14:33:00.522244 #1874]  INFO -- : OS: Linux 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020
I, [2020-03-28T14:33:00.524485 #1874]  INFO -- : Distribution: Raspbian GNU/Linux 10 \n \l
I, [2020-03-28T14:33:00.526666 #1874]  INFO -- : Desktop session:
I, [2020-03-28T14:33:00.526771 #1874]  INFO -- : ---------------------------------------------
I, [2020-03-28T14:33:00.526825 #1874]  INFO -- : Enabled Plugins:
I, [2020-03-28T14:33:00.527016 #1874]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-03-28T14:33:00.527075 #1874]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-03-28T14:33:00.527124 #1874]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-03-28T14:33:00.527171 #1874]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-03-28T14:33:00.527217 #1874]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-03-28T14:33:00.527264 #1874]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-03-28T14:33:00.527310 #1874]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-03-28T14:33:00.527355 #1874]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-03-28T14:33:00.527400 #1874]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-03-28T14:33:00.527445 #1874]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-03-28T14:33:00.527491 #1874]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-03-28T14:33:00.527539 #1874]  INFO -- : ---------------------------------------------
ILITEK ILITEK-TP Mouse
```

Is my feeling correct ?

Best regards,
Remy

iberianpig, 2020-03-28:
Try the following settings.

```config.yml
plugins:
  filters:
    libinput_device_filter:
      keep_device_names:
        - ILITEK ILITEK-TP
        - ILITEK ILITEK-TP Mouse
```
Add this to the bottom of the `~/.config/fusuma/config.yml` .

iberianpig, 2020-03-28:
Fusuma detects the device as a gesture-enabled touchpad if the libinput-list-devices has a natural-scrolling entry.

You can customize `config.yml` to add or refine any device to be detected by Fusuma.
The `plugins:` section in `config.yml` is used to customize the behavior of fusuma's (internal and external) plugins, but these features are not yet documented.

remynicol, 2020-03-28:
No improvement with this config.yml :
```r
swipe:
  3:
    left:
      command: 'echo left'
    right:
      command: 'echo right'

plugins:
  filters:
    libinput_device_filter:
      keep_device_names:
        - ILITEK ILITEK-TP
        - ILITEK ILITEK-TP Mouse
```

Output of "fusuma -l -c config.yml -v"
```r
I, [2020-03-29T00:41:27.751044 #3623]  INFO -- : reload config : /home/xuser/.config/fusuma/config.yml
I, [2020-03-29T00:41:27.752171 #3623]  INFO -- : reload config : /home/xuser/.config/fusuma/config.yml
I, [2020-03-29T00:41:27.834655 #3623]  INFO -- : ---------------------------------------------
I, [2020-03-29T00:41:27.834817 #3623]  INFO -- : Fusuma: 1.7.0
I, [2020-03-29T00:41:27.844652 #3623]  INFO -- : libinput: 1.12.6
I, [2020-03-29T00:41:27.846927 #3623]  INFO -- : OS: Linux 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020
I, [2020-03-29T00:41:27.849229 #3623]  INFO -- : Distribution: Raspbian GNU/Linux 10 \n \l
I, [2020-03-29T00:41:27.851558 #3623]  INFO -- : Desktop session:
I, [2020-03-29T00:41:27.851699 #3623]  INFO -- : ---------------------------------------------
I, [2020-03-29T00:41:27.851912 #3623]  INFO -- : Enabled Plugins:
I, [2020-03-29T00:41:27.852134 #3623]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-03-29T00:41:27.852216 #3623]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-03-29T00:41:27.852284 #3623]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-03-29T00:41:27.852359 #3623]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-03-29T00:41:27.852438 #3623]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-03-29T00:41:27.852506 #3623]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-03-29T00:41:27.852576 #3623]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-03-29T00:41:27.852643 #3623]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-03-29T00:41:27.852710 #3623]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-03-29T00:41:27.852775 #3623]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-03-29T00:41:27.852841 #3623]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-03-29T00:41:27.852908 #3623]  INFO -- : ---------------------------------------------
D, [2020-03-29T00:41:27.862875 #3623] DEBUG -- : {:list_devices=>"libinput list-devices"}
D, [2020-03-29T00:41:27.993470 #3623] DEBUG -- : {:available_devices=>[#<Fusuma::Device:0x00273420 @id="event0", @name="ILITEK ILITEK-TP Mouse", @available=true>]}
ILITEK ILITEK-TP Mouse
```

There is only one device in the "available_devices".

iberianpig, 2020-03-29:
When fusuma runs (without -l option), it only uses the configured device.

But for now, the output of `fusuma -l` shows only devices with natural scrolling, not the configuration of `config.yml`.
(I'll improve on this point later.)

remynicol, 2020-03-29:
I am sorry but I do not manage to recognize any gesture.
I tried swipe and pinch and multiple command, but nothing happens.

What stange is that in debug mode (fusuma -v), I can see the inputs on the touch screen.

iberianpig, 2020-03-29:
If swipe or pinch logs are not shown in outputs of `libinput debug-events`, it is difficult to detect gestures with fusuma.

remynicol, 2020-03-29:
Indeed, sorry for my misunderstanding. libinput gets events but not transform them in gesture.

Thanks for your help !

closed 2020-03-29 by remynicol