iberianpig/fusuma #159: Zoom-in not really usable on Vivaldi

l0f4r0, 2020-03-29:
Hi,
Maybe this is not fusuma to be blamed but Vivaldi browser itself.
I'm just posting here to get some advice, hints and so on so I can revert to Vivaldi team afterwards.
Of course, if it's a fusuma bug, feel free to take that report into account...
Thank you :)

**Describe the bug**
Zoom-in on a webpage opened with Vivaldi browser doesn't work great.
Most of the time (~90% ) nothing happens.
fusuma recognizes the zoom-in gesture and triggers xdotool command but it's not enough to make it working 100% of the time.
The same zoom-in command shortcut (ctrl+plus) works everytime in Vivaldi (that's why I'm not opening the bug on Vivaldi for the moment).
Zoom-out works normally as other gestures (wipe left/right).

**To Reproduce**
1. Open any webpage on Vivaldi
2. Zoom-in while fusuma is running (pinching IN, swiping UP...)

**Expected behavior**
Everything on the webpage should be bigger but here it works with great difficulties (10% of the time maybe...)

**Versions**
```
~/.gem/ruby/2.5.0/bin/fusuma --version
I, [2020-03-29T18:46:05.926660 #26153]  INFO -- : ---------------------------------------------
I, [2020-03-29T18:46:05.926693 #26153]  INFO -- : Fusuma: 1.8.0
I, [2020-03-29T18:46:05.928576 #26153]  INFO -- : libinput: 1.12.6
I, [2020-03-29T18:46:05.928977 #26153]  INFO -- : OS: Linux 5.4.0-0.bpo.2-amd64 #1 SMP Debian 5.4.8-1~bpo10+1 (2020-01-07)
I, [2020-03-29T18:46:05.929362 #26153]  INFO -- : Distribution: Debian GNU/Linux 10 \n \l
I, [2020-03-29T18:46:05.929774 #26153]  INFO -- : Desktop session: lightdm-xsession x11
I, [2020-03-29T18:46:05.929791 #26153]  INFO -- : ---------------------------------------------
I, [2020-03-29T18:46:05.929799 #26153]  INFO -- : Enabled Plugins: 
I, [2020-03-29T18:46:05.929842 #26153]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-03-29T18:46:05.929851 #26153]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-03-29T18:46:05.929857 #26153]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-03-29T18:46:05.929862 #26153]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-03-29T18:46:05.929867 #26153]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-03-29T18:46:05.929872 #26153]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-03-29T18:46:05.929877 #26153]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-03-29T18:46:05.929882 #26153]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-03-29T18:46:05.929901 #26153]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-03-29T18:46:05.929907 #26153]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-03-29T18:46:05.929912 #26153]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-03-29T18:46:05.929917 #26153]  INFO -- : ---------------------------------------------
```

**Additional context**
```
cat ~/.config/fusuma/config.yml 
swipe:
  3:
    left:
      command: 'xdotool key alt+Left'
    right:
      command: 'xdotool key alt+Right'
    down:
      command: 'xdotool key ctrl+minus'
    up:
      command: 'xdotool key ctrl+plus'

threshold:
  swipe: 0.5

interval:
  swipe: 0.1
```

NB: 

- Before that, I modified config.yml to zoom-in via pinching-in. It didn't really work either...
- I'm using Vivaldi 2.11.1811.49 (Stable channel) (64 bits)

iberianpig, 2020-03-30:
Try `xdotool key ctrl+minus`.

BTW, please try `fusuma-plugin-sendkey` I released recently, which has lower latency than xdotool.
https://github.com/iberianpig/fusuma-plugin-sendkey

```yaml
pinch:
  2:
    in:
      sendkey: LEFTCTRL+EQUAL
    out:
      sendkey: LEFTCTRL+MINUS
```

l0f4r0, 2020-03-30:
Thank you :)

I've installed fusuma-plugin-sendkey. All swipe gestures work except for zoom out...
I've tried:

- LEFTCTRL+MINUS
- LEFTCTRL+LEFTSHIFT+MINUS

...with no avail.

PS:

- I have a Lenovo ThinkPad x390 laptop with no numeric keypad so I don't have access to key KPPLUS.

- I think MINUS is the correct key as when pressing "-" key, xev returns the following:

```
KeyPress event, serial 37, synthetic NO, window 0x3a00001,
    root 0x16a, subw 0x0, time 18393293, (68,73), root:(939,519),
    state 0x0, keycode 15 (keysym 0x2d, minus), same_screen YES,
    XLookupString gives 1 bytes: (2d) "-"
    XmbLookupString gives 1 bytes: (2d) "-"
    XFilterEvent returns: False
```

iberianpig, 2020-03-30:
Could you try following settings?

```
pinch:
  2:
    in:
      command: "xdotool keydown ctrl click 4 keyup ctrl"
    out:
      command: "xdotool keydown ctrl click 5 keyup ctrl"
```

Click 4 / 5 mean wheel scroll up / down.

l0f4r0, 2020-03-30:
Not very seamlessly but it works ;)

Do you know:

- why one needs to specify keydown/keyup here wheareas we didn't use it before for 'xdotool key ctrl+minus' for example? I confirm it's important nonetheless because future gestures won't work without it.
- if there is an equivalent click4/5 with sendkey?

iberianpig, 2020-03-30:
> why one needs to specify keydown/keyup here wheareas we didn't use it before for 'xdotool key ctrl+minus' for example? I confirm it's important nonetheless because future gestures won't work without it.

If you need to hold the key a little longer, sometimes using keydown/up will work, such as when displaying Gnome's Dash.

Actually about this, I could find out by trial and error how to pinch-zoom well on Google Maps :sweat_smile:

> if there is an equivalent click4/5 with sendkey?
No, click method couldn't implemented implemented now.

l0f4r0, 2020-03-30:
Too bad for my MINUS key... but OK, I'll make do with it.
If by any chance you have another idea (more seamlessly), I'll take it ;)
Thanks a lot, I'm closing this issue!

closed 2020-03-30 by l0f4r0