iberianpig/fusuma #160: Specific threshold/interval values for each kind of gestures

l0f4r0, 2020-03-30:
Hi,

Is it possible to set threshold & interval values specifically for certain gestures independently from each other?

For example the following doesn't seem to work:
```
swipe:
  3:
    left:
      command: 'xdotool key alt+Left'
    right:
      command: 'xdotool key alt+Right'
    down:
      command: 'xdotool key ctrl+minus'
    up:
      command: 'xdotool key ctrl+plus'
    threshold:
      swipe: 0.5
    interval:
      swipe: 0.1

  4:
    left:
      workspace: 'prev'
    right:
      workspace: 'next'
    down:
      window: 'prev'
    up:
      window: 'next'
    threshold:
      swipe: 0.8
    interval:
      swipe: 0.8
```

Only last threshold/interval values are taken int account for all swipe gestures...

If it's possible, I could be interested to set different values for swipe down/up versus swipe left/right as well.

Maybe I didn't search well but I think documentation doesn't deal with this specific use-cases ;)
Tell me if I'm wrong!

Thank you in advance :)

iberianpig, 2020-03-30:
Certainly, this is a specification that is not made clear on the documentation.

There are three priority of thresholds and intervals.
The individual settings (under "direction") have a higher priority than the general settings ("root")

1. child elements in the direction (left/right/down/up)
2. root child elements
3. default value == 1

```config.yml
swipe:
  3:
    left: 
      command: "echo threshold is 0.2"
      threshold: 0.2 # Priority High
    threshold: 1.5 # Invalid Syntax

threshold:
  swipe: 0.1 # Priority Low

```

l0f4r0, 2020-03-30:
Thanks :)
How can I set threshold/interval values for a X-finger swipe and different ones for Y-finger swipe?
I mean is there a different way than setting those values for each direction inside X or Y (redundant)?

iberianpig, 2020-03-30:
In that case, you will need to make them ”redundant”.
It must be set in each direction.

l0f4r0, 2020-03-30:
Ok, thank you for your prompt answers so far!

Finally, I have the following config.yml:
```
swipe:
  3:
    left:
      command: 'xdotool key alt+Left'
      threshold: 0.5
      interval: 1
    right:
      command: 'xdotool key alt+Right'
      threshold: 0.5
      interval: 1
    down:
      command: 'xdotool key ctrl+minus'
      threshold: 0.1
      interval: 0.1
    up:
      command: 'xdotool key ctrl+plus'
      threshold: 0.1
      interval: 0.1

  4:
    left:
      workspace: 'prev'
      threshold: 0.6
      interval: 0.6
    right:
      workspace: 'next'
      threshold: 0.6
      interval: 0.6
    down:
      window: 'prev'
      threshold: 0.6
      interval: 0.6
    up:
      window: 'next'
      threshold: 0.6
      interval: 0.6
```
There are a lot of redundant information but it works ;)

Then may I suggest you provide a way to specify threshold/interval values for X-finger gestures only?
Maybe you could allow something like the following:
```
swipe:
  3:
    threshold: 1.5
```
Or like this:
```
threshold:
  swipe:
    3: 1.5
```
Or this:
```
threshold:
  3:
    swipe: 1.5
```
We could even imagine specific values for any kind of gestures for X-finger:
```
3:
  threshold: 1.5
```

Maybe there is still a better way. What do you think about it?

iberianpig, 2020-03-30:
Thanks. However, this is a difficult question. 
The configuration flexibility is a good thing, but it also means that the application code becomes more complex.
The syntax in this config.yml makes the fusuma code complicated, and honestly, I don't want to complicate it any more.

l0f4r0, 2020-03-30:
Ok, understood.
I will not insist then but if you could just update the documentation to show what is syntactically correct or not would be great ;)
Thanks!

iberianpig, 2020-03-30:
OK, I'll update README and implement a validator for config.yml.

iberianpig, 2020-04-12:
Implement validator #163 
Update README https://github.com/iberianpig/fusuma/commit/390d777716ce91ac9afda1010f06a2c9f83e9974

closed 2020-04-12 by iberianpig