iberianpig/fusuma #162: pinch_detector.rb: String can't be coerced into Float (TypeError)

SeluMesu, 2020-04-05:
**Describe the bug**
A clear and concise description of what the bug is.

When using two finger pinch it show this error
"/var/lib/gems/2.7.0/gems/fusuma-1.9.0/lib/fusuma/plugin/detectors/pinch_detector.rb:92:in `*': String can't be coerced into Float (TypeError)" and stops working.

**To Reproduce**
Steps to reproduce the behavior:
1. [First Step]
on terminal run:  fusuma
2. [Second Step]
start pinching with two fingers

**Versions**
You can get this information from copy and paste the output of `fusuma --version` from the command line. Also, please include the OS and what version of the OS you're running.
Fusuma: 1.9.0
I, [2020-04-04T20:22:15.931225 #24663]  INFO -- : ---------------------------------------------
I, [2020-04-04T20:22:15.931257 #24663]  INFO -- : Fusuma: 1.9.0
I, [2020-04-04T20:22:15.932356 #24663]  INFO -- : libinput: 1.15.4
I, [2020-04-04T20:22:15.932735 #24663]  INFO -- : OS: Linux 5.4.0-21-generic #25-Ubuntu SMP Sat Mar 28 13:10:28 UTC 2020
I, [2020-04-04T20:22:15.933084 #24663]  INFO -- : Distribution: Ubuntu Focal Fossa (development branch) \n \l
I, [2020-04-04T20:22:15.933467 #24663]  INFO -- : Desktop session: ubuntu x11
I, [2020-04-04T20:22:15.933484 #24663]  INFO -- : ---------------------------------------------
I, [2020-04-04T20:22:15.933492 #24663]  INFO -- : Enabled Plugins: 
I, [2020-04-04T20:22:15.933519 #24663]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-04-04T20:22:15.933527 #24663]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-04-04T20:22:15.933533 #24663]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-04-04T20:22:15.933538 #24663]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-04-04T20:22:15.933543 #24663]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-04-04T20:22:15.933549 #24663]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-04-04T20:22:15.933554 #24663]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-04-04T20:22:15.933559 #24663]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-04-04T20:22:15.933565 #24663]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-04-04T20:22:15.933570 #24663]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-04-04T20:22:15.933575 #24663]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-04-04T20:22:15.933581 #24663]  INFO -- : ---------------------------------------------

iberianpig, 2020-04-05:
@SeluMesu Could you paste your config.yml?

SeluMesu, 2020-04-05:
Im so sorry. I just looked at the config file and i accidentally added a letter at the end of pinch interval. like this 
interval:
  pinch: 0.1Z
im sorry again. it was my fault.

iberianpig, 2020-04-05:
Thanks for your reporting 👍
This is not your fault.

I'll implement validator for pre-checking config.yml.

https://github.com/iberianpig/fusuma/issues/163

closed 2020-04-12 by iberianpig