iberianpig/fusuma #164: ruby2.7

minzukas, 2020-04-08:
after terminal: fusuma i get
_bash: /usr/local/bin/fusuma: /usr/bin/ruby2.5: bad interpreter: No such file or directory
currently ruby2.7 installed_

iberianpig, 2020-04-08:
Are you using rbenv?

Could you confirm that is there following script in your ~/.bashrc ?

```sh
export PATH="$HOME/.rbenv/bin:$PATH" 
eval "$(rbenv init -)
```

minzukas, 2020-04-09:
I'm currently on Focal Fossa development. Maybe it is too early to report...

./bashrc  is empty.

iberianpig, 2020-04-12:
Execute the following command to reinstall the software.
It is possible that an OS update may have left behind an old Ruby installation of Fusuma.
```sh
sudo gem uninstall fusuma
sudo gem install fusuma
```

minzukas, 2020-04-13:
Thank You. Had to

> sudo gem install fusuma

works like a charm

closed 2020-04-13 by minzukas