iberianpig/fusuma #165: Hot to recreate Windows 10-alike Alt-Tab switching with three fingers?

emvaized, 2020-04-11:
In Windows 10, task switcher opens on three-finger swipe start, and when you continue movement, you scroll through tasks, and select one on fingers release.

Is is possible to implement something similar using fusuma? If it is, maybe some example should be provided in the documentation (which is very great, by the way - thanks!).

iberianpig, 2020-04-12:
I haven't used Windows in about ten years, so if I'll try it :smile: 

I think it's still going to be challenging to achieve that function.
Many problems need to be solved to make this happen.

* Implementation of the three-finger scrolling gesture
* Recognition of gesture combinations
   * 3-finger scroll up → ALT+TAB &&Continue to press ALT
   * Three fingers left/right scroll → SHIFT+TAB OR TAB
   * End the three-fingered scrolling → release the TAB key.
* Integration with WindowManager
   * Fusuma needs to know about the UI
   *  Avoid keeping pressing ALT when another action finishes the task switching.

emvaized, 2020-04-12:
@iberianpig 
Here's little git I found with google to illustrate the principle :)
[image]

emvaized, 2020-04-12:
@iberianpig 
Oh, I see now - it's not possible yet. Well, I hope that some day this feature will be implemented, as it's actually very useful.

Then, I want to ask one more little question, for the matter of not opening new issue - I heard it is possible to regulate screen volume & brightness with fusuma the same way? E.g. fingers swipe regulate volume dynamically - the more far you swipe, the more volume/brightness increased.

It's not possible yet as well? Thanks.

iberianpig, 2020-04-12:
Yes, Fusuma can trigger volume up/down
 or brightness up/down keys continuously.
Set a short values like 0.3 to threshold and interval of gestures.

emvaized, 2020-04-12:
@iberianpig 
I see.
So, for example, I have an `Alt+F10` shortcut for increasing volume.
So, `config.yml` setting should like somewhat like that?.. (code below triggers volume-up only once in my case)
```
 4:
    left:
      command: "xdotool keydown alt; xdotool key F9;sleep 0.3;xdotool keyup alt" 
    right:
      command: "xdotool keydown alt; xdotool key F10;sleep 0.3;xdotool keyup alt" 
```

emvaized, 2020-04-12:
@iberianpig 
I understood. You were talking about those:
```
threshold:
  swipe: 0.3
interval:
  swipe: 0.3
```
It works now! Thanks a lot.

closed 2020-06-20 by stale[bot]

iberianpig, 2021-09-06:
ref: https://github.com/iberianpig/fusuma/wiki/3-fingers-Alt-Tab-Switcher(Windows-Style)