iberianpig/fusuma #168: Multi finger tab word lookup

volfgox, 2020-05-03:
I want to enable looking up a word for instance in a we page by 3 fingers tap over it in Ubuntu. Is this possible with fusuma (or maybe with additional fusuma plugins)?

iberianpig, 2020-05-03:
I think you can do that.
Try [fusuma-plugin-tap](https://github.com/iberianpig/fusuma-plugin-tap).

You need to make a script that copies the word you are highlighting and runs it as an argument in your browser or dictionary.

iberianpig, 2020-05-09:
volfgox Please install `xsel` and `fusuma-plugin-tap` and try following

```yaml
tap:
  3:
    command: 'xdg-open https://www.google.com/search?q="$(xsel -p)"'
```

volfgox, 2020-05-09:
Thank you. This helps a lot.
Furthermore, it would be great if the word is selected before opening the URL. Currently, I should select the word with a tap-2 and then a tap-3 to lookup. Merging both of these operations into just the tap-3 will be wonderful. Any idea?

iberianpig, 2020-05-10:
I think that it would be nice to have a chrome extension that search word under the cursor.

volfgox, 2020-05-10:
It would be nicer if this feature could be available everywhere not just in a browser. Just like in Mac OS.
The only missing feature to achieve this is highlighting the word under cursor with tap-3. What do you think? Is it possible this with fusuma? If not, is it possible to add the feature? I can contribute to add new features.

iberianpig, 2020-05-10:
You may need a small dictionary app that pops up at the cursor position.

I think it would be difficult to implement the ability to make fusuma highlight characters.
Instead, it seems easy to run two commands in a row: "double-click to select a character" and "search for the selected character".

volfgox, 2020-05-10:
Thank you very much. I finally come up with a solution that works properly for me:
```yaml
tap:
  3:
    command: 'xdotool click --repeat 2 1 && xdg-open https://www.google.com/search?q="$(xsel -p)"'
```

iberianpig, 2020-05-11:
I found the chrome extension as a dictionary app [Google Dictionary (by Google)](https://chrome.google.com/webstore/detail/google-dictionary-by-goog/mgijmajocgfcbeboacabfgobmjgjcoja/related)

# options
[image]

ctrl + double click

# config.yml

```
  3:
    command: 'xdotool keydown ctrl click --repeat 2 1 keyup ctrl'
```

ctrl press → double click → ctrl release

# demo
[image]

iberianpig, 2020-05-11:
http://goldendict.org/ would be better for you.
This local app has a nice popup window feature.

```
tap: 
  3:
    command: "xdotool click --repeat 2 1 && goldendict $(xsel -p)"
```

[image]

closed 2020-07-18 by stale[bot]