iberianpig/fusuma #171: Unable to reprogram Pinch

fookitt, 2020-05-10:
**Describe the bug**
Loaded config.yml with PopOS Default Gnome config. Tried to reprogram pinch in and out, but it is 'stuck' to commands Ctrl+Super+Up' (pinch in, move to workspace up) and Ctrl+Super+Down' (pinch out, move to workspace down). I currently have no config for pinch actions and yet it is still 'stuck' to those commands.

I have reloaded Fusuma by running Fusuma with -d. 
I have log off and log back on, and yet the pinch actions are 'stuck' to those shortcut.
I am able to reprogram other gestures. 

Running the command fusuma to list actions/commands, I am able to see other gestures but not pinch (currently set to nothing) and yet it activates the command to move up/down to other workspace.

**Versions**
Fusuma: 1.10.0
libinput: 1.15.5
Linux 5.4.0-7626-generic #30~1588169883~20.04~bbe668a-Ubuntu SMP Wed Apr  Pop!_OS 20.04 LTS \n \l
Desktop session: pop x11

**Additional context**
```
swipe:
  4:
    up:
      command: 'xdotool key Ctrl+Super+Down'
    down:
      command: 'xdotool key Ctrl+Super+Up'
    left:
      command: 'xdotool key Ctrl+Super+Left'
    right:
      command: 'xdotool key Ctrl+Super+Right'

  3:
    up:
      command: 'xdotool key Super+s'
    down:
      command: 'xdotool key Super+d'
    left:
      command: 'xdotool key Alt+Right'
    right:
      command: 'xdotool key Alt+Left'

threshold:
  swipe: 1
  pinch: 1

interval:
  swipe: 1
  pinch: 1

```

iberianpig, 2020-05-10:
Please paste outputs of `fusuma --version`
I'd like to know that fusuma loaded your configuration correctly.

fookitt, 2020-05-10:
```
I, [2020-05-11T07:58:15.616341 #25248]  INFO -- : ---------------------------------------------
I, [2020-05-11T07:58:15.616373 #25248]  INFO -- : Fusuma: 1.10.0
I, [2020-05-11T07:58:15.617521 #25248]  INFO -- : libinput: 1.15.5
I, [2020-05-11T07:58:15.617938 #25248]  INFO -- : OS: Linux 5.4.0-7626-generic #30~1588169883~20.04~bbe668a-Ubuntu SMP Wed Apr 29 21:00:02 UTC
I, [2020-05-11T07:58:15.618339 #25248]  INFO -- : Distribution: Pop!_OS 20.04 LTS \n \l
I, [2020-05-11T07:58:15.618698 #25248]  INFO -- : Desktop session: pop x11
I, [2020-05-11T07:58:15.618720 #25248]  INFO -- : ---------------------------------------------
I, [2020-05-11T07:58:15.618728 #25248]  INFO -- : Enabled Plugins: 
I, [2020-05-11T07:58:15.618753 #25248]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-05-11T07:58:15.618760 #25248]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-05-11T07:58:15.618765 #25248]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-05-11T07:58:15.618771 #25248]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-05-11T07:58:15.618776 #25248]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-05-11T07:58:15.618781 #25248]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-05-11T07:58:15.618786 #25248]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-05-11T07:58:15.618791 #25248]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-05-11T07:58:15.618796 #25248]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-05-11T07:58:15.618800 #25248]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-05-11T07:58:15.618815 #25248]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-05-11T07:58:15.618820 #25248]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-05-11T07:58:15.618825 #25248]  INFO -- : ---------------------------------------------

```

iberianpig, 2020-05-11:
Thank you. 
Can you also paste the output of `fusuma -v`?

You can find line like following
```
I, [2020-05-11T10:13:20.004947 #29280]  INFO -- : reload config : /home/iberianpig/.config/fusuma/config.yml
```

fookitt, 2020-05-11:
```
I, [2020-05-11T13:55:34.605801 #25864]  INFO -- : ---------------------------------------------
I, [2020-05-11T13:55:34.605838 #25864]  INFO -- : Fusuma: 1.10.0
I, [2020-05-11T13:55:34.606993 #25864]  INFO -- : libinput: 1.15.5
I, [2020-05-11T13:55:34.607431 #25864]  INFO -- : OS: Linux 5.4.0-7626-generic #30~1588169883~20.04~bbe668a-Ubuntu SMP Wed Apr 29 21:00:02 UTC
I, [2020-05-11T13:55:34.607815 #25864]  INFO -- : Distribution: Pop!_OS 20.04 LTS \n \l
I, [2020-05-11T13:55:34.608301 #25864]  INFO -- : Desktop session: pop x11
I, [2020-05-11T13:55:34.608317 #25864]  INFO -- : ---------------------------------------------
I, [2020-05-11T13:55:34.608325 #25864]  INFO -- : Enabled Plugins: 
I, [2020-05-11T13:55:34.608408 #25864]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-05-11T13:55:34.608417 #25864]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-05-11T13:55:34.608424 #25864]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-05-11T13:55:34.608430 #25864]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-05-11T13:55:34.608436 #25864]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-05-11T13:55:34.608442 #25864]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-05-11T13:55:34.608447 #25864]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-05-11T13:55:34.608453 #25864]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-05-11T13:55:34.608473 #25864]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-05-11T13:55:34.608478 #25864]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-05-11T13:55:34.608483 #25864]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-05-11T13:55:34.608497 #25864]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-05-11T13:55:34.608503 #25864]  INFO -- : ---------------------------------------------
I, [2020-05-11T13:55:34.608941 #25864]  INFO -- : reload config : /home/foo/.config/fusuma/config.yml
D, [2020-05-11T13:55:34.609938 #25864] DEBUG -- : {:debug_events=>"stdbuf -oL -- libinput debug-events"}
D, [2020-05-11T13:55:34.624152 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event2   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k}
D, [2020-05-11T13:55:34.624477 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event4   DEVICE_ADDED     Video Bus                         seat0 default group2  cap:k}
D, [2020-05-11T13:55:34.624565 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event1   DEVICE_ADDED     Lid Switch                        seat0 default group3  cap:S}
D, [2020-05-11T13:55:34.624641 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event0   DEVICE_ADDED     Sleep Button                      seat0 default group4  cap:k}
D, [2020-05-11T13:55:34.625066 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event7   DEVICE_ADDED     Integrated Camera: Integrated C   seat0 default group5  cap:k}
D, [2020-05-11T13:55:34.625380 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event9   DEVICE_ADDED     HDA Intel PCH Mic                 seat0 default group6  cap:}
D, [2020-05-11T13:55:34.625568 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event10  DEVICE_ADDED     HDA Intel PCH Headphone           seat0 default group6  cap:}
D, [2020-05-11T13:55:34.625803 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event11  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=3       seat0 default group6  cap:}
D, [2020-05-11T13:55:34.626036 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event12  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=7       seat0 default group6  cap:}
D, [2020-05-11T13:55:34.626260 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event13  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=8       seat0 default group6  cap:}
D, [2020-05-11T13:55:34.626570 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event14  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=9       seat0 default group6  cap:}
D, [2020-05-11T13:55:34.626835 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event15  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=10      seat0 default group6  cap:}
D, [2020-05-11T13:55:34.626973 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event5   DEVICE_ADDED     Elan Touchpad                     seat0 default group7  cap:pg  size 98x54mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on}
D, [2020-05-11T13:55:34.627108 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:34 +1000, tag: libinput_command_input, record: -event6   DEVICE_ADDED     Elan TrackPoint                   seat0 default group8  cap:p left scroll-nat scroll-button}
[... cut ...]
D, [2020-05-11T13:55:35.741768 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:35 +1000, tag: libinput_command_input, record:  event5   POINTER_MOTION   +0.084s	 10.03/ -6.22 (+29.00/-18.00)}
D, [2020-05-11T13:55:35.748583 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:35 +1000, tag: libinput_command_input, record:  event5   POINTER_MOTION   +0.091s	 13.49/ -6.55 (+35.00/-17.00)}
D, [2020-05-11T13:55:35.755400 #25864] DEBUG -- : {:input_event=>time: 2020-05-11 13:55:35 +1000, tag: libinput_command_input, record:  event5   POINTER_MOTION   +0.098s	 15.01/ -6.09
```

iberianpig, 2020-05-11:
Please paste outputs of `ps aux | grep fusuma`.
You're running more than one Fusuma process, aren't you?
`-d` option is not a reload. Just run it and detach it from the terminal (daemonize).

If you want to stop the remaining Fusuma process, you only need to `kill PID`.

fookitt, 2020-05-11:
You're right! But they all have -d option though. 
How do I re-apply the config file after making changes?

```
foo        13316  0.2  0.1 367024 28960 ?        Sl   May10   3:02 /usr/bin/ruby2.7 /usr/local/bin/fusuma -d
foo        14713  0.2  0.1 366992 28764 ?        Sl   May10   2:49 /usr/bin/ruby2.7 /usr/local/bin/fusuma -d
foo        16170  0.2  0.1 366984 28740 ?        Sl   May10   2:36 /usr/bin/ruby2.7 /usr/local/bin/fusuma -d
foo        16331  0.1  0.1 299396 28580 ?        Sl   May10   2:34 /usr/bin/ruby2.7 /usr/local/bin/fusuma -d
foo        16984  0.1  0.1 366848 28720 ?        Sl   May10   2:34 /usr/bin/ruby2.7 /usr/local/bin/fusuma -d
foo        20580  0.1  0.1 366984 28364 ?        Sl   May10   2:11 /usr/bin/ruby2.7 /usr/local/bin/fusuma -d
foo        37561  0.0  0.0  18804   728 pts/0    S+   17:49   0:00 grep --color=auto fusuma
```

iberianpig, 2020-05-11:
Just kill Fusuma processes and type `fusuma -d` in terminal.

closed 2020-05-15 by fookitt

elvisisvan, 2023-03-27:
> Just kill Fusuma processes and type `fusuma -d` in terminal.

what's the process name of fusuma? there should be ```fusuma -s``` or something to stop daemon though