iberianpig/fusuma #172: Some gestures are not always recognized

Trickytram, 2020-05-13:
I am using fusuma only with swipe gestures configured.
I am using three finger swipes and four finger swipes. I've never run into any issues using the four finger swipe (which switches between desktops for me), whereas the three finger swipe (which I use to go back between pages in the browser) frequently does not work. However, when I change the desktops using the four finger swipe there is a high probability that the three finger swipe gesture will work again.

Sadly I dont know if I can provide any helpful logs, so if I can please tell me.

Thanks for your help.

iberianpig, 2020-05-13:
@Trickytram 
Could you paste outputs of `fusuma --version`?

Trickytram, 2020-05-14:
Here you go:

I, [2020-05-14T08:49:16.615568 #14931]  INFO -- : ---------------------------------------------
I, [2020-05-14T08:49:16.615606 #14931]  INFO -- : Fusuma: 0.11.1
I, [2020-05-14T08:49:16.618431 #14931]  INFO -- : libinput: 1.10.4
I, [2020-05-14T08:49:16.618984 #14931]  INFO -- : OS: Linux 5.5.15-050515-generic #202004021032 SMP Thu Apr 2 14:37:37 UTC 2020
I, [2020-05-14T08:49:16.619928 #14931]  INFO -- : Distribution: Ubuntu 18.04.4 LTS \n \l
I, [2020-05-14T08:49:16.620394 #14931]  INFO -- : Desktop session: plasma
I, [2020-05-14T08:49:16.620417 #14931]  INFO -- : ---------------------------------------------

iberianpig, 2020-05-14:
Your Fusuma version is too old!
Please update with `sudo gem update fusuma`

Trickytram, 2020-05-14:
Alright I will do that.
Thank you. I will close this issue.
If the problem reappears I will open another issue.

closed 2020-05-14 by Trickytram