iberianpig/fusuma #173: Three finger drag

buzuddha, 2020-05-13:
First, let me say: WOW! What a wonderful piece of software you've written! This truly makes ubuntu feel closer to the mac trackpad experience and I really value what you've done here.

Two mac trackpad features that I use constantly are the three-finger select and the three-finger drag. Perhaps you are aware, but if not, they work as follows:

hold down three fingers on text and swipe -> selects text as if left click was held during swipe
hold down three fingers on the top of a window or a file icon -> drags file/window as if left click was held during swipe
 
I have considered this program:
https://martys.typepad.com/blog/2015/08/3-finger-drag-on-linux.html
However I really want to start using two touchpad programs at the same time. That sound like it will just cause problems, plus this page looks old anyhow and may not be compatible with ubuntu 20.04.

I'm sure you're quite busy, and I don't know how widely used the three-finger finger drag/select is. Mac took it out of the main touchpad settings and hid it deeper in the preferences, so perhaps I'm alone in liking this one. My wife doesn't use it on her work computer so perhaps it's not popular.

I'm not sure I would like to sign up as a patreon long term, but I'll be glad to send $10 your way for fusuma as is!

Thanks!

iberianpig, 2020-05-13:
Hi @buzuddha, 
I was surprised to find PayPal and this issue 10 minutes ago. Thank you, I'm very happy.

I'm thinking of implementing a three-finger scrolling feature. #170 

I believe this is closer to the drag function you describe.
Simply put, the idea is for a feature that detects slow swipes.

iberianpig, 2020-05-13:
Assume that I add a new parser plugin to Fusuma for a scrolling event.
It generates events like "scroll begin", "scroll update", "scroll end".
By assigning actions to these, the 3 or 4 finger scrolling function can be achieved.

I think there is another issue with the create the Drag function.
The libinput that Fusuma depends on will stop moving the cursor when it detects a 3,4 finger gesture. I think that It's hard to unlock this.

But, I noticed that external touchpad or mouse can also moves cursor.
Moving the cursor on that uinput virtual device to emulate the cursor's movement.
The cursor follows even with three fingers, so it should look like it's behaving like a real drag function.

buzuddha, 2020-05-14:
Interesting. I love the idea of creating a virtual second trackpad that doesn't stop. Are you suggesting you can just create a second input device, then switch to the second (that doesn't stop) once the three fingers are detected on the first input device?

I love your creativity with this!
Hugh

iberianpig, 2020-05-14:
Yes, that's right!
It's going to take some time, but I believe this can be implemented.

buzuddha, 2020-05-14:
I'm not sure if I have the expertise, but let me know if you need help
testing it!

pgiraud, 2020-11-25:
> so perhaps I'm alone in liking this one.

No you're not! I use it a lot on my mac as well.
It works for anything that can be dragged: for example (window) resize handles.
I'd love to see something implemented.

ggrrll, 2020-12-08:
Thanks also from my side for the nice tool! -- it really helps me transitioning back to Ubuntu :sweat_smile: 

btw, I would  also appreciate very much the 3-fingers text selection  -- happy to help testing of course 
(unfortunately I don't have experience with ruby :/ )

andrenbrandao, 2020-12-20:
@ggrrll I also would want to have the 3-fingers text selection. Anyway, I am already happy by having the 3-finger and 4-finger movements!

petko10, 2021-01-01:
Here's a workaround https://github.com/daveriedstra/draggy written in python, using evdev (not mine). It can work alongside fusuma, but mind you there's no conditional suppression of the three finger gestures by draggy (so both the drag event - if you have one configured -  and the fusuma gesture are activated). The other option I saw was a fork of libinput-gestures by the same dev, but I had trouble testing it. Draggy seems like the best available option until this functionality is integrated in fusuma, since the libinput-gesture devs will not be integrating such functionality.

iberianpig, 2021-02-24:
I'm implementing real-time, high-frequency updating gestures right now.

[image]

petko10, 2021-03-01:
Given the timing I'll guess you're using the implementation in the new libinput release? https://www.phoronix.com/scan.php?page=news_item&px=Libinput-1.17-Released

iberianpig, 2021-03-01:
I was very surprised as I did not know about the new 1.17 release.

My implementation works with the old libinput so far. It's just a simple hack to bind the start, update and end of swipe and pinch events to yaml.

However, I'm curious about the API update for libiput 1.17. Thanks for letting me know!

buzuddha, 2021-03-05:
That looks just like what I'm hoping for! Please let us know when you release this and also how to implement it on our own computers!

iberianpig, 2021-03-06:
I'm planning to release fusuma v2 soon. Then 3-finger dragging will be available, so please wait a bit👍
@buzuddha 
I'll let you know in this thread😊

buzuddha, 2021-03-06:
Cool! Congratulations!

iberianpig, 2021-04-12:
@buzuddha 

Thanks for your patience. fusuma v2 has been released.
Now you can support continuous gestures and contexts.

see: https://github.com/iberianpig/fusuma/pull/224

Please update fusuma and plugins
```sh
sudo gem update fusuma
sudo gem update fusuma-plugin-XXXX
````

Three-finger drag can be configured with the following config.
```yaml
swipe:
  3:
    begin:
      command: xdotool mousedown 1
    update:
      command: xdotool mousemove_relative -- $move_x, $move_y
      interval: 0.01
      accel: 2
    end:
      command: xdotool mouseup 1
```

petko10, 2021-04-12:
Thanks for implementing that @iberianpig , it works great.

closed 2021-04-24 by iberianpig

iberianpig, 2024-03-22:
I get the need for this feature. 

We must change the swipe detector or create an another detector to keep dragging gesture mode. It could be tough. It might need to ignore when fingers reduce to one or lift off to keep in drag mode.

jncsotk30, 2024-04-16:
For anyone looking for help additional help with this on wayland, you have to use [ydotool](https://github.com/ReimuNotMoe/ydotool) instead of xdotool. The syntax is a little bit different, but it still works. Note, I'm using Fedora 39 | 6.8.5 Kernel. Shouldn't matter too much, but just letting you know.

First install ydotool via `dnf` or `yum`...

`sudo dnf install ydotool`

Once installed, if you try to run it, it will be started as root which won't work for this purpose. You need to modify the ydotool service to start as your user as well as set the socket path for the tool. To do that, modify the `ydotool.service` file under `/usr/lib/systemd/system`. 

You can use...
`sudo vim /usr/lib/systemd/system/ydotool.service`

In the `ExecStart` line, you'll need to add two parameters for ydotool...

1. `--socket-own=[UID]:[GID]`
2. `--socket-path=/home/[your-user]/ydotool_socket`

By default, I think the socket-path is set to `/tmp/ydotool_socket`, but because we're running ydotool as our user in the service, it instead looks under home dir. I think you can set this to wherever, but I did my home dir.

To get the UID & GID, you can use a terminal to echo those environmental variables.

`echo $UID`
`echo $GID`

Those two numbers will be put in place in the `--socket-own` parameter separated by colon(:).

Once the ExecStart line is edited, it should look like this...

`ExecStart=/usr/bin/ydotool --socket-own=1000:1000 --socket-path=/home/[your-username]/.ydotool_socket`

**Of course, replace [your-username] with your actual username**

After that, you can save and exit vim, then enable & start your ydotool service.

`sudo systemctl enable ydotool`
`sudo systemctl start ydotool`

Check the status of the service to ensure startup was okay.

---

After ydotool is configured properly, follow the instructions for adding your user to the `input` group in fusuma docs (best to logout after adding so the change sticks). After that, setup your fusuma config file with the following to enable three finger drag.

```
swipe:
  3:
    begin:
      command: ydotool click 40
    update:
      command: ydotool mousemove -- $move_x, $move_y
      interval: 0.00
      accel: 2
    end:
      command: ydotool click 80
```

tomyrseth, 2024-05-05:
Sorry not sure where to ask, but is fusuma supported on ubuntu 24.04 LTS? 3 finger text select works somewhat, had to disable multiple workspaces (desktops), but when I do 3 finger swipe up it does the same as pressing windows button (super button) which is annoying I havent found a way to disable it in settings.

3 finger swipe to move windows around does not work, the mouse stays still in place while output in terminal is:

```
I, [2024-05-05T20:46:45.890870 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.97, :move_y=>0.0, :unaccelerated_x=>-1.02, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:45.911393 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.99, :move_y=>0.0, :unaccelerated_x=>-1.02, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:45.930954 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.51, :move_y=>0.0, :unaccelerated_x=>-0.68, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:45.950907 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.32, :move_y=>0.0, :unaccelerated_x=>-0.45, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:45.971025 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.51, :move_y=>0.0, :unaccelerated_x=>-0.68, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:46.032017 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.47, :move_y=>0.09, :unaccelerated_x=>-0.57, :unaccelerated_y=>0.11, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:46.110884 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.39, :move_y=>0.0, :unaccelerated_x=>-0.57, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:46.120695 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.68, :move_y=>0.0, :unaccelerated_x=>-0.68, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:46.140232 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.37, :move_y=>0.09, :unaccelerated_x=>-0.45, :unaccelerated_y=>0.11, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:46.160475 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.43, :move_y=>0.0, :unaccelerated_x=>-0.68, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:46.220973 #12346]  INFO -- : {:command=>"xdotool mousemove_relative -- $move_x, $move_y", :args=>{:move_x=>-0.31, :move_y=>0.0, :unaccelerated_x=>-0.45, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
I, [2024-05-05T20:46:46.493037 #12346]  INFO -- : {:command=>"xdotool mouseup 1", :args=>{:move_x=>-0.05, :move_y=>0.0, :unaccelerated_x=>-0.11, :unaccelerated_y=>0.0, :zoom=>0.0, :rotate=>0.0}}
```

I followed this https://github.com/iberianpig/fusuma/wiki/3-fingers-Drag-(OS-X-Style)

iberianpig, 2024-05-06:
Are you using Wayland? 
The three-finger swipe is controlled by the compositor on GNOME wayland and cannot change it, so on the fusuma side, you could assign the swipe to a different number of fingers, or create a Gnome override like this https://gist.github.com/iberianpig/6e7049a02489950f570b4a27cff8df5e to change the default swipe tracker from 3 fingers to 4, or to 5 to disable it.

lVentus, 2024-05-16:
> I get the need for this feature.
> 
> We must change the swipe detector or create an another detector to keep dragging gesture mode. It could be tough. It might need to ignore when fingers reduce to one or lift off to keep in drag mode.

I''m thinking, can we do it like this. give 'end' event a property 'delay', for example 500 ms. when 'end' event happen, it delay to run the command. if before the time finish, 'begin' happened again, then do not run 'end' command, and keep 'update' command.

Kaeen, 2024-10-18:
> I''m thinking, can we do it like this. give 'end' event a property 'delay', for example 500 ms. when 'end' event happen, it delay to run the command. if before the time finish, 'begin' happened [...]

Yes, currently the only difference between native MacOS behavior and the three-finger-drag-n-drop one is that using fusuma I can't quickly resume to drag-n-drop action after I put fingers off my touchpad (e.g. if I need longer drag while my fingers met touchpad border). 

Maybe there could be a hack like delayed command launch (I mean launching script with delayed 'end' command and cancelling that delayed 'end' command in 'begin' command). Have strong concern on performance. :)

Nashenas88, 2024-12-20:
@Kaeen I got it working with the following configuration:
```
swipe:
  3:
    begin:
      command: 'rm "$HOME/.config/fusuma/delay"; ydotool click 40'
      interval: 0.00
    update:
      command: ydotool mousemove -- $move_x, $move_y
      interval: 0.01
      accel: 2.00
    end:
      command: 'touch "$HOME/.config/fusuma/delay"; sleep 1; if [ -f "$HOME/.config/fusuma/delay" ]; then ydotool click 80; fi'
      interval: 0.0
```

The only issue I'm seeing is that moving the mouse (even with just one finger) will still move the highlighted area until the timer is done. It's not perfect but definitely lets you lift up and continue the drag where you left off.

leonardorifeli, 2024-12-20:
tks @iberianpig, it works great!