iberianpig/fusuma #174: Pinch (2, in) does not work.

HubKing, 2020-05-14:
Pinch (2, out) works, but pinch (2, in) does not work. I have checked that libinput debug and it got the pinch in event.

Here is my config.yml I had used an empty setting for pinch out, because I was not sure whether the gesture's (two fingers moving into opposite directions)  name is in or out.

```
swipe:
 
pinch:
  2: 
    in:
       sendkey: "LEFTCTRL+EQUAL" # threshold: 0.5, interval: 0.5
  2: 
    out:
       sendkey: "" # threshold: 0.5, interval: 0.5

threshold:
  pinch: 0.5

interval:
  swipe: 0.75
  pinch: 0.5
```

Here is the output of libinput when I performed pinch in.

```
-event5   GESTURE_PINCH_BEGIN +3.590s	2
 event5   GESTURE_PINCH_UPDATE +3.590s	2  1.85/-1.85 ( 9.84/-9.84 unaccelerated)  1.04 @  0.37
 event5   GESTURE_PINCH_UPDATE +3.597s	2  1.76/-3.08 ( 6.56/-11.48 unaccelerated)  1.09 @  0.49
 event5   GESTURE_PINCH_UPDATE +3.604s	2  0.44/-5.72 ( 1.64/-21.33 unaccelerated)  1.17 @  0.28
 event5   GESTURE_PINCH_UPDATE +3.611s	2 -1.76/-7.04 (-6.56/-26.25 unaccelerated)  1.25 @  0.25
 event5   GESTURE_PINCH_UPDATE +3.618s	2 -2.20/-4.84 (-8.20/-18.04 unaccelerated)  1.33 @  0.50
 event5   GESTURE_PINCH_UPDATE +3.626s	2  0.00/-7.92 ( 0.00/-29.53 unaccelerated)  1.44 @  0.51
 event5   GESTURE_PINCH_UPDATE +3.633s	2 -3.52/-3.52 (-13.12/-13.12 unaccelerated)  1.52 @  0.27
 event5   GESTURE_PINCH_UPDATE +3.640s	2 -4.40/-0.44 (-16.40/-1.64 unaccelerated)  1.59 @  0.41
 event5   GESTURE_PINCH_UPDATE +3.647s	2 -3.96/-2.64 (-14.76/-9.84 unaccelerated)  1.66 @  0.51
 event5   GESTURE_PINCH_UPDATE +3.654s	2 -3.52/-2.64 (-13.12/-9.84 unaccelerated)  1.73 @  0.63
 event5   GESTURE_PINCH_UPDATE +3.661s	2 -3.96/-3.08 (-14.76/-11.48 unaccelerated)  1.77 @  0.70
 event5   GESTURE_PINCH_UPDATE +3.669s	2 -3.96/-1.76 (-14.76/-6.56 unaccelerated)  1.83 @  1.04
 event5   GESTURE_PINCH_UPDATE +3.676s	2  2.64/-1.76 ( 9.84/-6.56 unaccelerated)  1.85 @  0.21
 event5   GESTURE_PINCH_UPDATE +3.683s	2  0.00/-0.44 ( 0.00/-1.64 unaccelerated)  1.85 @  0.05
 event5   GESTURE_PINCH_END +3.689s	2
```

Here is the output of Fusuma from the moment before I did the libinput test above till now (the moment I am writing this). I had done some pinch out gestures before the test.

```
I, [2020-05-14T01:23:49.407187 #3120]  INFO -- : reload config : /home/user/.config/fusuma/config.yml
I, [2020-05-14T01:23:57.074437 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:23:57.196394 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:23:57.246917 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:23:57.297170 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:23:57.355030 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:24:39.113313 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:24:39.163722 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:24:39.220623 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:24:39.271234 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:24:39.321517 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:24:39.415447 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:52.826423 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:52.877131 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:52.934274 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:52.984779 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:53.035313 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:53.185810 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:53.903409 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:53.954039 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:54.004943 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:54.062029 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:54.118473 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:54.169192 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:54.219088 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:56.662928 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:56.719580 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:56.769744 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:56.826836 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:56.877665 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:28:56.928048 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:29:02.631550 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:29:02.682082 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:29:02.731927 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:29:02.782401 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:43.964511 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:44.021759 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:44.072159 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:44.122516 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:44.172703 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:48.829378 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:48.880141 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:48.930297 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:48.980599 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:49.031113 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:49.082050 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:49.139087 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:49.196427 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:49.246742 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:49.297542 #3120]  INFO -- : {:sendkey=>""}
I, [2020-05-14T01:30:49.349088 #3120]  INFO -- : {:sendkey=>""}
```

**To Reproduce**
Steps to reproduce the behavior:
1. [First Step] Install Ubuntu 20.04
2. [Second Step] Pinch in (two fingers moving in opposite directions)
3. [and so on...]

**Expected behavior**
Work?

**Versions**
Must be the latest, because I installed it a few tens of minutes ago using the commands at the front page.

**Additional context**
A Precision touchpad.

iberianpig, 2020-05-20:
Hi @HubKing,

I think you have invalid syntax.

Please remove `2:` key above "out:" in config.yml.
```diff
swipe:
 
pinch:
  2: 
    in:
       sendkey: "LEFTCTRL+EQUAL" # threshold: 0.5, interval: 0.5
-   2: 
    out:
       sendkey: "" # threshold: 0.5, interval: 0.5

threshold:
  pinch: 0.5

interval:
  swipe: 0.75
  pinch: 0.5
```

Fusuma parse the config.yml and convert to Hash from YAML.

`Pinch in` key is overwritten by `Pinch out` key because Hash cannot have the same key.

HubKing, 2020-05-20:
Yeah, now it works. Thanks.
If two instances of the same number of fingers (I guess "2" represents the number of fingers involved), even for different directions, is not valid, maybe showing some warning message like "Multiple instances of Pinch/2 are defined and preceding instance(s) are ignored, are you sure this is what you want?" in the console output would be helpful to prevent this kind of thing.

closed 2020-05-20 by HubKing

iberianpig, 2020-05-21:
I prefer to avoid interactive interfaces.
I would implement an option to output the assigned values instead.