iberianpig/fusuma #175: Fusuma stops working sometimes

Ru5h4d, 2020-05-15:
I have set fusuma to run on starup in KDE's system settings. It stops working after a few moments sometimes. The log for two instances of the scenario is provided. 

**Describe the bug**
Just stops working and gives an error message. Noticed while toggling workspace overview or opening / closing tabs.

**Versions**
```
I, [2020-05-15T16:22:04.003994 #24351]  INFO -- : ---------------------------------------------
I, [2020-05-15T16:22:04.004027 #24351]  INFO -- : Fusuma: 1.10.0
I, [2020-05-15T16:22:04.004601 #24351]  INFO -- : libinput: 1.15.5
I, [2020-05-15T16:22:04.005128 #24351]  INFO -- : OS: Linux 5.6.11-arch1-1 #1 SMP PREEMPT Wed, 06 May 2020 17:32:37 +0000
I, [2020-05-15T16:22:04.005573 #24351]  INFO -- : Distribution: Arch Linux \r (\l)
I, [2020-05-15T16:22:04.007681 #24351]  INFO -- : Desktop session: /usr/share/xsessions/plasma x11
I, [2020-05-15T16:22:04.007702 #24351]  INFO -- : ---------------------------------------------
I, [2020-05-15T16:22:04.007711 #24351]  INFO -- : Enabled Plugins: 
I, [2020-05-15T16:22:04.007746 #24351]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-05-15T16:22:04.007756 #24351]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-05-15T16:22:04.007764 #24351]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2020-05-15T16:22:04.007771 #24351]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-05-15T16:22:04.007778 #24351]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-05-15T16:22:04.007785 #24351]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-05-15T16:22:04.007792 #24351]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-05-15T16:22:04.007799 #24351]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2020-05-15T16:22:04.007807 #24351]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-05-15T16:22:04.007814 #24351]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-05-15T16:22:04.007821 #24351]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-05-15T16:22:04.007828 #24351]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-05-15T16:22:04.007835 #24351]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-05-15T16:22:04.007842 #24351]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2020-05-15T16:22:04.007849 #24351]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-05-15T16:22:04.007869 #24351]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-05-15T16:22:04.007876 #24351]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-05-15T16:22:04.007883 #24351]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-05-15T16:22:04.007890 #24351]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-05-15T16:22:04.007897 #24351]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-05-15T16:22:04.007903 #24351]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2020-05-15T16:22:04.007911 #24351]  INFO -- : ---------------------------------------------

```
**Log (1)**
```
I, [2020-05-15T16:34:41.368091 #24782]  INFO -- : ---------------------------------------------
I, [2020-05-15T16:34:41.368123 #24782]  INFO -- : Fusuma: 1.10.0
I, [2020-05-15T16:34:41.368669 #24782]  INFO -- : libinput: 1.15.5
I, [2020-05-15T16:34:41.369142 #24782]  INFO -- : OS: Linux 5.6.11-arch1-1 #1 SMP PREEMPT Wed, 06 May 2020 17:32:37 +0000
I, [2020-05-15T16:34:41.369582 #24782]  INFO -- : Distribution: Arch Linux \r (\l)
I, [2020-05-15T16:34:41.371773 #24782]  INFO -- : Desktop session: /usr/share/xsessions/plasma x11
I, [2020-05-15T16:34:41.371794 #24782]  INFO -- : ---------------------------------------------
I, [2020-05-15T16:34:41.371803 #24782]  INFO -- : Enabled Plugins: 
I, [2020-05-15T16:34:41.371859 #24782]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-05-15T16:34:41.371885 #24782]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-05-15T16:34:41.371911 #24782]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2020-05-15T16:34:41.371919 #24782]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-05-15T16:34:41.371926 #24782]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-05-15T16:34:41.371933 #24782]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-05-15T16:34:41.371955 #24782]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-05-15T16:34:41.371979 #24782]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2020-05-15T16:34:41.371987 #24782]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-05-15T16:34:41.371994 #24782]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-05-15T16:34:41.372001 #24782]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-05-15T16:34:41.372007 #24782]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-05-15T16:34:41.372074 #24782]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-05-15T16:34:41.372082 #24782]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2020-05-15T16:34:41.372089 #24782]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-05-15T16:34:41.372096 #24782]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-05-15T16:34:41.372103 #24782]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-05-15T16:34:41.372110 #24782]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-05-15T16:34:41.372118 #24782]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-05-15T16:34:41.372125 #24782]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-05-15T16:34:41.372131 #24782]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
[... 92 lines cut ...]
        20: from /usr/bin/fusuma:23:in `load'
        19: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/exe/fusuma:41:in `<top (required)>'
        18: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma.rb:18:in `run'
        17: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma.rb:67:in `run'
        16: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:30:in `run'
        15: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma/libinput_command.rb:45:in `debug_events'
        14: from /usr/lib/ruby/2.7.0/open3.rb:101:in `popen3'
        13: from /usr/lib/ruby/2.7.0/open3.rb:219:in `popen_run'
        12: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma/libinput_command.rb:46:in `block in debug_events'
        11: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma/libinput_command.rb:46:in `loop'
        10: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma/libinput_command.rb:54:in `block (2 levels) in debug_events'
         9: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma/plugin/inputs/libinput_command_input.rb:31:in `block in run'
         8: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma.rb:72:in `block in run'
         7: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma.rb:93:in `detect'
         6: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma.rb:93:in `each_with_object'
         5: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma.rb:93:in `each'
         4: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.0/lib/fusuma.rb:94:in `block in detect'
         3: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-plugin-tap-0.3.0/lib/fusuma/plugin/detectors/tap_detector.rb:28:in `detect'
         2: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-plugin-tap-0.3.0/lib/fusuma/plugin/detectors/tap_detector.rb:61:in `hold?'
         1: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-plugin-tap-0.3.0/lib/fusuma/plugin/buffers/tap_buffer.rb:36:in `finger'
/usr/lib/ruby/gems/2.7.0/gems/fusuma-plugin-tap-0.3.0/lib/fusuma/plugin/buffers/tap_buffer.rb:36:in `max': comparison of Integer with String failed (ArgumentError)
```

**Additional context**
Using KDE to mimic MacOS config provided in the wiki on EndeavourOS running KDE plasma 5.18.5.

iberianpig, 2020-05-15:
Thanks for reporting.
It seems that fusuma-plugin-tap parse number of fingers as string.
I'll fix it.

iberianpig, 2020-05-15:
Please update fusuma and fusuma-plugin-tap.

```shell
$ sudo gem update fusuma fusuma-plugin-tap
```

Ru5h4d, 2020-05-15:
Updated. Thanks.

closed 2020-05-18 by Ru5h4d