iberianpig/fusuma #176: Can't use alternative .config.yml

LazyGeniusMan, 2020-05-21:
**Describe the bug**
I cant load alternative config.yml file. If I run `fusuma` it will load `/usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/config.yml` then if I run `fusuma --config=~/.config/fusuma/config.yml` it's still load `/usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/config.yml`. 

This default config.yml location will be a problem, right?. If `ruby` or `fusuma` is updated the folder path will change because it has version number in folder's name. A few days ago I use fusuma and it will automatically load `~/.config/fusuma/config.yml`

**To Reproduce**
1. Run `sudo gpasswd -a $USER input` and reboot.
2. Install `fusuma` and all plugins from AUR 
3. Run `fusuma`

**Expected behavior**
Fusuma load correct config.yml file

**Versions**
Fusuma: 1.10.1
libinput: 1.15.5
OS: Linux 5.6.12-1-MANJARO #1 SMP PREEMPT Sun May 10 14:36:43 UTC 2020
Distribution: Manjaro Linux 

**Additional context**
```
[lazygeniusman@Swift3-Manjaro ~]$ fusuma --config=~/.config/fusuma/config.yml
I, [2020-05-21T13:48:14.465831 #28428]  INFO -- : reload config : /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/config.yml
I, [2020-05-21T13:48:14.466226 #28428]  INFO -- : reload config : /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/config.yml
I, [2020-05-21T13:48:14.522230 #28428]  INFO -- : ---------------------------------------------
I, [2020-05-21T13:48:14.522266 #28428]  INFO -- : Fusuma: 1.10.1
I, [2020-05-21T13:48:14.522874 #28428]  INFO -- : libinput: 1.15.5
I, [2020-05-21T13:48:14.523433 #28428]  INFO -- : OS: Linux 5.6.12-1-MANJARO #1 SMP PREEMPT Sun May 10 14:36:43 UTC 2020
I, [2020-05-21T13:48:14.523962 #28428]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2020-05-21T13:48:14.526528 #28428]  INFO -- : Desktop session: /usr/share/xsessions/plasma x11
I, [2020-05-21T13:48:14.526553 #28428]  INFO -- : ---------------------------------------------
I, [2020-05-21T13:48:14.526564 #28428]  INFO -- : Enabled Plugins: 
I, [2020-05-21T13:48:14.526623 #28428]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-05-21T13:48:14.526636 #28428]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-05-21T13:48:14.526645 #28428]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2020-05-21T13:48:14.526655 #28428]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-05-21T13:48:14.526665 #28428]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-05-21T13:48:14.526673 #28428]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-05-21T13:48:14.526683 #28428]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-05-21T13:48:14.526692 #28428]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2020-05-21T13:48:14.526700 #28428]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-05-21T13:48:14.526711 #28428]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-05-21T13:48:14.526720 #28428]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-05-21T13:48:14.526728 #28428]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-05-21T13:48:14.526737 #28428]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-05-21T13:48:14.526745 #28428]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2020-05-21T13:48:14.526753 #28428]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2020-05-21T13:48:14.526766 #28428]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-05-21T13:48:14.526775 #28428]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-05-21T13:48:14.526783 #28428]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-05-21T13:48:14.526793 #28428]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-05-21T13:48:14.526802 #28428]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-05-21T13:48:14.526812 #28428]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-05-21T13:48:14.526826 #28428]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2020-05-21T13:48:14.526840 #28428]  INFO -- : ---------------------------------------------

```

iberianpig, 2020-05-21:
Could you paste outputs of `$ ls -l ~/.config/fusuma/config.yml` ?

LazyGeniusMan, 2020-05-21:
> Could you paste outputs of `$ ls -l ~/.config/fusuma/config.yml` ?

@iberianpig 
```
ls: cannot access '/home/lazygeniusman/.config/fusuma/config.yml': No such file or directory

```
hhmm weird, if i `cd` to `~/.config/fusuma/` and then `ls -l` the output is:
```
[lazygeniusman@Swift3-Manjaro ~]$ cd ~/.config/fusuma/
[lazygeniusman@Swift3-Manjaro fusuma]$ ls -l
total 4
-rwxrwxrwx 1 lazygeniusman lazygeniusman 905 May 17 14:59 config,yml
```

EDIT : Apparently it's just a typo in file naming haha, I mistype `,`, it should be `.`. I know this when you asked me to do `ls -l`. I did'nt realized it's mistype before, I thought it would be told there's no such a file when run `fusuma` if it's typo (not detected) or something. My bad, All good now.

closed 2020-05-21 by LazyGeniusMan

iberianpig, 2020-05-22:
Thank you for reporting.
I could know that every users can typo. This is good feedback for me.

I will modify the behavior to stop if config file does not exists.