iberianpig/fusuma #180: Right and left swipe with 3 fingers are not working in Ubuntu 20.04.

SasukeeUchiha, 2020-05-25:
I have attached my `config.yml` file here as a `.txt` as `.yml` is apparently not supported.
[config .yml.txt](https://github.com/iberianpig/fusuma/files/4677236/config.yml.txt)

I can't find any issues with this file. Also, the action is recognized when I run `$ fusuma` in the terminal as you can see from the output below. This issue is only with 3 finger swipe right and left.

me@mylaptop:~$ fusuma<br>
I, [2020-05-25T18:22:26.938651 #6145]  INFO -- : ---------------------------------------------<br>
I, [2020-05-25T18:22:26.938700 #6145]  INFO -- : Fusuma: 1.10.1<br>
I, [2020-05-25T18:22:26.940427 #6145]  INFO -- : libinput: 1.15.5<br>
I, [2020-05-25T18:22:26.941111 #6145]  INFO -- : OS: Linux 5.4.0-31-generic #35-Ubuntu SMP Thu May 7 20:20:34 UTC 2020<br>
I, [2020-05-25T18:22:26.941775 #6145]  INFO -- : Distribution: Ubuntu 20.04 LTS \n \l<br>
I, [2020-05-25T18:22:26.942394 #6145]  INFO -- : Desktop session: ubuntu x11<br>
I, [2020-05-25T18:22:26.942422 #6145]  INFO -- : ---------------------------------------------<br>
I, [2020-05-25T18:22:26.942442 #6145]  INFO -- : Enabled Plugins: <br>
I, [2020-05-25T18:22:26.942513 #6145]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer<br>
I, [2020-05-25T18:22:26.942529 #6145]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector<br>
I, [2020-05-25T18:22:26.942555 #6145]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector<br>
I, [2020-05-25T18:22:26.942574 #6145]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector<br>
I, [2020-05-25T18:22:26.942590 #6145]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord<br>
I, [2020-05-25T18:22:26.942600 #6145]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord<br>
I, [2020-05-25T18:22:26.942610 #6145]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord<br>
I, [2020-05-25T18:22:26.942619 #6145]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor<br>
I, [2020-05-25T18:22:26.942629 #6145]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter<br>
I, [2020-05-25T18:22:26.942638 #6145]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter<br>
I, [2020-05-25T18:22:26.942652 #6145]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput<br>
I, [2020-05-25T18:22:26.942661 #6145]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser<br>
I, [2020-05-25T18:22:26.942670 #6145]  INFO -- : ---------------------------------------------<br>
I, [2020-05-25T18:22:26.943318 #6145]  INFO -- : reload config : /home/vidath_dissanayake/.config/fusuma/config.yml<br>
I, [2020-05-25T18:22:28.849487 #6145]  INFO -- : {:command=>"xdotool key alt+tab"}<br>
I, [2020-05-25T18:22:30.237735 #6145]  INFO -- : {:command=>"xdotool key alt+shift+tab"}<br>

iberianpig, 2020-05-27:
Thank you for reporting.

I told you via email, did you confirm commands?
Please confirm that commands work correctly you assigned to fusuma in a terminal.

SasukeeUchiha, 2020-05-28:
Yes. The commands work.

iberianpig, 2020-05-28:
Ok.
Could you open your terminal?

Then, please paste `xdotool key alt+tab` to the terminal.(The command you assign to fusuma)
Run it with enter key.

I think you have a error message. Please fix it.

SasukeeUchiha, 2020-05-28:
`me@mylaptop:~$ xdotool key alt+tab`<br>
`(symbol) No such key name 'tab'. Ignoring it.`<br>
`(symbol) No such key name 'tab'. Ignoring it.`

Then what am I supposed to use instead of tab?

iberianpig, 2020-05-29:
I think this is difficult only fusuma now. see: https://github.com/iberianpig/fusuma/issues/94

You know, Switching application to reach 4th window requires the following process,
1. `xdotool keydown alt`
2. `xdotool key Tab`
3. `xdotool key Tab`
4. `xdotool key Tab`
5. `xdotool keyup alt`

**fusuma triggers keys combination as one shot command.** 
I think we should not assign a command requires complicated process or state.

It may be great that there is a simple one shot command that behaves that foreground application move to background in the same workspace.

SasukeeUchiha, 2020-05-29:
Ok. Then I will try that command. If I set the command to only `Tab` and hold `alt` and swipe will it work?

iberianpig, 2020-05-29:
Do you mean to send the Tab key in Fusuma while holding down the Alt key in another hand?
It's a simple and interesting idea. I think it's going to work.

SasukeeUchiha, 2020-05-29:
Yes. That is what I meant. And it does seem to work even though I have to do a separate swipe for each window. I have shift+Tab for swiping left. Now I put 3 finger tap to `xdotool keydown alt` and 4 finger tap to `xdotool keyup alt`. So now when I want to switch, I do a 3 finger tap, go to the one I want by swiping and do a 4 finger tap.

iberianpig, 2020-05-29:
> It may be great that there is a simple one shot command that behaves that foreground application move to background in the same workspace.

I wrote simple shellscript for you. please try this.

```
#!/bin/bash -x

desktop_id=$(wmctrl -d | grep \* | cut -d ' ' -f1)
cursor_file=/tmp/wmctrl.log

cursor=$(cat "$cursor_file" | head -n 1)

windows=$(wmctrl -l | grep "\ $desktop_id\ " | cut -d ' ' -f1)

if [ -z $cursor ]; then
  cursor="this is dummy"
fi

next_window=$(echo "$windows" | grep "$cursor" -A1 | tail -n 1)

if [ -z $next_window ]; then
  next_window=$(echo "$windows" | head -n 1)
fi

if [ "$cursor" = "$next_window" ]; then
  next_window=$(echo "$windows" | head -n 1)
fi

echo "$next_window" > "$cursor_file"

wmctrl -i -a  "$next_window"
```

Save as `cycle.sh` and  set `command: "sh /path/to/cycle.sh"` to config.yml

SasukeeUchiha, 2020-05-30:
This is my code now and I am satisfied with it. Thank you for your help.
[config .yml.txt](https://github.com/iberianpig/fusuma/files/4705236/config.yml.txt)

iberianpig, 2020-05-30:
I would like to port this window switching feature to [fusuma-plugin-tap](https://github.com/iberianpig/fusuma-plugin-tap/).

SasukeeUchiha, 2020-05-30:
Tap? Ok...

iberianpig, 2020-05-30:
LOL Sorry, I mistook it for `fusuma-plugin-wmctrl`.

This is the script of Window Switcher, an alternative to Alt-Tab, which I created above. I' ll rewrite this in Ruby and add it as a feature of `fusuma-plugin-wmctrl`.

SasukeeUchiha, 2020-05-30:
Yes, that is a good idea. Though you better say that it sends windows back instead of switching them. That is a very useful function. Also, when I want to directly switch to another window I use `alt+swipe with 3 fingers` thanks to `keypress`...

iberianpig, 2020-05-30:
See https://github.com/iberianpig/fusuma-plugin-wmctrl#window-property

The following parameters move to next/previous **workspaces** not switching window.
* `prev` is value to move the active window to the previous workspace.
* `next` is value to move the active window to the next workspace.

SasukeeUchiha, 2020-05-30:
Ah I see. Thanks for the clarification. Ping me when you add that feature to `wmctrl`. I will add the plugin then.

closed 2020-08-08 by stale[bot]

dakhrapov, 2021-04-01:
Hi all. I have a problem with fusuma installed on stock Ubuntu 20.04. I did all by instructions, but it still not working. May somebody can help me?