iberianpig/fusuma #181: Fusuma crashes while switching between workspaces on KDE Plasma

Ru5h4d, 2020-06-02:
**Describe the bug**
Fusuma crashes while switching between workspaces.

**To Reproduce**
Steps to reproduce the behavior:
1. 4 figure swipe left or right after running fusuma with the following in the config file
```
 LEFTMETA:
          command: 'xdotool key --clearmodifiers super+Tab'
```

**Expected behavior**
Switch to the next / previous workspace

**Versions**
```
I, [2020-06-02T11:19:26.346574 #3889]  INFO -- : ---------------------------------------------
I, [2020-06-02T11:19:26.346606 #3889]  INFO -- : Fusuma: 1.10.1
I, [2020-06-02T11:19:26.347098 #3889]  INFO -- : libinput: 1.15.5
I, [2020-06-02T11:19:26.347542 #3889]  INFO -- : OS: Linux 5.6.15-arch1-1 #1 SMP PREEMPT Wed, 27 May 2020 23:42:26 +0000
I, [2020-06-02T11:19:26.347980 #3889]  INFO -- : Distribution: Arch Linux \r (\l)
I, [2020-06-02T11:19:26.350145 #3889]  INFO -- : Desktop session: /usr/share/xsessions/plasma x11
I, [2020-06-02T11:19:26.350167 #3889]  INFO -- : ---------------------------------------------
I, [2020-06-02T11:19:26.350177 #3889]  INFO -- : Enabled Plugins: 
I, [2020-06-02T11:19:26.350216 #3889]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-06-02T11:19:26.350240 #3889]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-06-02T11:19:26.350249 #3889]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2020-06-02T11:19:26.350256 #3889]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-06-02T11:19:26.350263 #3889]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-06-02T11:19:26.350269 #3889]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-06-02T11:19:26.350275 #3889]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-06-02T11:19:26.350281 #3889]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2020-06-02T11:19:26.350287 #3889]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-06-02T11:19:26.350293 #3889]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-06-02T11:19:26.350300 #3889]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-06-02T11:19:26.350306 #3889]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-06-02T11:19:26.350312 #3889]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-06-02T11:19:26.350318 #3889]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2020-06-02T11:19:26.350324 #3889]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-06-02T11:19:26.350332 #3889]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-06-02T11:19:26.350344 #3889]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-06-02T11:19:26.350353 #3889]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-06-02T11:19:26.350359 #3889]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-06-02T11:19:26.350365 #3889]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-06-02T11:19:26.350371 #3889]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2020-06-02T11:19:26.350378 #3889]  INFO -- : ---------------------------------------------
```

**Log**
```
I, [2020-06-02T10:45:08.083714 #3057]  INFO -- : ---------------------------------------------
I, [2020-06-02T10:45:08.083748 #3057]  INFO -- : Fusuma: 1.10.1
I, [2020-06-02T10:45:08.084255 #3057]  INFO -- : libinput: 1.15.5
I, [2020-06-02T10:45:08.084721 #3057]  INFO -- : OS: Linux 5.6.15-arch1-1 #1 SMP PREEMPT Wed, 27 May 2020 23:42:26 +0000
I, [2020-06-02T10:45:08.085158 #3057]  INFO -- : Distribution: Arch Linux \r (\l)
I, [2020-06-02T10:45:08.087308 #3057]  INFO -- : Desktop session: /usr/share/xsessions/plasma x11
I, [2020-06-02T10:45:08.087329 #3057]  INFO -- : ---------------------------------------------
I, [2020-06-02T10:45:08.087339 #3057]  INFO -- : Enabled Plugins: 
I, [2020-06-02T10:45:08.087374 #3057]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-06-02T10:45:08.087385 #3057]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-06-02T10:45:08.087392 #3057]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2020-06-02T10:45:08.087399 #3057]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-06-02T10:45:08.087405 #3057]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-06-02T10:45:08.087411 #3057]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-06-02T10:45:08.087417 #3057]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-06-02T10:45:08.087423 #3057]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2020-06-02T10:45:08.087441 #3057]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-06-02T10:45:08.087450 #3057]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-06-02T10:45:08.087457 #3057]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-06-02T10:45:08.087464 #3057]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-06-02T10:45:08.087470 #3057]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-06-02T10:45:08.087476 #3057]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2020-06-02T10:45:08.087482 #3057]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-06-02T10:45:08.087488 #3057]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-06-02T10:45:08.087494 #3057]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-06-02T10:45:08.087500 #3057]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-06-02T10:45:08.087506 #3057]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-06-02T10:45:08.087513 #3057]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-06-02T10:45:08.087519 #3057]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2020-06-02T10:45:08.087525 #3057]  INFO -- : ---------------------------------------------
I, [2020-06-02T10:45:08.087935 #3057]  INFO -- : reload config : /home/ru5h4d/.config/fusuma/config.yml
Traceback (most recent call last):
        23: from /home/ru5h4d/.gem/ruby/2.7.0/bin/fusuma:23:in `<main>'
        22: from /home/ru5h4d/.gem/ruby/2.7.0/bin/fusuma:23:in `load'
        21: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/exe/fusuma:41:in `<top (required)>'
        20: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:18:in `run'
        19: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:67:in `run'
        18: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:30:in `run'
        17: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma/libinput_command.rb:45:in `debug_events'
        16: from /usr/lib/ruby/2.7.0/open3.rb:101:in `popen3'
        15: from /usr/lib/ruby/2.7.0/open3.rb:219:in `popen_run'
        14: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma/libinput_command.rb:46:in `block in debug_events'
        13: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma/libinput_command.rb:46:in `loop'
        12: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma/libinput_command.rb:54:in `block (2 levels) in debug_e
        11: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:31:in `block i
        10: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:74:in `block in run'
         9: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:114:in `execute'
         8: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:114:in `find'
         7: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:114:in `each'
         6: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:115:in `block in execute'
         5: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-plugin-wmctrl-0.3.0/lib/fusuma/plugin/executors/wmctrl_executor.rb:29:inble?'
         4: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-plugin-wmctrl-0.3.0/lib/fusuma/plugin/executors/wmctrl_executor.rb:36:incommand'
         3: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-plugin-wmctrl-0.3.0/lib/fusuma/plugin/executors/wmctrl_executor.rb:50:inworkspace_command'
         2: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-plugin-wmctrl-0.3.0/lib/fusuma/plugin/executors/wmctrl_executor.rb:90:inmmand'
         1: from /home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-plugin-wmctrl-0.3.0/lib/fusuma/plugin/executors/wmctrl_executor.rb:83:in_workspace_num'
/home/ru5h4d/.gem/ruby/2.7.0/gems/fusuma-plugin-wmctrl-0.3.0/lib/fusuma/plugin/executors/wmctrl_executor.rb:83:in ``': No such filctory - wmctrl (Errno::ENOENT)
```

**Additional context**
Using KDE to mimic MacOS config provided in the wiki with the changes mention in steps to reproduce, on EndeavourOS running KDE plasma 5.18.5.

iberianpig, 2020-06-02:
Hi, thanks for reporting.
You are using fusuma-plugin-wmctrl, but it seems to not install wmctrl.

 Please install wmctrl

```shell
$ sudo apt install wmctrl
```

Ru5h4d, 2020-06-03:
Installed wmctrl from AUR and the issue is resolved. Thanks.

closed 2020-06-03 by Ru5h4d