iberianpig/fusuma #182: undefined method `deep_symbolize_keys' for false:FalseClass (NoMethodError)

dipakmiglani5, 2020-06-04:
I get the following error while launching fusuma. Am I missing some dependency? I am a newbie so this might just be due to my ignorance. Any help is appreciated.

...
I, [2020-06-04T23:25:45.123292 #4935]  INFO -- : ---------------------------------------------
I, [2020-06-04T23:25:45.123325 #4935]  INFO -- : Fusuma: 1.10.1
I, [2020-06-04T23:25:45.124537 #4935]  INFO -- : libinput: 1.15.5
I, [2020-06-04T23:25:45.125018 #4935]  INFO -- : OS: Linux 5.4.0-33-generic #37-Ubuntu SMP Thu May 21 12:53:59 UTC 2020
I, [2020-06-04T23:25:45.125642 #4935]  INFO -- : Distribution: Ubuntu 20.04 LTS \n \l
I, [2020-06-04T23:25:45.126089 #4935]  INFO -- : Desktop session: ubuntu x11
I, [2020-06-04T23:25:45.126111 #4935]  INFO -- : ---------------------------------------------
I, [2020-06-04T23:25:45.126120 #4935]  INFO -- : Enabled Plugins: 
I, [2020-06-04T23:25:45.126156 #4935]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-06-04T23:25:45.126166 #4935]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-06-04T23:25:45.126173 #4935]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-06-04T23:25:45.126179 #4935]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-06-04T23:25:45.126186 #4935]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-06-04T23:25:45.126192 #4935]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-06-04T23:25:45.126198 #4935]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-06-04T23:25:45.126214 #4935]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-06-04T23:25:45.126221 #4935]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-06-04T23:25:45.126227 #4935]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-06-04T23:25:45.126234 #4935]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-06-04T23:25:45.126240 #4935]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2020-06-04T23:25:45.126246 #4935]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2020-06-04T23:25:45.126252 #4935]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-06-04T23:25:45.126259 #4935]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-06-04T23:25:45.126267 #4935]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-06-04T23:25:45.126274 #4935]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-06-04T23:25:45.126281 #4935]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-06-04T23:25:45.126287 #4935]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-06-04T23:25:45.126294 #4935]  INFO -- : ---------------------------------------------
Traceback (most recent call last):
	14: from /usr/local/bin/fusuma:23:in `<main>'
	13: from /usr/local/bin/fusuma:23:in `load'
	12: from /var/lib/gems/2.7.0/gems/fusuma-1.10.1/exe/fusuma:41:in `<top (required)>'
	11: from /var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:18:in `run'
	10: from /var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma.rb:67:in `run'
	 9: from /var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:25:in `run'
	 8: from /var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/plugin/inputs/libinput_command_input.rb:36:in `libinput_options'
	 7: from /var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/plugin/base.rb:32:in `config_params'
	 6: from /var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/config.rb:16:in `search'
	 5: from /usr/lib/ruby/2.7.0/singleton.rb:123:in `instance'
	 4: from /usr/lib/ruby/2.7.0/singleton.rb:123:in `synchronize'
	 3: from /usr/lib/ruby/2.7.0/singleton.rb:125:in `block in instance'
	 2: from /usr/lib/ruby/2.7.0/singleton.rb:125:in `new'
	 1: from /var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/config.rb:31:in `initialize'
/var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/config.rb:41:in `reload': undefined method `deep_symbolize_keys' for false:FalseClass (NoMethodError)
...

iberianpig, 2020-06-04:
Could you paste your config.yml?

dipakmiglani5, 2020-06-05:
I realised there were some typing issues within the config.yml. Correcting them solved the problem. Thankyou for your help even though it was not really an issue and just my ignorance.

closed 2020-06-05 by dipakmiglani5