iberianpig/fusuma #184: Pinch in gesture not recognized in Popos 20.04 (Ubuntu 20.04)

john-ghatas, 2020-06-15:
**Describe the bug**
Pinch in gesture is not recognized

**To Reproduce**
Steps to reproduce the behavior:
1. Run fusuma command in the terminal 
```bash
fusuma
```
2. Pinch out, it shows up in the debugger
3. Pinch in, it doesn't show up in the debugger

**Expected behavior**
Pinching out works as expected, pinching in does not

**Versions**
You can get this information from copy and paste the output of `fusuma --version` from the command line. Also, please include the OS and what version of the OS you're running.
[image]

**Additional context**
Any additional context, your `~/.config/fusuma/config.yml` or data that might be necessary to reproduce the issue.
The current setup looks like this
```yaml
swipe:
  3:
    left:
      command: "xdotool key alt+Right" # History forward 
    right:
      command: "xdotool key alt+Left" # History back
    up:
      command: "xdotool key ctrl+Super+Down" # Switch to next workspace
    down:
      command: "xdotool key ctrl+Super+Up" # Switch to previous workspace

  4:
    up:
      command: "xdotool key super" # Activity
    down:
      command: "xdotool key super" # Activity
    
pinch:
  2: 
    in:
      command: "xdotool keydown ctrl click 4 && xdotool keyup ctrl" # threshold: 0.5, interval: 0.5
  2: 
    out:
      command: "xdotool keydown ctrl click 5 keyup ctrl" # threshold: 0.5, interval: 0.5

threshold:
  pinch: 0.5

interval:
  swipe: 0.75
  pinch: 0.5
```

iberianpig, 2020-06-15:
Could you also check your log of `fusuma`?

You can find a line including `reload config: /path/to/your/config.yml.`

Fusuma use the config reloaded last.
```
$ iberianpig@iberianpig-XPS-13-9360:~$ fusuma
I, [2020-03-03T13:59:03.837509 #22169]  INFO -- : reload config : /home/iberianpig/.config/fusuma/config.yml   #← THIS IS CONFIG FILE USING FUSUMA
```

john-ghatas, 2020-06-15:
Yes, all other gestures are functioning fine as well :)

john-ghatas, 2020-06-16:
P.S. This is the case on both my XPS 7590 and Inspiron 5490

iberianpig, 2020-06-16:
I got it. You have a duplicate key. 

delete line `2:` last

```diff
swipe:
  3:
    left:
      command: "xdotool key alt+Right" # History forward 
    right:
      command: "xdotool key alt+Left" # History back
    up:
      command: "xdotool key ctrl+Super+Down" # Switch to next workspace
    down:
      command: "xdotool key ctrl+Super+Up" # Switch to previous workspace

  4:
    up:
      command: "xdotool key super" # Activity
    down:
      command: "xdotool key super" # Activity
    
pinch:
  2: 
    in:
      command: "xdotool keydown ctrl click 4 && xdotool keyup ctrl" # threshold: 0.5, interval: 0.5
-  2: 
    out:
      command: "xdotool keydown ctrl click 5 keyup ctrl" # threshold: 0.5, interval: 0.5

threshold:
  pinch: 0.5

interval:
  swipe: 0.75
  pinch: 0.5
```

john-ghatas, 2020-06-16:
Awesome, that worked. Thanks for the help :D, closing the issue now

closed 2020-06-16 by john-ghatas

john-ghatas, 2020-06-16:
I copied the config from the README, it showed the double key
[image]

iberianpig, 2020-06-16:
Thank you for reporting. I wrote invalid config.yml.
I will fix soon :+1:

commit f895c199 (2020-06-16): fix(README.md): fix wrong syntax

Fix #184