iberianpig/fusuma #185: pinch accuracy

gwanwoochoi, 2020-06-19:
I think the pinch gesture accuracy is very incorrect. 
When multiple attempts of pinch gesture are made, only one or two of them are executed. 
I'm using fusuma well and fusuma makes me feel more convenient, but also I felt uncomfortable because I used a lot of pinch options when browse the web.

amalija-ramljak, 2020-07-10:
I have this same issue. It just does not recognise every pinch very easily. I have the latest version of fusuma (installed 09 July). At first it did not recognise _any_ pinch, but after rerunning it a few times now it recognises some of them, but not reliably.

iberianpig, 2020-07-10:
Sorry for the delay to reply.

Try setting the `threshold` and `interval` in Fusuma's config.yml.

If you still think the recognition accuracy is poor, this is actually a problem with the `libinput` itself, on which Fusuma depends.

Even if I do a pinch gesture, sometimes it's recognized as a swipe or a scroll.

Libinput is still in its development stage, So I think the problem is not easy to solve only Fusuma.

OrionDB5, 2020-07-16:
I have this problem too, after trying different `threshold` and `interval` settings I found a temporary workaround: pinch with three fingers so that `libinput` can't recognize it as a scroll. I get 100% accuracy this way. 
While not being a solution, I think it's the best we can do until the fix is found in libinput's code.

closed 2020-09-27 by stale[bot]

digitalsignalperson, 2023-04-10:
I'm finding this issue too. It maybe feels a little better reducing threshold and interval to like 0.01, but the issues I have:
- often pinch is interpreted as scroll
- often it will scroll for a moment before switching to pinch
- often it will switch from pinch to scroll

I tried using 3 finger pinching as suggested, but my brain is struggling with how that works with one hand. 

It does look like the issue is from the libinput level though as far as the events fusuma uses. Maybe I will open an issue over there, as I don't see any relevant "pinch" issues on their gitlab. Is the synaptics driver any better?