iberianpig/fusuma #188: I installed fusuma by following every step as given but its not working. I am attaching the output after fusuma starts

anris03, 2020-06-26:
I, [2020-06-26T18:36:33.162872 #11009]  INFO -- : ---------------------------------------------
I, [2020-06-26T18:36:33.162974 #11009]  INFO -- : Fusuma: 1.10.1
I, [2020-06-26T18:36:33.166194 #11009]  INFO -- : libinput: 1.10.4
I, [2020-06-26T18:36:33.167822 #11009]  INFO -- : OS: Linux 5.3.0-61-generic #55~18.04.1-Ubuntu SMP Mon Jun 22 16:40:20 UTC 2020
I, [2020-06-26T18:36:33.169165 #11009]  INFO -- : Distribution: elementary OS Hera \n \l
I, [2020-06-26T18:36:33.170440 #11009]  INFO -- : Desktop session: pantheon x11
I, [2020-06-26T18:36:33.170514 #11009]  INFO -- : ---------------------------------------------
I, [2020-06-26T18:36:33.170555 #11009]  INFO -- : Enabled Plugins: 
I, [2020-06-26T18:36:33.170708 #11009]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-06-26T18:36:33.170750 #11009]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-06-26T18:36:33.170790 #11009]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-06-26T18:36:33.170828 #11009]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-06-26T18:36:33.170865 #11009]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-06-26T18:36:33.170901 #11009]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-06-26T18:36:33.170938 #11009]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-06-26T18:36:33.170974 #11009]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-06-26T18:36:33.171011 #11009]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-06-26T18:36:33.171126 #11009]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-06-26T18:36:33.171162 #11009]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-06-26T18:36:33.171198 #11009]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-06-26T18:36:33.171234 #11009]  INFO -- : ---------------------------------------------
I, [2020-06-26T18:36:33.172238 #11009]  INFO -- : reload config : /home/erikl/.config/fusuma/config.yml

iberianpig, 2020-06-26:
Have you restarted after adding input group (`$ sudo gpasswd -a $USER input`) ?

anris03, 2020-06-26:
> Have you restarted after adding input group (`$ sudo gpasswd -a $USER input`) ?

Yes i did restart

anris03, 2020-06-26:
> Have you restarted after adding input group (`$ sudo gpasswd -a $USER input`) ?

I am using elementary os 5.1
Only one 3 finger gesture which is working is when all windows are shown and I swipe to the right with 3 fingers then the current window is closed.
But this was working even before I installed fusuma.
Previously i installed libinput gesture, but they too didn't work, so I removed them. Then I installed fusuma

exhile, 2020-08-12:
I too, am unable to get fusuma to work on my machine even though I have followed all the steps.
```

exhile@XPS13:~$ fusuma
I, [2020-08-11T17:25:00.056858 #3100]  INFO -- : reload config: /home/exhile/.config/fusuma/config.yml
I, [2020-08-11T17:25:00.089033 #3100]  INFO -- : ---------------------------------------------
I, [2020-08-11T17:25:00.089070 #3100]  INFO -- : Fusuma: 1.11.1
I, [2020-08-11T17:25:00.090970 #3100]  INFO -- : libinput: 1.15.5
I, [2020-08-11T17:25:00.091551 #3100]  INFO -- : OS: Linux 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020
I, [2020-08-11T17:25:00.092170 #3100]  INFO -- : Distribution: Ubuntu 20.04.1 LTS \n \l
I, [2020-08-11T17:25:00.092685 #3100]  INFO -- : Desktop session: ubuntu x11
I, [2020-08-11T17:25:00.092708 #3100]  INFO -- : ---------------------------------------------
I, [2020-08-11T17:25:00.092717 #3100]  INFO -- : Enabled Plugins: 
I, [2020-08-11T17:25:00.092772 #3100]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-08-11T17:25:00.092782 #3100]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-08-11T17:25:00.092789 #3100]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-08-11T17:25:00.092795 #3100]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-08-11T17:25:00.092802 #3100]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-08-11T17:25:00.092809 #3100]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-08-11T17:25:00.092815 #3100]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-08-11T17:25:00.092822 #3100]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-08-11T17:25:00.092829 #3100]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-08-11T17:25:00.092836 #3100]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-08-11T17:25:00.092843 #3100]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-08-11T17:25:00.092850 #3100]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-08-11T17:25:00.092857 #3100]  INFO -- : ---------------------------------------------

```
None of the 3 finger gestures work for me.

exhile, 2020-08-12:
Actually, I got fusuma to work by typing the "fusuma" command in the CLI of terminal. I also re-configured the config file. Is there a way to automatically execute the fusuma command so I don't have to open terminal all the time?

iberianpig, 2020-08-12:
Please check the following section on README.
https://github.com/iberianpig/fusuma#autostart-gnome-session-properties

closed 2020-10-22 by stale[bot]