iberianpig/fusuma #189: Horizontal 4 finger movement not recognised

Nandete84, 2020-07-05:
The 4 fingers vertical function don't work on my Lenovo ideaPad Y700-15ISK with Ubuntu 20.04
Rest of movements (including 4 fingers horizontal) work as a charm

**To Reproduce**
Using the shortcuts from 'Settings', I modified the config file
`gedit ~/.config/fusuma/config.yml`

For the 4 fingers movement I have:

`
  4:
    left:
      command: "xdotool key ctrl+alt+Up" # Switch to previous workspace
    right:
      command: "xdotool key ctrl+alt+Down" # Switch to next workspace
    up:
      command: "xdotool key ctrl+Super+F" # Toggle full mode
    down:
      command: "xdotool key ctrl+Super+D" # Hide all normal windows
`
Movements left and right work perfectly.

For the up and down, no matter what configuration I write, they don't work.

I am using:
Fusuma 1.10.1
libinput 1.15.5
Ubuntu 20.04 LTS (Linux 5.4.0-40-generic #44 Ubuntu),
gnome shell version 5.0.16(1)

I have not added any additional configuration to the config file (threshold).

iberianpig, 2020-07-06:
@Nandete84 

1. Have you tested in xdotool command in your terminal?
1. Could you paste output `fusuma --version` or  `fusuma -v`?

BTW, the following line's indent level in config.yml is wrong.
```yml
  right:
  command: "xdotool key ctrl+alt+Down" # Switch to next workspace
```

Nandete84, 2020-07-06:
This is the `fusuma --version` log:

`
I, [2020-07-06T22:23:41.905816 #27739]  INFO -- : ---------------------------------------------
I, [2020-07-06T22:23:41.905876 #27739]  INFO -- : Fusuma: 1.10.1
I, [2020-07-06T22:23:41.907556 #27739]  INFO -- : libinput: 1.15.5
I, [2020-07-06T22:23:41.908179 #27739]  INFO -- : OS: Linux 5.4.0-40-generic #44-Ubuntu SMP Tue Jun 23 00:01:04 UTC 2020
I, [2020-07-06T22:23:41.908863 #27739]  INFO -- : Distribution: Ubuntu 20.04 LTS \n \l
I, [2020-07-06T22:23:41.909623 #27739]  INFO -- : Desktop session: ubuntu x11
I, [2020-07-06T22:23:41.909665 #27739]  INFO -- : ---------------------------------------------
I, [2020-07-06T22:23:41.909678 #27739]  INFO -- : Enabled Plugins: 
I, [2020-07-06T22:23:41.909749 #27739]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-07-06T22:23:41.909760 #27739]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-07-06T22:23:41.909788 #27739]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-07-06T22:23:41.909796 #27739]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-07-06T22:23:41.909820 #27739]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-07-06T22:23:41.909845 #27739]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-07-06T22:23:41.909853 #27739]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-07-06T22:23:41.909862 #27739]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-07-06T22:23:41.909877 #27739]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-07-06T22:23:41.909901 #27739]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-07-06T22:23:41.909909 #27739]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-07-06T22:23:41.909917 #27739]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-07-06T22:23:41.909926 #27739]  INFO -- : ---------------------------------------------
`

(The wrong indent was made during pasting it in here; the horizontal swipe works)

iberianpig, 2020-07-06:
OK,  thanks.

Please wrap code blocks using ``` triple backquotes for indenting. (I edited your post)

It's become like following.
```
code blocks
```

Please check following

> 1. Have you tested in xdotool command in your terminal?

```shell
$ xdotool key ctrl+Super+F
```

Have you checked the log in launching fusuma and the following lines?

```
I, [2020-07-06T18:59:58.239722 #1478]  INFO -- : reload config : /path/to/config.yml # ← Is config file path correctly?
```

If fusuma don't load your config, fusuma loads default config.yml.

Nandete84, 2020-07-07:
Thanks for that.

None of the commands work in xdotool.

I've double check the configuration file launching fusuma from terminal and it is reading the correct file; I configured all the movements and all they work but the 4 fingers up and down (and I can't make the rotation work either).

Is there any more info I can get?

iberianpig, 2020-07-08:
> None of the commands work in xdotool.

Do you mean that you couldn't execute `$ xdotool key ctrl+Super+F` in your terminal?

If you couldn't execute in a terminal, it is xdotool's issue, not fusuma.

Nandete84, 2020-07-08:
All the commands I have configured works through xdotool in a terminal.
But I've been trying with some of the ones in my 'Settings>Keyboard shortcuts' and there are a few that don't work with xdotool.

Specifically:

```xdotool key Ctrl+Super+D``` To minimise (toggle) all windows in one workspace, and 

```xdotool key Super+F``` (I removed the ```Ctrl``` it on 'Settings>Keyboard shortcuts', to try if it worked).

Is it then an xdotool issue?

Nandete84, 2020-07-11:
@iberianpig I finally worked it out.

You were right, of course, and it was an issue with xdotool.

Instead of simple shortcuts, this is the configuration I have now:

```
    up:
      command: "xdotool search --onlyvisible --name '.*' windowactivate %@" # Unhide all windows
      threshold: 0.25
    down:
      command: "xdotool key --clearmodifiers Super+d" # Show desktop
      threshold: 0.25
```

It works like a charm!!

Thanks for your support. And I love your nickname, I grew up in Southern West Spain, where the Iberian pigs are from :)

closed 2020-07-11 by Nandete84