iberianpig/fusuma #190: Non-working rotate and keypress plugin in Elementaty OS distribution

pawlobanano, 2020-07-05:
**Describe the bug**
I'd like to use option for rotation and keypress but it does not work.
I verified the commands for rotation (volume up/down and brightness up/down) pasted alone in the terminal. They are working fine. The rotation and keypress plugin seem to somehow be turned off - I don't see them in fusuma console output.
I think it is related to Elementary OS distribution. Maybe someone already knows the solution? I'd like to know if this distribution is still supported by fusuma.

**Versions**
Elementary OS 5.1.5 Hera
Built on Ubuntu 18.04.4 LTS
Linux 5.3.0-59-generic
GTK 3.22.30
```
fusuma --version
I, [2020-07-05T20:54:28.054207 #17721]  INFO -- : ---------------------------------------------
I, [2020-07-05T20:54:28.054261 #17721]  INFO -- : Fusuma: 1.10.1
I, [2020-07-05T20:54:28.055989 #17721]  INFO -- : libinput: 1.10.4
I, [2020-07-05T20:54:28.056734 #17721]  INFO -- : OS: Linux 5.3.0-59-generic #53~18.04.1-Ubuntu SMP Thu Jun 4 14:58:26 UTC 2020
I, [2020-07-05T20:54:28.057501 #17721]  INFO -- : Distribution: elementary OS Hera \n \l
I, [2020-07-05T20:54:28.058647 #17721]  INFO -- : Desktop session: pantheon x11
I, [2020-07-05T20:54:28.058676 #17721]  INFO -- : ---------------------------------------------
I, [2020-07-05T20:54:28.058691 #17721]  INFO -- : Enabled Plugins: 
I, [2020-07-05T20:54:28.058772 #17721]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-07-05T20:54:28.058792 #17721]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-07-05T20:54:28.058804 #17721]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2020-07-05T20:54:28.058814 #17721]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-07-05T20:54:28.058830 #17721]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-07-05T20:54:28.058850 #17721]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-07-05T20:54:28.058873 #17721]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-07-05T20:54:28.058896 #17721]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2020-07-05T20:54:28.058922 #17721]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-07-05T20:54:28.058944 #17721]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-07-05T20:54:28.058966 #17721]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-07-05T20:54:28.058989 #17721]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-07-05T20:54:28.059011 #17721]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-07-05T20:54:28.059037 #17721]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2020-07-05T20:54:28.059060 #17721]  INFO -- :   Fusuma::Plugin::Executors::WmctrlExecutor
I, [2020-07-05T20:54:28.059085 #17721]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-07-05T20:54:28.059109 #17721]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-07-05T20:54:28.059136 #17721]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-07-05T20:54:28.059159 #17721]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-07-05T20:54:28.059185 #17721]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-07-05T20:54:28.059206 #17721]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-07-05T20:54:28.059229 #17721]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2020-07-05T20:54:28.059251 #17721]  INFO -- : ---------------------------------------------
```

**Additional context**
My `~/.config/fusuma/config.yml` is an exact copy of https://github.com/iberianpig/fusuma/wiki/elementary-OS

iberianpig, 2020-07-06:
@pawlobanano 
I develop fusuma on elementary OS.
I checked now, but rotate and keypress are working correctly.

Did you check the log in launching fusuma and the following lines?

```
I, [2020-07-06T18:59:58.239722 #1478]  INFO -- : reload config : /path/to/config.yml # ← Is config file path correctly?
```

pawlobanano, 2020-07-07:
Bingo! I'm impressed Sir - you hit the problem with only one shot :)
`I, [2020-07-07T19:47:43.512839 #29320]  INFO -- : reload config : /var/lib/gems/2.5.0/gems/fusuma-1.10.1/lib/fusuma/config.yml
`
I added `-c <config_path>` option for starting command.

It works. Thank you @iberianpig . This project is great and a massive improvement of the way Linux users works on laptops.

I now see a quirk. When I have two-finger scrolling, two-finger pinch to zoom, and two-finger audio control they, unfortunately, they all interfere with each other. The cases I checked:

1. Two-finger scrolling, two-finger pinch to zoom, and two-finger audio control = total interference. 
Scrolling seems to have priority, then zoom, then audio control.
2. Two-finger pinch to zoom, and two-finger audio control = still evident interference. 
Zoom seems to have priority, then audio control.

Maybe it is a case of a config tweak or I'm just lacking a clear video tutorial on finger movement of a pro fusuma user :)

iberianpig, 2020-07-28:
Rotation and pinch-in/out gestures are easy to interfere with each other.

Try adjusting the intervals and thresholds.

Or try a less interfering setting, such as assigning rotation to three fingers only and pinch in/out to two fingers only.

pawlobanano, 2020-07-30:
Thank you @iberianpig  

You're right. Rotation with 3 fingers works flawlessly. So it is just a small interference with default double finger scroll set by default in the OS.

Anyway, my original issue is resolved. I'm closing the issue.

closed 2020-07-30 by pawlobanano