iberianpig/fusuma #191: How to uninstall fusuma

helios1321, 2020-07-08:
I don't know what I've done wrong but it only recognizes the touchpad if I do "sudo fusuma". If not it has not permission to the touchpad. 

Also after configurating the config.yaml file it doesn't update. I changed it both in my location and in this location that it keeps reloading it  /var/lib/gems/2.7.0/gems/fusuma-1.10.1/lib/fusuma/config.yml

After chaging this last one, when I do "sudo fusuma" the commands that it shows are correct but the actions are the previous ones. 

I think I shoud unistall it to start it from the bottom and see if I can make it work.

iberianpig, 2020-07-15:
@helios1321 Please try `$ sudo gem uninstall fusuma`.

codeniko, 2020-07-21:
1) did you assign the `input` group to your user account? This should likely help remove the need for `sudo`. See readme
2) It's `config.yml`, not `config.yaml`. Path would be `~/.config/fusuma/config.yml`. You can also pass a custom path to your config file with an option in the cli command like `fusuma -c '~/.config/fusuma/config.yml'`

mathpluscode, 2020-08-28:
@iberianpig Hi, thx for the great package, I suggest to indicate that `reboot` is required so that we can run `fusuma` without sudo ;)

closed 2020-12-19 by stale[bot]