iberianpig/fusuma #192: Gestures from libinput are not recognised

haydenrobins, 2020-07-21:
**Issue**
Sorry, this might be a feature request because it's not a very standard environment, I was not sure if this was supposed to be working or not. 
Fusuma does not seem to be recognising gestures from the libinput messages on my touchscreen device. 

When I run:
`fusuma -v` 
it shows that the touchscreen messages are received but the gestures to not invoke the keypresses.

I expect the three finger swipe down to open a terminal using the xdotool binding in the config file. 

** Environment**
ruby-fusuma 1.10.1-1
ruby-fusuma-plugin-sendkey 0.4.0-3
ruby-fusuma-plugin-tap 0.3.0-1

- Raspberry Pi 4b
- Arch Linux Arm 5.4.51-1-ARCH
- WaveShare 5.5" AMOLED HDMI display with USB touch input

** Configs **

This config tests sendkey plugin and xdotool as I am getting another error with `sendkey`
`fusuma-sendkey -l `
> sendkey: Keyboard /keyboard/ is not found

```
plugin:
  filters:
    libinput_device_filter:
      keep_device_names:
        - "WaveShare WaveShare"

swipe:
  3:
    up:
      sendkey: "SUPER+RETURN" # Open Terminal using sendkey
    down:
      command: 'xdotool key super+Return' # Open Terminal using xdotool

```

** Debug information **

`libinput list-devices`
>Device:           WaveShare WaveShare
Kernel:           /dev/input/event0
Group:            1
Seat:             seat0, default
Size:             540x320mm
Capabilities:     touch
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

The following is showing a 3 finger swipe down. 
```
fusuma -v
I, [2020-07-21T23:47:36.504331 #18501]  INFO -- : ---------------------------------------------
I, [2020-07-21T23:47:36.504481 #18501]  INFO -- : Fusuma: 1.10.1
I, [2020-07-21T23:47:36.509492 #18501]  INFO -- : libinput: 1.15.6
I, [2020-07-21T23:47:36.511145 #18501]  INFO -- : OS: Linux 5.4.51-1-ARCH #1 SMP PREEMPT Wed Jul 15 02:48:49 UTC 2020
[... 53 lines cut ...]
D, [2020-07-21T23:47:38.548946 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_FRAME      +0.136s       }
D, [2020-07-21T23:47:38.552341 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.144s       0 (0) 47.92/62.21 (259.00/199.17mm)}
D, [2020-07-21T23:47:38.554531 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.144s       1 (1) 51.90/47.58 (280.50/152.33mm)}
D, [2020-07-21T23:47:38.556809 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.144s       2 (2) 45.24/29.78 (244.50/95.33mm)}
D, [2020-07-21T23:47:38.558820 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_FRAME      +0.144s       }
D, [2020-07-21T23:47:38.561944 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.152s       0 (0) 44.22/62.05 (239.00/198.67mm)}
D, [2020-07-21T23:47:38.564258 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.152s       1 (1) 48.38/47.37 (261.50/151.67mm)}
D, [2020-07-21T23:47:38.565942 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.152s       2 (2) 42.00/29.78 (227.00/95.33mm)}
D, [2020-07-21T23:47:38.567943 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_FRAME      +0.152s       }
D, [2020-07-21T23:47:38.576212 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.168s       0 (0) 40.80/61.84 (220.50/198.00mm)}
D, [2020-07-21T23:47:38.577690 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.168s       1 (1) 45.14/47.42 (244.00/151.83mm)}
D, [2020-07-21T23:47:38.579130 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.168s       2 (2) 38.67/29.83 (209.00/95.50mm)}
D, [2020-07-21T23:47:38.580944 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_FRAME      +0.168s       }
D, [2020-07-21T23:47:38.584484 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.176s       0 (0) 38.30/61.63 (207.00/197.33mm)}
D, [2020-07-21T23:47:38.586171 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.176s       1 (1) 42.65/47.58 (230.50/152.33mm)}
D, [2020-07-21T23:47:38.587863 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.176s       2 (2) 35.52/29.98 (192.00/96.00mm)}
D, [2020-07-21T23:47:38.589190 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_FRAME      +0.176s       }
D, [2020-07-21T23:47:38.592447 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_UP         +0.184s       }
D, [2020-07-21T23:47:38.594470 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_UP         +0.184s       }
D, [2020-07-21T23:47:38.596563 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_MOTION     +0.184s       2 (2) 33.02/30.19 (178.50/96.67mm)}
D, [2020-07-21T23:47:38.598415 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_FRAME      +0.184s       }
D, [2020-07-21T23:47:38.608283 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_UP         +0.200s       }
D, [2020-07-21T23:47:38.610238 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record:  event0   TOUCH_FRAME      +0.200s       }
D, [2020-07-21T23:47:38.912734 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:38 +1000, tag: libinput_command_input, record: LIBINPUT TIMEOUT}
D, [2020-07-21T23:47:39.215242 #18501] DEBUG -- : {:input_event=>time: 2020-07-21 23:47:39 +1000, tag: libinput_command_input, record: LIBINPUT TIMEOUT}```

iberianpig, 2020-07-21:
Currently, touch screen is not supported.
However, it can be handled by libinput, so if you make a new parser plugin or something, it may work. Pull requests are welcome!

haydenrobins, 2020-07-22:
Understood. I will contribute if I can. Thank you.

closed 2020-07-22 by haydenrobins