iberianpig/fusuma #194: "ERROR -- : unknown keyword: filename" thrown after upgrading from 1.10.0 to 1.11.0

kats86, 2020-07-26:
hi,
first of all, thanks for the _great work_ so far :+1:. however, I stumpled upon the following error after upgrading to **fusuma_1.11.0**:
```
I, [2020-07-26T23:28:59.143073 #9193]  INFO -- : reload config: /home/kay/.config/fusuma/config.yml
E, [2020-07-26T23:28:59.143162 #9193] ERROR -- : unknown keyword: filename
Traceback (most recent call last):
	10: from /usr/local/bin/fusuma:23:in `<main>'
	 9: from /usr/local/bin/fusuma:23:in `load'
	 8: from /var/lib/gems/2.5.0/gems/fusuma-1.11.0/exe/fusuma:41:in `<top (required)>'
	 7: from /var/lib/gems/2.5.0/gems/fusuma-1.11.0/lib/fusuma.rb:17:in `run'
	 6: from /var/lib/gems/2.5.0/gems/fusuma-1.11.0/lib/fusuma.rb:32:in `read_options'
	 5: from /var/lib/gems/2.5.0/gems/fusuma-1.11.0/lib/fusuma.rb:51:in `load_custom_config'
	 4: from /var/lib/gems/2.5.0/gems/fusuma-1.11.0/lib/fusuma/config.rb:24:in `custom_path='
	 3: from /var/lib/gems/2.5.0/gems/fusuma-1.11.0/lib/fusuma/config.rb:39:in `custom_path='
	 2: from /var/lib/gems/2.5.0/gems/fusuma-1.11.0/lib/fusuma/config.rb:46:in `reload'
	 1: from /var/lib/gems/2.5.0/gems/fusuma-1.11.0/lib/fusuma/config.rb:52:in `validate'
/var/lib/gems/2.5.0/gems/fusuma-1.11.0/lib/fusuma/config.rb:67:in `rescue in validate': unknown keyword: filename (Fusuma::Config::InvalidFileError)
```
**fusuma_1.10.0** works fine (config.yml unchanged):
```
I, [2020-07-26T23:51:11.839560 #3350]  INFO -- : ---------------------------------------------
I, [2020-07-26T23:51:11.839608 #3350]  INFO -- : Fusuma: 1.10.1
I, [2020-07-26T23:51:11.841559 #3350]  INFO -- : libinput: 1.10.4
I, [2020-07-26T23:51:11.842358 #3350]  INFO -- : OS: Linux 5.4.0-42-generic #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020
I, [2020-07-26T23:51:11.843227 #3350]  INFO -- : Distribution: elementary OS Hera \n \l
I, [2020-07-26T23:51:11.844010 #3350]  INFO -- : Desktop session: pantheon x11
I, [2020-07-26T23:51:11.844040 #3350]  INFO -- : ---------------------------------------------
I, [2020-07-26T23:51:11.844056 #3350]  INFO -- : Enabled Plugins: 
I, [2020-07-26T23:51:11.844119 #3350]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-07-26T23:51:11.844139 #3350]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-07-26T23:51:11.844150 #3350]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-07-26T23:51:11.844161 #3350]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-07-26T23:51:11.844171 #3350]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-07-26T23:51:11.844187 #3350]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-07-26T23:51:11.844208 #3350]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-07-26T23:51:11.844221 #3350]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-07-26T23:51:11.844233 #3350]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-07-26T23:51:11.844244 #3350]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-07-26T23:51:11.844255 #3350]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-07-26T23:51:11.844279 #3350]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-07-26T23:51:11.844296 #3350]  INFO -- : ---------------------------------------------
I, [2020-07-26T23:51:11.844869 #3350]  INFO -- : reload config : /home/kay/.config/fusuma/config.yml
```
my **config.yml** (nothing special I suppose):
```
swipe:
  3:
    left:
      command: 'xdotool key ctrl+alt+Right'
    right:
      command: 'xdotool key ctrl+alt+Left'
    up:
      command: 'xdotool key super+s'
    #down:
    #  command: 'xdotool key super+s'
  4:
    left:
      command: 'xdotool key super+alt+Right'
    right:
      command: 'xdotool key super+alt+Left'
    #up:
      #command: 'xdotool key super+a'
    #down:
      #command: 'xdotool key super+s'

pinch:
  in:
    command: 'xdotool key ctrl+plus'
  out:
    command: 'xdotool key ctrl+minus'

threshold:
  swipe: 1
  pinch: 0.5

interval:
  swipe: 1
  pinch: 0.5
```
rolling back to fusuma_1.10.0 resolves the issue completely. _take care & stay save!_

TomasZsiva, 2020-07-26:
Hi!

I just tried to make the fusuma work today, already with the 1.11.0 version, and I'm getting the same error. I tried many configurations, tried to reinstall nothing worked. Can you please help me how can I 'roll back' even if I never had this program before? Is there a way to downgrade?

Thanks a lot

Tomi

kats86, 2020-07-26:
hey Tomi,
good question! I recovered the former version from a system backup I had available. However, some google research brought me there: https://rubygems.org/gems/fusuma/versions/1.10.1. It seems, that `$ gem install fusuma -v 1.10.1` might already do the trick.
Also, I just figured out I mentioned the wrong version in my initial posting (fixed now): the working version for me was actually 1.10.**1**, not 1.10.0 (never used that one, I also don't know about 1.10.2 which is available at rubygems.org). Hope, that helps!

kats86, 2020-07-26:
...I just played a little bit with `$ gem install fusuma -v 1.10.X` and it **does** the trick, also 1.10.2 works (in my case).

iberianpig, 2020-07-27:
Thank you for reporting :+1:  
I'm trying to reproduce.

iberianpig, 2020-07-27:
This issue occurred by`psych`package, YAML parser builtin Ruby.
I'll push fixed package.

closed 2020-07-27 by iberianpig

iberianpig, 2020-07-27:
Pushed fusuma 1.11.1 to rubygems.org

@kats86 @TomasZsiva Please update fusuma

TomasZsiva, 2020-07-27:
Everything works from now. Big up for you! Thanks a lot!

kats86, 2020-07-27:
same here, **1.11.1** works like a charm! *thank you!*

commit 024b072b (2020-07-27): Psych kwargs require Version 3.1.0 or later

* Psych v3.1.0 or older uses incompatible keyword arguments `filename:`
* Ruby v2.5.0(ubuntu 18.04) ships with old psych less than v3.1.0
* Use keyword arguments of YAML.parse if Psych v3.1.0 or later

Closes #194