iberianpig/fusuma #196: Unable to run fusuma even when Gem directory is in PATH

kmandarin-orange, 2020-07-28:
**Describe the bug**
After following all instructions to install, including adding `PATH="$PATH:$(ruby -e 'print Gem.user_dir')/bin"` to etc/bash.bashrc, then in ~/.bashrc (according to issue #96 ), then both, the fusuma executable won't run. It returns `bash: fusuma: command not found`. Seemingly, I've missed something with my PATH but I'm not sure what.

**To Reproduce**
Not sure exactly what is causing it, last time I used fusuma on this computer it was with an older version of Manjaro.
To reproduce just install and run fusuma on the latest stable Manjaro release.

**Expected behavior**
The fusuma gem to start

**Versions**
When running `fusuma --version`, `bash: fusuma: command not found` is returned because the executable isn't working. Reinstalling the gem says it is version 1.11.1. This is on a Manjaro KDE machine running the stable branch.

**Additional context**

As stated previously, I'm running Manjaro KDE on stable, with no .config for fusuma. Both ~/.bashrc and /etc/bash.bashrc have  been tested with both `PATH="$PATH:$(ruby -e 'print Gem.user_dir')/bin"` and `PATH="$PATH:/home/user/.gem/ruby/2.7.0/bin` at the bottom of the file. I made sure to restart every time I edited one of those files so changes would be applied.

- Laptop Specs (from inxi)
CPU: Dual Core Intel Celeron N4000 (-MCP-) speed/min/max: 796/800/2600 MHz 
Kernel: 5.6.19-2-MANJARO x86_64 Up: 16h 40m Mem: 2020.9/3760.7 MiB (53.7%) 
Storage: 88.07 GiB (38.4% used) Procs: 182 Shell: bash 5.0.18 inxi: 3.0.37

iberianpig, 2020-07-28:
Could you retry to install Ruby?
https://wiki.archlinux.org/index.php/Ruby

kmandarin-orange, 2020-07-28:
After reinstalling Ruby and the fusuma gem, I realized I had been using sudo to install the gem the whole time, which was affecting my gem path, so I tried installing without sudo and it worked! I guess these things always end up being easy mistakes haha, thanks for your help and time. I absolutely love this project!

closed 2020-07-28 by kmandarin-orange