iberianpig/fusuma #197: After Update it Crahses

pheinze, 2020-07-28:
After update fusuma with pamac (and gem) i can not use fusuma anymore. It crahses after starting.

the CLI gives that output.

********************
I, [2020-07-28T13:20:47.970593 #2721]  INFO -- : reload config: /home/user/.config/fusuma/config.yml
E, [2020-07-28T13:20:47.972009 #2721] ERROR -- : /home/user/.config/fusuma/config.yml: 4 is duplicated
E, [2020-07-28T13:20:47.972082 #2721] ERROR -- : /home/user/.config/fusuma/config.yml: pinch is duplicated
E, [2020-07-28T13:20:47.972177 #2721] ERROR -- : Detect duplicate keys ["4", "pinch"]
Traceback (most recent call last):
	9: from /usr/bin/fusuma:23:in `<main>'
	8: from /usr/bin/fusuma:23:in `load'
	7: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/exe/fusuma:41:in `<top (required)>'
	6: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma.rb:17:in `run'
	5: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma.rb:32:in `read_options'
	4: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma.rb:51:in `load_custom_config'
	3: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:24:in `custom_path='
	2: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:39:in `custom_path='
	1: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:46:in `reload'
/usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:58:in `validate': Detect duplicate keys ["4", "pinch"] (Fusuma::Config::InvalidFileError)
	10: from /usr/bin/fusuma:23:in `<main>'
	 9: from /usr/bin/fusuma:23:in `load'
	 8: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/exe/fusuma:41:in `<top (required)>'
	 7: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma.rb:17:in `run'
	 6: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma.rb:32:in `read_options'
	 5: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma.rb:51:in `load_custom_config'
	 4: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:24:in `custom_path='
	 3: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:39:in `custom_path='
	 2: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:46:in `reload'
	 1: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:52:in `validate'
/usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.rb:67:in `rescue in validate': Detect duplicate keys ["4", "pinch"] (Fusuma::Config::InvalidFileError)
************************************

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Versions**
You can get this information from copy and paste the output of `fusuma --version` from the command line. Also, please include the OS and what version of the OS you're running.

**Additional context**
Any additional context, your `~/.config/fusuma/config.yml` or data that might be necessary to reproduce the issue.

iberianpig, 2020-07-28:
```
E, [2020-07-28T13:20:47.972177 #2721] ERROR -- : Detect duplicate keys ["4", "pinch"]
```

Please paste your ~/.config/fusuma/config.yml

Do you have duplicate keys in your ~/.config/fusuma/config.yml?

pheinze, 2020-07-28:
Thanks for replying.
Here is my config:

https://pastebin.com/MXRwxyq5

I just copied the config.yml somewhere.
Would be great to have some modern useful config templates.
btw. my cursor changed sometime into across-hair and have strange behavior.
Probably it is issued with duplicate keys, too.

THX

pheinze, 2020-07-28:
I already have seen the mistake.

iberianpig, 2020-07-28:
* You have some duplicate keys in YAML
    * first two sections (swipe, pinch section) are duplicated keys
    * Therefore, I modified duplicated sections to comment
* Second swipe section's indent level is invalid

The following setting is valid syntax that I fixed your invalid config.yml.

```yaml
# lines beginning from `#` are comments 
# swipe:
#   3:
#     left:
#       command: 'xdotool key alt+Right'
#     right:
#       command: 'xdotool key alt+Left'
#     up:
#       command: 'xdotool key ctrl+t'
#     down:
#       command: 'xdotool key ctrl+w'
#   4:
#     left:
#       command: 'xdotool key super+Right'
#     right:
#       command: 'xdotool key super+Left'
#     up:
#       command: 'xdotool key super+a'
#     down:
#       command: 'xdotool key super+s'
# 
# pinch:
#   2:
#     in:
#       command: 'xdotool key ctrl+equal'
#     out:
#       command: 'xdotool key ctrl+minus'
#   4:
#     in:
#       command: 'xdotool key super+a'
#     out:
#       command: 'xdotool key super+s'
# 

swipe:
  3:
    left:
      command: 'xdotool key alt+Right' # History forward
      keypress:
        LEFTCTRL:
          command: 'xdotool key ctrl+Tab' # Next tab
    right:
      command: 'xdotool key alt+Left' # History back
      keypress:
        LEFTCTRL:
          command: 'xdotool key ctrl+shift+Tab' # Prev tab
    up:
      command: 'xdotool key ctrl+t' # Open new tab
      keypress:
        LEFTSHIFT:
          command: 'xdotool key --clearmodifiers ctrl+shift+t' # Open last closed tab
    down:
      command: 'xdotool key ctrl+w' # Close tab
  4:
    left:
      workspace: 'next' # Switch to next workspace
      keypress:
        LEFTSHIFT:
          window: 'next' # Move window to next workspace
        LEFTMETA:
          command: 'xdotool key --clearmodifiers super+ctrl+Left' # Move window to left side
    right:
      workspace: 'prev' # Switch to previous workspace
      keypress:
        LEFTSHIFT:
          window: 'prev' # Move window to previous workspace
        LEFTMETA:
          command: 'xdotool key --clearmodifiers super+ctrl+Right'  # Move window to right side
    up:
      command: 'xdotool key super+s' # Workspace overview
      keypress:
        LEFTMETA:
          window:
            maximized: 'toggle' # Toggle Maximize/Unmaximize Window
    down:
      command: 'xdotool key super+a' # Window overview
      keypress:
        LEFTMETA:
          window: 'close' # Close window

pinch:
  2:
    in:
      command: 'xdotool keydown ctrl click 4 keyup ctrl' # Zoom in
    out:
      command: 'xdotool keydown ctrl click 5 keyup ctrl' # Zoom out
  4:
    in:
      command: 'xdotool key super+a' # Window overview
    out:
      command: 'xdotool key super+s' # Workspace overview

rotate:
  2:
    clockwise:
      command: 'xdotool key XF86AudioRaiseVolume' # Volume up
    counterclockwise:
      command: 'xdotool key XF86AudioLowerVolume' # Volume down
  3:
    clockwise:
      command: 'xdotool key XF86MonBrightnessUp' # Brightness up
    counterclockwise:
      command: 'xdotool key XF86MonBrightnessDown' # Brightness down

threshold:
  swipe: 0.3
  pinch: 0.1

interval:
  swipe: 1
  pinch: 1

plugin:
  inputs:
    libinput_command_input: # options for lib/plugin/inputs/libinput_command_input
      enable-tap: true # click to tap
      enable-dwt: true # disable tap while typing
      show-keycodes: true # https://github.com/iberianpig/fusuma-plugin-keypress#add-show-keycode-option
```

closed 2020-07-28 by iberianpig