iberianpig/fusuma #201: fusuma without sudo fails, although user is included in input group. It makes impossible configuring it on startup

jesfi, 2020-08-20:
**Describe the bug**
fusuma does not work without sudo, User is included in group input, but it fails.

Traceback (most recent call last):
	2: from /usr/local/bin/fusuma:23:in `<main>'
	1: from /usr/local/lib/site_ruby/2.5.0/rubygems.rb:296:in `activate_bin_path'
/usr/local/lib/site_ruby/2.5.0/rubygems.rb:277:in `find_spec_for_exe': can't find gem fusuma (>= 0.a) with executable fusuma (Gem::GemNotFoundException)

it works running sudo, but i need to add it to startup applications so sudo won't be able.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow Readme installation.

**Expected behavior**
Run fusuma normaly before adding user to input group and rebooting, but it does not work. It only works with sudo.

**Versions**
OS: Ubuntu Mate 1.20.1
ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]O

sudo fusuma --version
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
I, [2020-08-20T17:40:26.887305 #23769]  INFO -- : reload config: /home/fco-escribano-fierro/.config/fusuma/config.yml
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
I, [2020-08-20T17:40:26.915192 #23769]  INFO -- : ---------------------------------------------
I, [2020-08-20T17:40:26.915250 #23769]  INFO -- : Fusuma: 1.11.1
I, [2020-08-20T17:40:26.917338 #23769]  INFO -- : libinput: 1.10.4
I, [2020-08-20T17:40:26.918330 #23769]  INFO -- : OS: Linux 5.7.0-050700rc6-generic #202005172030 SMP Mon May 18 00:33:00 UTC 2020
I, [2020-08-20T17:40:26.919302 #23769]  INFO -- : Distribution: ALCS 4.54.1-X23 LTS
I, [2020-08-20T17:40:26.920190 #23769]  INFO -- : Desktop session:
I, [2020-08-20T17:40:26.920235 #23769]  INFO -- : ---------------------------------------------
I, [2020-08-20T17:40:26.920266 #23769]  INFO -- : Enabled Plugins: 
I, [2020-08-20T17:40:26.920380 #23769]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-08-20T17:40:26.920445 #23769]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-08-20T17:40:26.920460 #23769]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-08-20T17:40:26.920473 #23769]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-08-20T17:40:26.920492 #23769]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-08-20T17:40:26.920508 #23769]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-08-20T17:40:26.920523 #23769]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-08-20T17:40:26.920536 #23769]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-08-20T17:40:26.920551 #23769]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-08-20T17:40:26.920577 #23769]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-08-20T17:40:26.920607 #23769]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-08-20T17:40:26.920623 #23769]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-08-20T17:40:26.920661 #23769]  INFO -- : ---------------------------------------------

**Additional context**
cat ~/.config/fusuma/config.yml
swipe:
  3: 
    left: 
      command: 'xdotool key alt+Right'
    right: 
      command: 'xdotool key alt+Left'
    up: 
      command: 'xdotool key ctrl+w'
    down: 
      command: 'xdotool key super'
  4:
    left: 
      command: 'xdotool key ctrl+alt+Down'
    right: 
      command: 'xdotool key ctrl+alt+Up'
    up: 
      command: 'xdotool key ctrl+alt+Down'
    down: 
      command: 'xdotool key ctrl+alt+Up'
pinch:
  in:
    command: 'xdotool key ctrl+plus'
  out:
     command: 'xdotool key ctrl+minus'

threshold:
  swipe: 0.4
  pinch: 0.4

interval:
  swipe: 0.8
  pinch: 0.1

iberianpig, 2020-08-20:
How do you install ruby?

Please paste outputs of following commands,

```
$ sudo which fusuma
```

```
$ which fusuma
```

jesfi, 2020-08-21:
ruby installation:
sudo apt install libinput-tools
sudo apt install ruby

$ sudo which fusuma
/usr/local/bin/fusuma

$ which fusuma
/usr/local/bin/fusuma

iberianpig, 2020-08-26:
> Traceback (most recent call last):
2: from /usr/local/bin/fusuma:23:in <main>' 1: from /usr/local/lib/site_ruby/2.5.0/rubygems.rb:296:in activate_bin_path'
/usr/local/lib/site_ruby/2.5.0/rubygems.rb:277:in `find_spec_for_exe': can't find gem fusuma (>= 0.a) with executable fusuma (Gem::GemNotFoundException)

I think you installed local ruby and system-wide ruby.

Please try to uninstall fusuma from local ruby and system-wide ruby.

* `gem uninstall fusuma` (uninstall from local)
* `sudo gem uninstall fusuma` (uninstall from system-wide)
* `sudo gem install fusuma` (install to system-wide)

jesfi, 2020-08-30:
I followed your steps but I got the same result.

~$ gem uninstall fusuma
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
Gem 'fusuma' is not installed
~$ sudo gem uninstall fusuma
[sudo] password for fco-escribano-fierro: 
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated

Select gem to uninstall:
 1. fusuma-1.10.1
 2. fusuma-1.11.1
 3. All versions
> 3
Successfully uninstalled fusuma-1.10.1
Remove executables:
	fusuma

in addition to the gem? [Yn]  y
Removing fusuma
Successfully uninstalled fusuma-1.11.1
$ sudo gem uninstall fusuma
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
Gem 'fusuma' is not installed
$ sudo gem install fusuma
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:29: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
[... cut ...]
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:30: warning: constant Gem::ConfigMap is deprecated
/usr/lib/ruby/vendor_ruby/rubygems/defaults/operating_system.rb:10: warning: constant Gem::ConfigMap is deprecated
Traceback (most recent call last):
	2: from /usr/local/bin/fusuma:23:in `<main>'
	1: from /usr/local/lib/site_ruby/2.5.0/rubygems.rb:296:in `activate_bin_path'
/usr/local/lib/site_ruby/2.5.0/rubygems.rb:277:in `find_spec_for_exe': can't find gem fusuma (>= 0.a) with executable fusuma (Gem::GemNotFoundException)

closed 2020-12-19 by stale[bot]

HritwikSinghal, 2021-03-20:
hey i am also facing this same issue, but on manjaro

here is my error 
```
    ~  fusuma                                                                                                                                         ✔  3s  
I, [2021-03-21T04:23:54.665735 #11757]  INFO -- : reload config: /home/hritwik/.config/fusuma/config.yml
Traceback (most recent call last):
	33: from /usr/bin/fusuma:23:in `<main>'
	32: from /usr/bin/fusuma:23:in `load'
	31: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/exe/fusuma:41:in `<top (required)>'
	30: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma.rb:17:in `run'
	29: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma.rb:34:in `read_options'
	28: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:68:in `require_base_plugins'
	27: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:68:in `require_relative'
	26: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/executors/executor.rb:5:in `<top (required)>'
	25: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/executors/executor.rb:6:in `<module:Fusuma>'
	24: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/executors/executor.rb:8:in `<module:Plugin>'
	23: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/executors/executor.rb:13:in `<module:Executors>'
	22: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/base.rb:13:in `inherited'
	21: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:56:in `add'
	20: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:16:in `require_siblings_from_plugin_dir'
	19: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:16:in `each'
	18: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:17:in `block in require_siblings_from_plugin_dir'
	17: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:85:in `require'
	16: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:85:in `require'
	15: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/executors/command_executor.rb:5:in `<top (required)>'
	14: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/executors/command_executor.rb:6:in `<module:Fusuma>'
	13: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/executors/command_executor.rb:7:in `<module:Plugin>'
	12: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/executors/command_executor.rb:9:in `<module:Executors>'
	11: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/base.rb:13:in `inherited'
	10: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:57:in `add'
	 9: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:23:in `require_siblings_from_gems'
	 8: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:23:in `each'
	 7: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/plugin/manager.rb:25:in `block in require_siblings_from_gems'
	 6: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:85:in `require'
	 5: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:85:in `require'
	 4: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-plugin-sendkey-0.5.1/lib/fusuma/plugin/executors/sendkey_executor.rb:3:in `<top (required)>'
	 3: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-plugin-sendkey-0.5.1/lib/fusuma/plugin/executors/sendkey_executor.rb:3:in `require_relative'
	 2: from /usr/lib/ruby/gems/2.7.0/gems/fusuma-plugin-sendkey-0.5.1/lib/fusuma/plugin/sendkey/keyboard.rb:3:in `<top (required)>'
	 1: from /usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:85:in `require'
/usr/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:85:in `require': cannot load such file -- revdev (LoadError)
```