iberianpig/fusuma #203: Gesture is being passed on to applicationA clear and concise description of what the bug is.

paperbenni, 2020-09-01:
**Describe the bug**
When executing any three finger gesture it still gets passed on to the application. 
Example: If you execute a three finger gesture while hovering over the tab bar in Firefox you will either open or close a tab because it still gets registered as a middle click

**To Reproduce**
Steps to reproduce the behavior:
1. Create a three finger gesture
2. Open an application that recognizes middle clicks
3. Execute the gesture

**Expected behavior**
The gesture should be triggered and the input should not be passed on to the application

**Versions**
```
W, [2020-09-01T10:51:02.419431 #7763]  WARN -- : config file: /home/benjamin/.config/fusuma/config.yml is NOT FOUND
I, [2020-09-01T10:51:02.419488 #7763]  INFO -- : reload config: /usr/lib/ruby/gems/2.7.0/gems/fusuma-1.11.1/lib/fusuma/config.yml
I, [2020-09-01T10:51:02.457533 #7763]  INFO -- : ---------------------------------------------
I, [2020-09-01T10:51:02.457575 #7763]  INFO -- : Fusuma: 1.11.1
I, [2020-09-01T10:51:02.458564 #7763]  INFO -- : libinput: 1.15.6
I, [2020-09-01T10:51:02.459319 #7763]  INFO -- : OS: Linux 5.7.15-1-MANJARO #1 SMP PREEMPT Tue Aug 11 15:00:37 UTC 2020
I, [2020-09-01T10:51:02.460244 #7763]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2020-09-01T10:51:02.465837 #7763]  INFO -- : Desktop session: default x11
I, [2020-09-01T10:51:02.465867 #7763]  INFO -- : ---------------------------------------------
I, [2020-09-01T10:51:02.465883 #7763]  INFO -- : Enabled Plugins:
I, [2020-09-01T10:51:02.465949 #7763]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-09-01T10:51:02.465963 #7763]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-09-01T10:51:02.465976 #7763]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-09-01T10:51:02.465994 #7763]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-09-01T10:51:02.466009 #7763]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-09-01T10:51:02.466025 #7763]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-09-01T10:51:02.466037 #7763]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-09-01T10:51:02.466048 #7763]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-09-01T10:51:02.466060 #7763]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-09-01T10:51:02.466073 #7763]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-09-01T10:51:02.466087 #7763]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-09-01T10:51:02.466105 #7763]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-09-01T10:51:02.466119 #7763]  INFO -- : ---------------------------------------------

```
OS: instantOS (the WM in use it a dwm fork with an identical input system so you should be able to treat this bug report as dwm related)

iberianpig, 2020-09-02:
Fusuma don't restrict tap to click(left/middle/right).
You should off middle click manually.

paperbenni, 2020-09-03:
Completely disabling middle click prevents me from using it inside applications and that's a big problem for most creative programs. 
It's also heavily used in browsers and some gtk apps like nautilus. 
The instances of a user wanting to execute a gesture and paste something in a text field at the same time are pretty rare. 
If faced with the choice between gestures and working middle click I think most would choose middle click. 
If it's not possible I understand that, but imho it is it is a must have for users that use any GUI applications at all.

iberianpig, 2020-09-03:
Fusuma is a simple system that only monitors events and issues commands. Therefore, you cannot eliminate the built-in gestures that are under the control by OS.
 **Note that existing clicks and pastes are handled by the OS.**

 1. Hack to disable paste by 3-finger tap. 

see: https://unix.stackexchange.com/questions/24330/how-can-i-turn-off-middle-mouse-button-paste-functionality-in-all-programs

edit `~/.xbindkeysrc`

```
"echo -n | xsel -n -i; pkill xbindkeys; xdotool click 2; xbindkeys"
b:2 + Release
```

2. Use fusuma-plugin-tap to set up Ctrl-V pasting for 3-finger taps.
see: https://github.com/iberianpig/fusuma-plugin-tap

```yaml
tap:
  3:
    sendkey: "LEFTCTRL+V"

plugin:
  inputs:
    libinput_command_input: # options for lib/plugin/inputs/libinput_command_input
      enable-tap: true
      enable-dwt: true
      show-keycodes: true
      verbose: true

```

closed 2020-12-19 by stale[bot]