iberianpig/fusuma #204: Touchpad not found?

MrInfusion, 2020-09-28:
I have an XMG Fusion 15. I installed everything like it was shown in the Installation Section. But unfortunately after I start fusuma in the Terminal the the output said that the Touchpad wasn't found.

iberianpig, 2020-09-28:
try `sudo gpasswd -a $USER input` and shutdown and restart.
This will add the user to the input group. Fusuma should be able to detect your touchpad after a restart.

shreyash, 2021-03-31:
Try using `sudo fusuma` in the terminal. Your touchpad doesn't get detected without root privileges for some reason.

iberianpig, 2021-03-31:
In general, the reason why touchpad input can be read by `sudo` is that normal users do not have access permission to `/dev/input/eventX`.
If you do `$ sudo gpasswd -a $USER input` before running fusuma, you don't need `sudo` for run fusuma and can access `/dev/input/eventX`. (Please see README)