iberianpig/fusuma #205: continuous gestures

Da-Viper, 2020-09-30:
**Is your feature request related to a problem? Please describe.**
using the gestures to increase or decrease volume / backlight. the volume is reduced/increased at  a rate of 5. 
what i want is for it to increase continously as i swipe up 

**Describe the solution you'd like**
the current command i am running is with a keyboard modifier key SHIFT and a 3 swipe up gesture 
if i hold the shift and continously swipe up the volume increases. 

**Describe alternatives you've considered**
i have considered setting the intervall/ threshold to 0.1 / 0 but because the gesture is recognised as one gesture is recognised as one gesture, it will only perform it once .

**Additional context**
if the gestures has something like a timer to say how long the gesture should be

iberianpig, 2020-09-30:
Are you using fusuma-plugin-keypress?
Could you paste your config.yml?

Da-Viper, 2020-10-01:
@iberianpig  here it is 
```
swipe:
  3:
    left:
      command: 'xdotool key alt+Right' # History forward
    right:
      command: 'xdotool key alt+Left' # History back
    up:
      command: 'qdbus org.kde.kglobalaccel /component/kwin org.kde.kglobalaccel.Component.invokeShortcut "ShowDesktopGrid"' # Open DesktopGrid
      keypress:
        LEFTSHIFT:
          command: 'xdotool key --clearmodifiers ctrl+shift+t' # Open last closed tab
        LEFTCTRL:
          command: 'qdbus org.kde.kglobalaccel /component/kmix invokeShortcut "increase_volume"' #Volume increase 
        threshold: 0.0
        interval: 0.0
    down:
      command: 'xdotool key ctrl+w' # Close tab
      keypress:
        LEFTCTRL:
          command: 'qdbus org.kde.kglobalaccel /component/kmix invokeShortcut "decrease_volume"'
        threshold: 0.0
        interval: 0.0
  4:
    left:
      workspace: 'next' # Switch to next workspace
      keypress:
        LEFTSHIFT: 
          command: 'qdbus org.kde.kglobalaccel /component/kwin org.kde.kglobalaccel.Component.invokeShortcut "Window to Previous Desktop"' # Move window to next workspace
        LEFTMETA:
          command: 'xdotool key --clearmodifiers super+ctrl+Left' # Move window to left side
    right:
      workspace: 'prev' # Switch to previous workspace
      keypress:
        LEFTSHIFT: 
          command: 'qdbus org.kde.kglobalaccel /component/kwin org.kde.kglobalaccel.Component.invokeShortcut "Window to Next Desktop"' # Move window to previous workspace
    up:
      command: 'qdbus org.kde.kglobalaccel /component/kwin org.kde.kglobalaccel.Component.invokeShortcut "ShowDesktopGrid"' # Workspace overview
      keypress:
        LEFTMETA:
          window:
            maximized: 'toggle' # Toggle Maximize/Unmaximize Window
    down:
      window: 'close' # Close window
      keypress:
        LEFTCTRL:
          command: 'qdbus org.kde.kglobalaccel /component/kwin org.kde.kglobalaccel.Component.invokeShortcut "ExposeClass"' #aaaaa

pinch:
  2:
    in:
      command: 'xdotool keydown ctrl click 4 keyup ctrl' # Zoom in
    out:
      command: 'xdotool keydown ctrl click 5 keyup ctrl' # Zoom out
  4:
    in:
      command: 'xdotool key super+a' # Window overview
    out:
      command: 'xdotool key super+s' # Workspace overview

plugin: 
  inputs:
    libinput_command_input: # options for lib/plugin/inputs/libinput_command_input
      enable-tap: true # click to tap
      enable-dwt: true # disable tap while typing
      show-keycodes: true # https://github.com/iberianpig/fusuma-plugin-keypress#add-show-keycode-option

```

HritwikSinghal, 2020-10-16:
it would be great to increase volume or backlight increase as we swipe up in one go and not in intervals. If i recall libinput only supports this type of behavior in wayland and not on Xorg. Still iberianpig if you can implement this, it would be very much helpful.

Here is my config on Ubuntu 20.04.1 Wayland.

'''
swipe:
  3:
    left:
      # sendkey: "LEFTALT+RIGHT" # history back
      keypress:
        LEFTCTRL:
          sendkey: "PREVIOUSSONG"
        # LEFTMETA+LEFTALT:
          # sendkey: 'BRIGHTNESSDOWN'

    right:
      # sendkey: "LEFTALT+LEFT" # history forward
      keypress:
        LEFTCTRL:
          sendkey: "NEXTSONG"
        # LEFTMETA+LEFTALT:
        #   sendkey: 'xdotool key --clearmodifiers XF86AudioLowerVolume'

    up:
      sendkey: "LEFTMETA" # Overview
      keypress:
        LEFTCTRL:
          sendkey: "LEFTCTRL+K" # open new tab
        LEFTCTRL+LEFTALT:
          sendkey: 'BRIGHTNESSUP'

    down:
      sendkey: "LEFTMETA" # Overview
      keypress:
        LEFTCTRL:
          sendkey: "LEFTCTRL+COMMA" # close tab
        LEFTCTRL+LEFTALT:
          sendkey: 'BRIGHTNESSDOWN'

  4:
    left:
      sendkey: "VOLUMEDOWN"
    right:
      sendkey: "VOLUMEUP"
#     up:
#       sendkey: "xdotool key ctrl+alt+Down" # Switch to next workspace
#     down:
#       sendkey: "xdotool key ctrl+alt+Up" # Switch to previous workspace
pinch:
  in:
    sendkey: "ZOOMOUT" # Zoom in
  out:
    sendkey: "ZOOMIN" # Zoom out

threshold:
  pinch: 0.4
  swipe: 0.2

interval:
  swipe: 0.8
  pinch: 0.5
  tap: 0.2

tap:
  # 1:
  #   sendkey: "echo ----------------tap1----------------------------"
  # 2:
  #   sendkey: "echo ----------------tap2----------------------------"
  # 3:
  #   sendkey: "echo ----------------tap3----------------------------"
  # 4:
    # sendkey: "PLAYPAUSE"

hold:
#   1:
#     command: "echo ----------------hold1----------------------------"
#   2:
#     command: "echo ----------------hold2----------------------------"
#   3:
#     command: "echo ----------------hold3----------------------------"
  4:
    sendkey: "PLAYPAUSE"

plugin: 
  inputs:
    libinput_command_input:
      enable-tap: true
      verbose: true
      show-keycodes: true
'''

iberianpig, 2021-04-12:
@Da-Viper @HritwikSinghal 

I've implemented continuous gestures. https://github.com/iberianpig/fusuma/pull/224

I'm using the following settings on my laptop.

* 3 fingers rotate
    * => Sound Volume Up/Down
* pressing LEFT ALT key + 3 fingers rotate
    * => Brightness Up/Down

```yaml
rotate:
  3:
    clockwise:
      update:
        sendkey: VOLUMEUP
        LEFTALT:
          command: gdbus call --session --dest org.gnome.SettingsDaemon.Power --object-path /org/gnome/SettingsDaemon/Power --method org.gnome.SettingsDaemon.Power.Screen.StepUp
    counterclockwise:
      update:
        sendkey: VOLUMEDOWN
        LEFTALT:
          command: gdbus call --session --dest org.gnome.SettingsDaemon.Power --object-path /org/gnome/SettingsDaemon/Power --method org.gnome.SettingsDaemon.Power.Screen.StepDown
```

HritwikSinghal, 2021-04-13:
> @Da-Viper @HritwikSinghal
>
> I've implemented continuous gestures. #224
>
> I'm using the following settings on my laptop.
>
>     * 3 fingers rotate
>
>       * => Sound Volume Up/Down
>
>     * pressing [...]

I tried continuous gestures for volume and they work flawlessly. Thanks for this update!

HritwikSinghal, 2021-04-13:
> @Da-Viper @HritwikSinghal
> 
> I've implemented continuous gestures. #224

I tried continuous gestures for volume and they work flawlessly. Thanks for this update!

Edit: How can i have continuous gesture only when holding special key (like left ctrl) and not for normal gesture?

Also for brightness, i am using sendkey and it is not working. the command ```fusuma-sendkey BRIGHTNESSDOWN``` works on terminal ( but first prints out ```fatal: not a git repository (or any of the parent directories): .git``` and then lowers brightness) but ```sendkey: "BRIGHTNESSUP"``` does nothing (it logs that ```[2021-04-13T09:57:55.125914 #1690119]  INFO -- : {:sendkey=>"BRIGHTNESSDOWN"}``` but does nothing.

iberianpig, 2021-04-13:
> `fatal: not a git repository (or any of the parent directories): .git`

I released fixed versions. This message occurs when loading each plugin's gemspec. 

> it logs that [2021-04-13T09:57:55.125914 #1690119] INFO -- : {:sendkey=>"BRIGHTNESSDOWN"} but does nothing.

I think that it does not work.
Press the `BRIGHTNESSDOWN` and `LEFTALT` keys on your keyboard at the same time. It should not work.
This is because fusuma-plugin-keypress and fusuma-plugin-sendkey do not prevent key events currently sending from real devices.

ahmed2m, 2021-04-20:
I've a concerns with this approach that I hope there is a way around for them.
If I have my virtual desktops switch on swipe left<->right with 3 fingers and volume on swipe up<->down with 3 fingers like this:
```yml
swipe:
  3:
    left:
      update:
          command: 'xdotool key super+Left'
    right:
      update:
        command: 'xdotool key super+Right'

    up:
      update:
        command: 'xdotool key XF86AudioRaiseVolume'
    down:
      update:
        command: 'xdotool key XF86AudioLowerVolume'

```

This setup implicitly triggers both when I start my gesture as volume control (y-axis) but my fingers go ever slightly to the right I trigger the x-axis combination.

Also I thought about using the new `begin, end` but I couldn't figure how to only apply them on one section but not the other?
Like with this the up, down don't function (because of the alt press with `begin`)

```yml
swipe:
  3:
    begin:
      command: xdotool keydown super
    right:
      update:
        command: xdotool key Right
        interval: 5
    left:
      update:
        command: xdotool key Left
        interval: 5
    end:
      command: xdotool keyup super+ctrl

    up:
      update:
        command: 'xdotool key XF86AudioRaiseVolume'
    down:
      update:
        command: 'xdotool key XF86AudioLowerVolume'
```

iberianpig, 2021-04-22:
Hi @ahmed2m 

> virtual desktops switch on swipe left<->right with 3 fingers and volume on swipe up<->down with 3 fingers like this:

I don't think continuous gestures that are triggered frequently are suitable for switching virtual desktops. Please use normal one-shot gestures.

```yml
swipe:
  3:
    right:
      command: 'xdotool key super+Right'
    left:
      command: 'xdotool key super+Left'
    up:
      update:
        command: 'xdotool key XF86AudioRaiseVolume'
    down:
      update:
        command: 'xdotool key XF86AudioLowerVolume'
```

iberianpig, 2021-04-22:
> Also I thought about using the new "begin"/"end" but I couldn't figure how to only apply them on one section but not the other?
> Like with this the up, down don't function (because of the alt press with begin

As shown in the example below, `begin` / `end` can also be used under the direction (`right`/`left`/`up`/`down`).

```yml
swipe:
  3:
    right:
      command: 'xdotool key Right'
      begin:
        command: 'xdotool keydown super'
    left:
      command: 'xdotool key super+Left'
      begin:
        command: 'xdotool keydown super'
    up:
      update:
        command: 'xdotool key XF86AudioRaiseVolume'
    down:
      update:
        command: 'xdotool key XF86AudioLowerVolume'
    end:
      command: 'xdotool keyup super'
```

Note that keys used with `begin` will be kept pressed.
To ensure that keys are released at `end`, it is a good idea to set keyup to the parent property (set to whole).

closed 2021-06-26 by iberianpig