iberianpig/fusuma #206: Magic Trackpad 2 via Bluetooth does not work

defnotrobbie, 2020-09-30:
**Describe the bug**
Magic Trackpad 2 does not work when connected to Ubuntu 20.04 via Bluetooth. When connected via USB, the trackpad works great, gestures work, etc.

**To Reproduce**
Steps to reproduce the behavior:
1. setup Fusuma with Magic Trackpad 2 via USB
2. verify everything works as expected
3. disconnect Magic Trackpad 2
4. connect Magic Trackpad 2 via Bluetooth
5. verify only clicks register

**Expected behavior**
Magic Trackpad 2 works the same when connected via USB and Bluetooth.

**Versions**
```
INFO -- : Fusuma: 1.11.1
INFO -- : libinput: 1.15.5
INFO -- : OS: Linux 5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020
INFO -- : Distribution: Ubuntu 20.04.1 LTS
INFO -- : Desktop session: ubuntu x11
```

**Additional context**

`libinput debug-events` selected output (trackpad is recognized, but only clicks register).
```
-event22  DEVICE_ADDED     Apple Inc. Magic Trackpad 2       seat0 default group8  cap:pg  size 162x115mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger
 event22  POINTER_BUTTON   +0.871s	BTN_LEFT (272) pressed, seat count: 1
 event22  POINTER_BUTTON   +0.901s	BTN_LEFT (272) released, seat count: 0
```

iberianpig, 2020-09-30:
If libinput-debug-events outputs swipe logs, fusuma can detect swipes.
I think it's a problem on the libinput side.

defnotrobbie, 2020-10-12:
I tried messing with all the different libinput quirks files, building the old pre-4.18 magic trackpad 2 driver and trying to use it, all to no avail. I ended up reinstalling Ubuntu fresh and only every connecting the trackpad via bluetooth and it works. Not a great solution but I never found the source of the problem so ¯\\_(ツ)_/¯.

closed 2020-10-12 by defnotrobbie