iberianpig/fusuma #207: Gestures not working but sendkey works

charansai612, 2020-10-09:
[charansai@skipper ~]$ fusuma-sendkey LEFTCTRL+F9     
[charansai@skipper ~]$ fusuma
I, [2020-10-09T15:50:19.560152 #10071]  INFO -- : reload config: /home/charansai/.config/fusuma/config.yml
I, [2020-10-09T15:50:19.601352 #10071]  INFO -- : ---------------------------------------------
I, [2020-10-09T15:50:19.601401 #10071]  INFO -- : Fusuma: 1.11.1
I, [2020-10-09T15:50:19.602064 #10071]  INFO -- : libinput: 1.16.1
I, [2020-10-09T15:50:19.602678 #10071]  INFO -- : OS: Linux 5.8.11-1-MANJARO #1 SMP PREEMPT Wed Sep 23 14:35:40 UTC 2020
I, [2020-10-09T15:50:19.603284 #10071]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2020-10-09T15:50:19.604588 #10071]  INFO -- : Desktop session: /usr/share/xsessions/plasma x11
I, [2020-10-09T15:50:19.604621 #10071]  INFO -- : ---------------------------------------------
I, [2020-10-09T15:50:19.604635 #10071]  INFO -- : Enabled Plugins: 
I, [2020-10-09T15:50:19.604692 #10071]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-10-09T15:50:19.604748 #10071]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-10-09T15:50:19.604770 #10071]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-10-09T15:50:19.604787 #10071]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-10-09T15:50:19.604804 #10071]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-10-09T15:50:19.604821 #10071]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-10-09T15:50:19.604838 #10071]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-10-09T15:50:19.604854 #10071]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-10-09T15:50:19.604871 #10071]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2020-10-09T15:50:19.604888 #10071]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-10-09T15:50:19.604905 #10071]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-10-09T15:50:19.604922 #10071]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-10-09T15:50:19.604938 #10071]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-10-09T15:50:19.604955 #10071]  INFO -- : ---------------------------------------------
I, [2020-10-09T15:50:35.776187 #10071]  INFO -- : {:sendkey=>"LEFTCTRL+F9"}

Typing fusuma-sendkey LEFTCTRL+F9 works. But when I try using the track pad gesture the sendkey is getting registered in terminal but I can't see any change. I waited for few minutes but can't see any change.

Here is sample config.yml
swipe:
  3:
    up:
      sendkey: "LEFTCTRL+F9"

plugin:
  executors:
    sendkey_executor:
      device_name: 'Acer WMI hotkeys'

iberianpig, 2020-10-09:
Could you paste output of `fusuma-sendkey -l`?

charansai612, 2020-10-10:
> Could you paste output of `fusuma-sendkey -l`?

RESERVED
ESC
1
2
3
4
5
6
7
8
9
0
MINUS
EQUAL
BACKSPACE
TAB
Q
W
E
R
T
Y
U
I
O
P
LEFTBRACE
RIGHTBRACE
ENTER
LEFTCTRL
A
S
D
F
G
H
J
K
L
SEMICOLON
APOSTROPHE
GRAVE
LEFTSHIFT
BACKSLASH
Z
X
C
V
B
N
M
COMMA
DOT
SLASH
RIGHTSHIFT
KPASTERISK
LEFTALT
SPACE
CAPSLOCK
F1
F2
F3
F4
F5
F6
F7
F8
F9
F10
NUMLOCK
SCROLLLOCK
KP7
KP8
KP9
KPMINUS
KP4
KP5
KP6
KPPLUS
KP1
KP2
KP3
KP0
KPDOT
ZENKAKUHANKAKU
102ND
F11
F12
RO
KATAKANA
HIRAGANA
HENKAN
KATAKANAHIRAGANA
MUHENKAN
KPJPCOMMA
KPENTER
RIGHTCTRL
KPSLASH
SYSRQ
RIGHTALT
LINEFEED
HOME
UP
PAGEUP
LEFT
RIGHT
END
DOWN
PAGEDOWN
INSERT
DELETE
MACRO
MUTE
VOLUMEDOWN
VOLUMEUP
POWER
KPEQUAL
KPPLUSMINUS
PAUSE
SCALE
KPCOMMA
HANGEUL
HANGUEL
HANJA
YEN
LEFTMETA
RIGHTMETA
COMPOSE
STOP
AGAIN
PROPS
UNDO
FRONT
COPY
OPEN
PASTE
FIND
CUT
HELP
MENU
CALC
SETUP
SLEEP
WAKEUP
FILE
SENDFILE
DELETEFILE
XFER
PROG1
PROG2
WWW
MSDOS
COFFEE
SCREENLOCK
DIRECTION
CYCLEWINDOWS
MAIL
BOOKMARKS
COMPUTER
BACK
FORWARD
CLOSECD
EJECTCD
EJECTCLOSECD
NEXTSONG
PLAYPAUSE
PREVIOUSSONG
STOPCD
RECORD
REWIND
PHONE
ISO
CONFIG
HOMEPAGE
REFRESH
EXIT
MOVE
EDIT
SCROLLUP
SCROLLDOWN
KPLEFTPAREN
KPRIGHTPAREN
NEW
REDO
F13
F14
F15
F16
F17
F18
F19
F20
F21
F22
F23
F24
PLAYCD
PAUSECD
PROG3
PROG4
DASHBOARD
SUSPEND
CLOSE
PLAY
FASTFORWARD
BASSBOOST
PRINT
HP
CAMERA
SOUND
QUESTION
EMAIL
CHAT
SEARCH
CONNECT
FINANCE
SPORT
SHOP
ALTERASE
CANCEL
BRIGHTNESSDOWN
BRIGHTNESSUP
MEDIA
SWITCHVIDEOMODE
KBDILLUMTOGGLE
KBDILLUMDOWN
KBDILLUMUP
SEND
REPLY
FORWARDMAIL
SAVE
DOCUMENTS
BATTERY
BLUETOOTH
WLAN
UWB
UNKNOWN
VIDEO_NEXT
VIDEO_PREV
BRIGHTNESS_CYCLE
BRIGHTNESS_ZERO
DISPLAY_OFF
WIMAX
RFKILL
OK
SELECT
GOTO
CLEAR
POWER2
OPTION
INFO
TIME
VENDOR
ARCHIVE
PROGRAM
CHANNEL
FAVORITES
EPG
PVR
MHP
LANGUAGE
TITLE
SUBTITLE
ANGLE
ZOOM
MODE
KEYBOARD
SCREEN
PC
TV
TV2
VCR
VCR2
SAT
SAT2
CD
TAPE
RADIO
TUNER
PLAYER
TEXT
DVD
AUX
MP3
AUDIO
VIDEO
DIRECTORY
LIST
MEMO
CALENDAR
RED
GREEN
YELLOW
BLUE
CHANNELUP
CHANNELDOWN
FIRST
LAST
AB
NEXT
RESTART
SLOW
SHUFFLE
BREAK
PREVIOUS
DIGITS
TEEN
TWEN
VIDEOPHONE
GAMES
ZOOMIN
ZOOMOUT
ZOOMRESET
WORDPROCESSOR
EDITOR
SPREADSHEET
GRAPHICSEDITOR
PRESENTATION
DATABASE
NEWS
VOICEMAIL
ADDRESSBOOK
MESSENGER
DISPLAYTOGGLE
SPELLCHECK
LOGOFF
DOLLAR
EURO
FRAMEBACK
FRAMEFORWARD
CONTEXT_MENU
MEDIA_REPEAT
10CHANNELSUP
10CHANNELSDOWN
IMAGES
DEL_EOL
DEL_EOS
INS_LINE
DEL_LINE
FN
FN_ESC
FN_F1
FN_F2
FN_F3
FN_F4
FN_F5
FN_F6
FN_F7
FN_F8
FN_F9
FN_F10
FN_F11
FN_F12
FN_1
FN_2
FN_D
FN_E
FN_F
FN_S
FN_B
BRL_DOT1
BRL_DOT2
BRL_DOT3
BRL_DOT4
BRL_DOT5
BRL_DOT6
BRL_DOT7
BRL_DOT8
BRL_DOT9
BRL_DOT10
NUMERIC_0
NUMERIC_1
NUMERIC_2
NUMERIC_3
NUMERIC_4
NUMERIC_5
NUMERIC_6
NUMERIC_7
NUMERIC_8
NUMERIC_9
NUMERIC_STAR
NUMERIC_POUND
CAMERA_FOCUS
WPS_BUTTON
TOUCHPAD_TOGGLE
TOUCHPAD_ON
TOUCHPAD_OFF
CAMERA_ZOOMIN
CAMERA_ZOOMOUT
CAMERA_UP
CAMERA_DOWN
CAMERA_LEFT
CAMERA_RIGHT
MIN_INTERESTING
MAX
CNT

iberianpig, 2020-10-10:
Did you reboot after adding $USER into input group?

And if it's ok, did other action works,  like `command: xdotool`?

charansai612, 2020-10-10:
> Did you reboot after adding $USER into input group?
> 
> And if it's ok, did other action works, like `command: xdotool`?

Rebooted, and tried libinput gestures only and those are working fine.I tried libinput gestures after fusuma

iberianpig, 2020-10-10:
Thank you.

Could you paste output of `$ cat /home/charansai/.config/fusuma/config.yml` ?

I'll check and try your config.yml.

charansai612, 2020-10-10:
> Thank you.
> 
> Could you paste output of `$ cat /home/charansai/.config/fusuma/config.yml` ?
> 
> I'll check and try your config.yml.

swipe:
  3:
    up:
      sendkey: "LEFTCTRL+F9"

plugin:
  executors:
    sendkey_executor:
      device_name: 'Acer WMI hotkeys'

iberianpig, 2020-10-10:
Does it work with the following config?
```
swipe:
  3:
    up:
      command: "fusuma-sendkey LEFTCTRL+F9"

plugin:
  executors:
    sendkey_executor:
      device_name: 'Acer WMI hotkeys'
```

charansai612, 2020-10-10:
> Does it work with the following config?
> 
> ```
> swipe:
>   3:
>     up:
>       command: "fusuma-sendkey LEFTCTRL+F9"
> 
> plugin:
>   executors:
>     sendkey_executor:
>       device_name: 'Acer WMI hotkeys'
> ```

Works Thank you

closed 2020-10-10 by charansai612