iberianpig/fusuma #208: Fusuma wont work after resuming from suspend on wayland

HritwikSinghal, 2020-10-16:
**Describe the bug**
fusuma wont recognize gestures after waking up from suspend state on ubuntu 20.04.1 Wayland. It works fine on Xorg.

**To Reproduce**
Steps to reproduce the behavior:
1.Install fusuma (as mentioned in guide) with its plugins (sendkey and tap), and i configure my yml (given below).
2. add 'fusuma -d' or 'path/to/fusuma -d' in startup apps and all is working correctly. It is working across restarts and has expected behavior on gestures.
3. On wayland, click on suspend and after waking up from suspend, the gestures wont work. Fusuma service is running ("ps -ef | grep fusuma" shows fusuma running). If i start it again ("fusuma -d"),  it starts and works fine. 
Now note also that after starting the fusuma after waking up from suspend (i.e after step 3) it persists across suspends, meaning i have to start fusuma only after first suspend and it will function correctly after every suspend from now until i reboot.

**Expected  behavior**
fusuma should work after suspend without running it again

**Versions**
You can get this information from copy and paste the output of `fusuma --version` from the command line. Also, please include the OS and what version of the OS you're running.

```
I, [2020-10-16T21:09:56.541010 #24990]  INFO -- : reload config: /home/hritwik/.config/fusuma/config.yml
I, [2020-10-16T21:09:56.593344 #24990]  INFO -- : ---------------------------------------------
I, [2020-10-16T21:09:56.593377 #24990]  INFO -- : Fusuma: 1.11.1
I, [2020-10-16T21:09:56.594721 #24990]  INFO -- : libinput: 1.15.5
I, [2020-10-16T21:09:56.595227 #24990]  INFO -- : OS: Linux 5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020
I, [2020-10-16T21:09:56.595685 #24990]  INFO -- : Distribution: Ubuntu 20.04.1 LTS \n \l
I, [2020-10-16T21:09:56.596137 #24990]  INFO -- : Desktop session: ubuntu-wayland wayland
I, [2020-10-16T21:09:56.596153 #24990]  INFO -- : ---------------------------------------------
I, [2020-10-16T21:09:56.596162 #24990]  INFO -- : Enabled Plugins: 
I, [2020-10-16T21:09:56.596231 #24990]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-10-16T21:09:56.596243 #24990]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-10-16T21:09:56.596251 #24990]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2020-10-16T21:09:56.596258 #24990]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-10-16T21:09:56.596265 #24990]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-10-16T21:09:56.596272 #24990]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-10-16T21:09:56.596279 #24990]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-10-16T21:09:56.596287 #24990]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2020-10-16T21:09:56.596295 #24990]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-10-16T21:09:56.596304 #24990]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-10-16T21:09:56.596311 #24990]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-10-16T21:09:56.596318 #24990]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-10-16T21:09:56.596325 #24990]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-10-16T21:09:56.596333 #24990]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2020-10-16T21:09:56.596343 #24990]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-10-16T21:09:56.596351 #24990]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-10-16T21:09:56.596359 #24990]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-10-16T21:09:56.596367 #24990]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-10-16T21:09:56.596375 #24990]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-10-16T21:09:56.596385 #24990]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-10-16T21:09:56.596394 #24990]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2020-10-16T21:09:56.596402 #24990]  INFO -- : ---------------------------------------------

```

Ubuntu 20.04.1 Walyand.

**Additional context**
Any additional context, your `~/.config/fusuma/config.yml` or data that might be necessary to reproduce the issue.

```
swipe:
  3:
    left:
      # sendkey: "LEFTALT+RIGHT" # history back
      keypress:
        LEFTCTRL:
          sendkey: "PREVIOUSSONG"
        # LEFTMETA+LEFTALT:
          # sendkey: 'BRIGHTNESSDOWN'

    right:
      # sendkey: "LEFTALT+LEFT" # history forward
      keypress:
        LEFTCTRL:
          sendkey: "NEXTSONG"
        # LEFTMETA+LEFTALT:
        #   sendkey: 'xdotool key --clearmodifiers XF86AudioLowerVolume'

    up:
      sendkey: "LEFTMETA" # Overview
      keypress:
        LEFTCTRL:
          sendkey: "LEFTCTRL+K" # open new tab
        LEFTCTRL+LEFTALT:
          sendkey: 'BRIGHTNESSUP'

    down:
      sendkey: "LEFTMETA" # Overview
      keypress:
        LEFTCTRL:
          sendkey: "LEFTCTRL+COMMA" # close tab
        LEFTCTRL+LEFTALT:
          sendkey: 'BRIGHTNESSDOWN'

  4:
    left:
      sendkey: "VOLUMEDOWN"
    right:
      sendkey: "VOLUMEUP"
#     up:
#       sendkey: "xdotool key ctrl+alt+Down" # Switch to next workspace
#     down:
#       sendkey: "xdotool key ctrl+alt+Up" # Switch to previous workspace
pinch:
  in:
    sendkey: "ZOOMOUT" # Zoom in
  out:
    sendkey: "ZOOMIN" # Zoom out

threshold:
  pinch: 0.4
  swipe: 0.2

interval:
  swipe: 0.8
  pinch: 0.5
  tap: 0.2

tap:
  # 1:
  #   sendkey: "echo ----------------tap1----------------------------"
  # 2:
  #   sendkey: "echo ----------------tap2----------------------------"
  # 3:
  #   sendkey: "echo ----------------tap3----------------------------"
  # 4:
    # sendkey: "PLAYPAUSE"

hold:
#   1:
#     command: "echo ----------------hold1----------------------------"
#   2:
#     command: "echo ----------------hold2----------------------------"
#   3:
#     command: "echo ----------------hold3----------------------------"
  4:
    sendkey: "PLAYPAUSE"

plugin: 
  inputs:
    libinput_command_input:
      enable-tap: true
      verbose: true
      show-keycodes: true

```

Note that this issue happens on my dell laptop which completely suspends and not on my lenovo laptop which doesn't completely suspend (i can tell that since on my dell laptop fans stop working but on lenovo, they dont) (and i am not talking about hibernate here). I have also setup ubuntu without swap partition so ubuntu itself handles swap (currently 2GB), this maybe of help.

HritwikSinghal, 2020-10-16:
let me tell if i have to grab any logs

HritwikSinghal, 2020-10-16:
```
I, [2020-10-16T21:56:07.221207 #4571]  INFO -- : reload config: /home/hritwik/.config/fusuma/config.yml
I, [2020-10-16T21:56:07.272904 #4571]  INFO -- : ---------------------------------------------
I, [2020-10-16T21:56:07.272930 #4571]  INFO -- : Fusuma: 1.11.1
I, [2020-10-16T21:56:07.274318 #4571]  INFO -- : libinput: 1.15.5
I, [2020-10-16T21:56:07.274856 #4571]  INFO -- : OS: Linux 5.4.0-51-generic #56-Ubuntu SMP Mon Oct 5 14:28:49 UTC 2020
I, [2020-10-16T21:56:07.275354 #4571]  INFO -- : Distribution: Ubuntu 20.04.1 LTS \n \l
I, [2020-10-16T21:56:07.275860 #4571]  INFO -- : Desktop session: ubuntu-wayland wayland
I, [2020-10-16T21:56:07.275877 #4571]  INFO -- : ---------------------------------------------
I, [2020-10-16T21:56:07.275885 #4571]  INFO -- : Enabled Plugins: 
I, [2020-10-16T21:56:07.275939 #4571]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-10-16T21:56:07.275947 #4571]  INFO -- :   Fusuma::Plugin::Buffers::KeypressBuffer
I, [2020-10-16T21:56:07.275952 #4571]  INFO -- :   Fusuma::Plugin::Buffers::TapBuffer
I, [2020-10-16T21:56:07.275957 #4571]  INFO -- :   Fusuma::Plugin::Detectors::KeypressDetector
I, [2020-10-16T21:56:07.275968 #4571]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-10-16T21:56:07.275973 #4571]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-10-16T21:56:07.275978 #4571]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-10-16T21:56:07.275983 #4571]  INFO -- :   Fusuma::Plugin::Detectors::TapDetector
I, [2020-10-16T21:56:07.275988 #4571]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-10-16T21:56:07.275996 #4571]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-10-16T21:56:07.276001 #4571]  INFO -- :   Fusuma::Plugin::Events::Records::KeypressRecord
I, [2020-10-16T21:56:07.276006 #4571]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-10-16T21:56:07.276011 #4571]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-10-16T21:56:07.276016 #4571]  INFO -- :   Fusuma::Plugin::Executors::SendkeyExecutor
I, [2020-10-16T21:56:07.276020 #4571]  INFO -- :   Fusuma::Plugin::Filters::KeypressFilter
I, [2020-10-16T21:56:07.276025 #4571]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-10-16T21:56:07.276030 #4571]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-10-16T21:56:07.276035 #4571]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-10-16T21:56:07.276041 #4571]  INFO -- :   Fusuma::Plugin::Parsers::KeypressParser
I, [2020-10-16T21:56:07.276046 #4571]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-10-16T21:56:07.276054 #4571]  INFO -- :   Fusuma::Plugin::Parsers::TapParser
I, [2020-10-16T21:56:07.276060 #4571]  INFO -- : ---------------------------------------------
I, [2020-10-16T21:56:13.845481 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:14.506057 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:15.223116 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:15.702758 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:16.654756 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:17.322750 #4571]  INFO -- : {:sendkey=>"PLAYPAUSE"}
I, [2020-10-16T21:56:46.353354 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:46.801954 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:47.205204 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:48.353503 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
I, [2020-10-16T21:56:48.760617 #4571]  INFO -- : {:sendkey=>"LEFTMETA"}
```

and it happens after suspend and not hibernate. 

One more thing, after step 3 (mentioned in issue) if i kill all fusuma process and run only one, it still persists across further suspends. So by this i can conclude that it does not need 2 process to work after suspend. 

And if there are two process working on Xorg (after suspend or at any time), the gestures are recognized twice, but in wayland that only happens before suspend (or if i start 3 process after first suspend). So that means after suspend, it looks like one process is not working. (only after first suspend (i.e after step 3) )

iberianpig, 2020-10-17:
I have confirmed that the fusuma gesture works after the suspend.
No problem with either Wayland or X11.
It was tested on a Dell XPS13 9300.

HritwikSinghal, 2020-10-19:
there is another thing i noticed, 

1) i boot in ubuntu
2) i logout and log back in

now another fusuma process starts. So now there are two fusuma processes (and thus gestures are super sensitive), i have to stop one fusuma process to make gestures normal again.

iberianpig, 2020-10-19:
I checked on XPS 13 9360 (Ubuntu Wayland)
```shell
iberianpig@XPS-13-9300:~$ uname -a
Linux XPS-13-9300 5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
iberianpig@XPS-13-9300:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal
```

1. after boot(output of `ps aux | grep fusuma`)

```shell
iberianpig@XPS-13-9300:~$ cat /tmp/fusuma_ps
iberian+    3183  1.0  0.1 222060 27164 ?        Sl   22:09   0:01 /usr/bin/ruby2.7 /usr/local/bin/fusuma -d -c ~/.config/fusuma/config.gnome.yml
iberian+    6090  0.0  0.0   9956   728 pts/1    S+   22:11   0:00 grep --color=auto fusuma
iberian+    6091  0.0  0.0   9344   588 pts/1    S+   22:11   0:00 tee /tmp/fusuma_ps
```
2. after logout and log back in
```shell
iberianpig@XPS-13-9300:~$ ps aux | grep fusuma
iberian+    7291  1.1  0.1 222312 27464 ?        Sl   22:12   0:03 /usr/bin/ruby2.7 /usr/local/bin/fusuma -d -c ~/.config/fusuma/config.gnome.yml
iberian+   10517  0.0  0.0  10088  2696 pts/1    S+   22:17   0:00 grep --color=auto fusuma
```

=> There is only 1 process without grep in these sessions.

HritwikSinghal, 2021-01-02:
Hey just an update to this issue, this still happens on ubuntu 20.10 on my laptop. I did a clean install (actually did 2-3 clean installs) and it is still there. 

can you tell me how did you add fusuma to startup apps? maybe the bug is with that.

iberianpig, 2021-01-04:
I added fusuma to gnome-session-properties. 

```sh
iberianpig@iberianpig-XPS-13-9310:~$ cat ~/.config/autostart/fusuma.desktop
[Desktop Entry]
Name=fusuma(system ruby)
Comment=run fusuma
Exec=/usr/local/bin/fusuma -d
Icon=input-touchpad
X-GNOME-Autostart-enabled=true
Type=Application
Name[ja]=fusuma
```