iberianpig/fusuma #211: Executing any three finger gesture it still gets passed on to the application

zakhar-bykov, 2020-11-11:
**Describe the bug**
When executing any three finger gesture it still gets passed on to the application.
Example: When I scroll in Google Chrome on a webpage or any application with scrolling views with three fingers up / down (fusuma simulates pressing a super key and shows me all open apps in that workspace) or scrolls left / right (fusuma emulates switching tab / shift tab) the gesture does not start and fusuma doesn't show it in the console (not in --daemon mode). Sometimes it works, sometimes it doesn't.

**To Reproduce**
Steps to reproduce the behavior:
1. Open any site in Google Chrome
2. Scrolling with three fingers in any direction

**Expected behavior**
Three-fingers scrolling up / down - show all open apps in this workspace (press super key).
Three-fingers scrolling left / right - switching between tabs in browser (press tab / shift tab).

**Versions**
It hapends on versions 1.9.? and 1.11.1. Don't know about this issue on another versions.
`fusuma --version`
`I, [2020-11-11T13:27:15.268202 #32041]  INFO -- : reload config: /home/zakhar/.config/fusuma/config.yml
I, [2020-11-11T13:27:15.651645 #32041]  INFO -- : ---------------------------------------------
I, [2020-11-11T13:27:15.651716 #32041]  INFO -- : Fusuma: 1.11.1
I, [2020-11-11T13:27:15.985379 #32041]  INFO -- : libinput: 1.10.4
I, [2020-11-11T13:27:15.987133 #32041]  INFO -- : OS: Linux 5.4.0-52-generic #57~18.04.1-Ubuntu SMP Thu Oct 15 14:04:49 UTC 2020
I, [2020-11-11T13:27:16.088309 #32041]  INFO -- : Distribution: Ubuntu 18.04.5 LTS \n \l
I, [2020-11-11T13:27:16.089910 #32041]  INFO -- : Desktop session: ubuntu x11
I, [2020-11-11T13:27:16.090053 #32041]  INFO -- : ---------------------------------------------
I, [2020-11-11T13:27:16.090165 #32041]  INFO -- : Enabled Plugins: 
I, [2020-11-11T13:27:16.090464 #32041]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-11-11T13:27:16.090575 #32041]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-11-11T13:27:16.090669 #32041]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-11-11T13:27:16.090757 #32041]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-11-11T13:27:16.090842 #32041]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-11-11T13:27:16.090906 #32041]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-11-11T13:27:16.090966 #32041]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-11-11T13:27:16.091031 #32041]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-11-11T13:27:16.091092 #32041]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-11-11T13:27:16.091152 #32041]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-11-11T13:27:16.091212 #32041]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-11-11T13:27:16.091269 #32041]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-11-11T13:27:16.091345 #32041]  INFO -- : ---------------------------------------------
`

**Additional context**
My config.yml file:
`swipe:
#  2:
#    up:
#      command: 'xdotool key Up'
#    down:
#      command: 'xdotool key Down'
  3: 
    left: 
      command: 'xte "keydown Control_L" "key Tab" "keyup Control_L"'
      threshold: 0.8
      interval: 1
    right: 
      command: 'xte "keydown Control_L" "keydown Shift_L" "key Tab" "keyup Shift_L" "keyup Control_L"'
      threshold: 0.8
      interval: 1
    up: 
      command: 'xte "key Super_L"'
      threshold: 0.6
      interval: 1
    down: 
      command: 'xte "key Super_L"'
      threshold: 0.6
      interval: 1
  4:
    #left: 
      #command: 'xdotool key ctrl+alt+Down'
    #right: 
      #command: 'xdotool key ctrl+alt+Up'
    up: 
      command: 'xte "keydown Control_L" "keydown Super_L" "key Down" "keyup Super_L" "keyup Control_L"'
      threshold: 0.6
      interval: 0.8
    down:
      command: xte "keydown Control_L" "keydown Super_L" "key Up" "keyup Super_L" "keyup Control_L"
      threshold: 0.6
      interval: 0.8
pinch:
  in:
    command: 'xte "keydown Control_L" "key plus" "keyup Control_L"'
  out:
    command: 'xte "keydown Control_L" "key minus" "keyup Control_L"'

threshold:
  swipe: 0.4
  pinch: 0.4

interval:
  swipe: 0.2
  pinch: 2
`

iberianpig, 2020-11-24:
I checked with your config.yml and libinput v1.10.4, but I can't reproduce... 
What are you using the touchpad device? You can check it with `fusuma -l`

zakhar-bykov, 2020-11-25:
SynPS/2 Synaptics TouchPad

More info:
By running the command `fusuma -l`
```
I, [2020-11-26T01:35:59.154532 #31151]  INFO -- : reload config: /home/zakhar/.config/fusuma/config.yml
I, [2020-11-26T01:35:59.189719 #31151]  INFO -- : ---------------------------------------------
I, [2020-11-26T01:35:59.189774 #31151]  INFO -- : Fusuma: 1.11.1
I, [2020-11-26T01:35:59.192368 #31151]  INFO -- : libinput: 1.10.4
I, [2020-11-26T01:35:59.193368 #31151]  INFO -- : OS: Linux 5.4.0-54-generic #60~18.04.1-Ubuntu SMP Fri Nov 6 17:25:16 UTC 2020
I, [2020-11-26T01:35:59.194744 #31151]  INFO -- : Distribution: Ubuntu 18.04.5 LTS \n \l
I, [2020-11-26T01:35:59.195870 #31151]  INFO -- : Desktop session: ubuntu x11
I, [2020-11-26T01:35:59.195918 #31151]  INFO -- : ---------------------------------------------
I, [2020-11-26T01:35:59.195954 #31151]  INFO -- : Enabled Plugins: 
I, [2020-11-26T01:35:59.196091 #31151]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-11-26T01:35:59.196130 #31151]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-11-26T01:35:59.196167 #31151]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-11-26T01:35:59.196214 #31151]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-11-26T01:35:59.196244 #31151]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-11-26T01:35:59.196277 #31151]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-11-26T01:35:59.196310 #31151]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-11-26T01:35:59.196335 #31151]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-11-26T01:35:59.196359 #31151]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-11-26T01:35:59.196383 #31151]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-11-26T01:35:59.196406 #31151]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-11-26T01:35:59.196431 #31151]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-11-26T01:35:59.196456 #31151]  INFO -- : ---------------------------------------------
SynPS/2 Synaptics TouchPad
```

By running the command `xinput list`
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ HP Truevision HD: HP Truevision         	id=9	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=10	[slave  keyboard (3)]
    ↳ HP Wireless hotkeys                     	id=12	[slave  keyboard (3)]
    ↳ HP WMI hotkeys                          	id=13	[slave  keyboard (3)]
```

By running the command `cat /proc/bus/input/devices`
```
I: Bus=0011 Vendor=0002 Product=0007 Version=01b1
N: Name="SynPS/2 Synaptics TouchPad"
P: Phys=isa0060/serio1/input0
S: Sysfs=/devices/platform/i8042/serio1/input/input4
U: Uniq=
H: Handlers=mouse0 event3 
B: PROP=1
B: EV=b
B: KEY=e520 30000 0 0 0 0
B: ABS=660800011000003
```

iberianpig, 2020-11-26:
I would like to isolate the cause of this problem.

> Sometimes it works, sometimes it doesn't.

* When does it happen? Do you have any reproducibility hints? (e.g., does not occur after reboot, but after suspend, hibernate, etc.)
* The problem occurs only with three-fingered gestures?
* Does this problem occur not only on Google Chrome, but also on other applications?
* Does the xte command work properly when run on a console?

Also, launch a terminal and run `libinput debug-events`.
Check the log when a problem occurs. If `libinput debug-events` doesn't output any logs of three-finger swipes, Fusuma won't be able to detect them.
In that case, it's probably a bug in Libinput.

zakhar-bykov, 2020-12-05:
Some of your questions are already answered in my bug report.

This happens anytime the system is running.

When a gesture fails, it is not logged to the console. And when it works - log in. Conclusion - fusuma doesn't always track my gestures.

To activate my gestures, I use a four-finger gesture to switch to and from the next workspace. And again I activate my three-finger gesture and it works.

iberianpig, 2020-12-05:
I can't fix the issue that I can't reproduce.

Please try fusuma-plugin-sendkey instead of xte.

MaximoMachado, 2021-01-12:
I'm using xdotool and this occurs as well. For example, if I tap with 3 fingers, it'll mute my audio as I've configured it but in Firefox it'll close the tab I am hovering over.

It's possible that 4 finger gestures pass through but maybe there's no applications that use them as defaults.

zakhar-bykov, 2021-01-14:
I sometimes have a three-finger gesture that does not work (swipe right left to go through the tabs / up and down to show all applications on this workspace). And to make it work, I use four fingers to switch to another workspace and then switch back. And after this magic with four fingers, the three-finger gesture begins to work.

closed 2022-01-09 by stale[bot]