iberianpig/fusuma #212: Three finger gestures unreliable

marvinmai, 2020-11-22:
Very often, fusuma is not detecting three finger gestures. Some times it's working fine, but around half the time for a certain time it's not working. After some time it starts working again. I tried to observe if it's a certain action that reactivates it, but I wasn't able to figure sth. out. 
This is the content of my config.yml file.
````
swipe:
  3:
    left:
      command: 'xdotool key alt+Right'
    right:
      command: 'xdotool key alt+Left'
    up:
      command: 'xdotool key super'
    down:
      command: 'xdotool key super+a'
  4:
    left:
      command: 'xdotool key ctrl+super+Left'
    right:
      command: 'xdotool key ctrl+super+Right'
    up:
      command: 'xdotool key super+ctrl+Down'
    down:
      command: 'xdotool key super+ctrl+Up'
pinch:
  in:
    command: 'xdotool key ctrl+plus'
  out:
     command: 'xdotool key ctrl+minus'

threshold:
  swipe: 0.4
  pinch: 0.4

interval:
  swipe: 0.8
  pinch: 0.1
````

marvinmai, 2020-11-22:
The issue is always appearing for all 4 three finger gestures configured.

iberianpig, 2020-11-22:
Could you paste output of `fusuma --version`?

zakhar-bykov, 2020-11-23:
I have the same problem. I described this problem in #211

Haotian-Yang, 2020-11-26:
Got same issues

marvinmai, 2020-11-27:
this is the result:
```
fusuma --version
I, [2020-11-27T22:50:44.956278 #8399]  INFO -- : ---------------------------------------------
I, [2020-11-27T22:50:44.956310 #8399]  INFO -- : Fusuma: 1.10.1
I, [2020-11-27T22:50:44.957523 #8399]  INFO -- : libinput: 1.15.5
I, [2020-11-27T22:50:44.957958 #8399]  INFO -- : OS: Linux 5.8.0-7630-generic #32~1605108853~20.04~8bcf10e-Ubuntu SMP Wed Nov 11 22:42:16 UTC
I, [2020-11-27T22:50:44.958340 #8399]  INFO -- : Distribution: Pop!_OS 20.04 LTS \n \l
I, [2020-11-27T22:50:44.958750 #8399]  INFO -- : Desktop session: pop x11
I, [2020-11-27T22:50:44.958768 #8399]  INFO -- : ---------------------------------------------
I, [2020-11-27T22:50:44.958776 #8399]  INFO -- : Enabled Plugins: 
I, [2020-11-27T22:50:44.958810 #8399]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2020-11-27T22:50:44.958819 #8399]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2020-11-27T22:50:44.958826 #8399]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2020-11-27T22:50:44.958831 #8399]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2020-11-27T22:50:44.958837 #8399]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2020-11-27T22:50:44.958842 #8399]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2020-11-27T22:50:44.958848 #8399]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2020-11-27T22:50:44.958853 #8399]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2020-11-27T22:50:44.958859 #8399]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2020-11-27T22:50:44.958865 #8399]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2020-11-27T22:50:44.958871 #8399]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2020-11-27T22:50:44.958886 #8399]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2020-11-27T22:50:44.958893 #8399]  INFO -- : ---------------------------------------------
```
do you have any ideas? it's completely randomly appearing and disappearing and I have no clue how to fix it. If this works the workflow is really flawless and could keep up with gestures on mac os imo. But like that it's really tedious to use.

Something which is maybe also of interest: Always when I'm switching to German keyboard layout, the gestures are even working worse. E.g. the four finger gesture to switch to another desktop is delayed by around 2 seconds. Also, if they are working, the other gestures are delayed and working less fluent in general. This is 100% reproducible, also on 2 separate machines by now, with the same hardware setup and same OS. Maybe this belongs in a separate issue, but I want to mention it here first before creating a new one.

marvinmai, 2020-11-27:
I also just observed the behaviour of libinput debug-events. It just behaves like it's a two finger gesture when the three finger gesture is not detected. Does the mean the problem comes from libinput?

iberianpig, 2020-11-29:
> I also just observed the behaviour of libinput debug-events. It just behaves like it's a two finger gesture when the three finger gesture is not detected. Does the mean the problem comes from libinput?

Exactly. I think the cause is libinput behavior because fusuma reads outputs libinput debug-events and detects gestures.

> Something which is maybe also of interest: Always when I'm switching to German keyboard layout, the gestures are even working worse. E.g. the four finger gesture to switch to another desktop is delayed by around 2 seconds

This is a known issue of xdotool. xdotool command delay when using multiple keyboard layouts.
Please try https://github.com/iberianpig/fusuma-plugin-sendkey. It emulates key events in evdev, a lower layer X11. So it has no delays compared to xdotool.

nrbeaton, 2020-12-14:
I am having a similar problem. Using fusuma-plugin-sendkey. Four-finger swipe reliably works every time, but three-finger swipe is hit-and-miss. 

When three-finger swipe fails, libinput debug-events sometimes registers it as POINTER_AXIS (ie. behaves like two-finger scrolling) and sometimes as POINTER_MOTION.

Device is SynPS/2 Synaptics TouchPad.

vsantalov, 2020-12-15:
Same problem with both xdtool and fusuma-plugin-sendkey. The issue is most likely caused by libinput. Not surprisingly, the same problem also exits with libinput-gestures. I could never get reliable three-finger recognition with libinput. In the past llibinput would often fail to recognize three finger swipe and perform xcopy paste instead.
The only workaround I found was to avoid using three-finger gestures..

POP_OS 20.10
LENOVO P1 Gen2
TPPS/2 Elan TrackPoint

iberianpig, 2020-12-15:
> SynPS / 2 Synaptics TouchPad

It seems that you are using sypantics driver.
Could you setting to use libinput without synaptics?

vsantalov, 2020-12-15:
> > SynPS / 2 Synaptics TouchPad
> 
> It seems that you are using sypantics driver.
> Could you setting to use libinput without synaptics?

```
grep -i "Using input driver" /var/log/Xorg.0.log
[     6.284] (II) Using input driver 'libinput' for 'Power Button'
[     6.297] (II) Using input driver 'libinput' for 'Video Bus'
[     6.300] (II) Using input driver 'libinput' for 'Video Bus'
[     6.303] (II) Using input driver 'libinput' for 'Sleep Button'
[     6.305] (II) Using input driver 'libinput' for 'Integrated Camera: Integrated C'
[     6.307] (II) Using input driver 'libinput' for 'Integrated Camera: Integrated I'
[     6.310] (II) Using input driver 'libinput' for 'AT Translated Set 2 keyboard'
[     6.313] (II) Using input driver 'libinput' for 'SynPS/2 Synaptics TouchPad'
[     6.318] (II) Using input driver 'libinput' for 'TPPS/2 Elan TrackPoint'
[     6.327] (II) Using input driver 'libinput' for 'ThinkPad Extra Buttons'
```

nrbeaton, 2020-12-16:
I get similar output as @vsantalov -- in particular, using libinput for the touchpad.

`xserver-xorg-input-synaptics` is not installed.

iberianpig, 2021-04-24:
I can't reproduce this in my environment.
I think this problem is a bug in libinput and we need to wait for it to be fixed.
Since this is not a bug in Fusuma, I'm closing this issue.

closed 2021-04-24 by iberianpig

amosbird, 2022-03-19:
I'm experiencing exactly the same issue with Thinkpad X1 Extreme Gen 3. I guess it's related to both specific hardware and libinput. My old Thinkpad T480 works just fine with the same configuration.