iberianpig/fusuma #213: Uncontrolled switching

titaniumjones, 2020-12-06:
I've had Fusuma installed and running nicely for months. 
Unfortunately, I installed Tweaks, because I wanted to try other desktop themes. I'm pretty sure that was the moment when I started getting strange behaviour from Fusuma. Now when I use a 3 finger swipe up to switch desktops It literally flashes past all the desk tops to the top. and when I go down it flashes past all desktops to the bottom. If I use the 4 finger swipe ( which is assigned to "Activity" ) it shows for a second and then goes back to the desktop.
I've watched the output/input to Fusuma and It seems to only be getting one activation, but still flashes past many desktops.
I've now uninstalled Tweaks but I still get the same result. Any ideas/help/guidance would be greatly appreciated.

Oh, I also upgraded to the latest version.

titaniumjones, 2020-12-06:
Oh I also tried increasing the threshold and interval to 2,0 but it doesn't help.

titaniumjones, 2020-12-06:
I further found out is when there is 2 desktops everything behaves normally, once a 3rd desktop appears then the erratic behaviour begins.

iberianpig, 2020-12-07:
Fusuma only parse log of `libinput debug-events` and convert to command.

There is no relationship between the installation of Tweaks and fusuma.
I think you may install different gesture extensions.

titaniumjones, 2020-12-07:
That is entirely possible. Do you have any thoughts as to what? Could it have been part of Tweaks?

iberianpig, 2020-12-07:
Did you check extensions you installed from GNOME Shell Extensions?

titaniumjones, 2020-12-09:
Hello
Well I seem to have fixed the problem.
Unfortunately I'm not sure what I did to fix it.
Before finding Fusuma I did use xdotools and gestures GUI. 
I then moved to Fusuma and the Sendkey plugin.
I guess something to do with tweaks reenabled XDOTools some how or something....
Anyway I removed Xdotools (and also libinput-tools) 
Reinstalled the Fusuma sendkey plugin (and also libinput-tools)
and everything now seems to be working again.

iberianpig, 2020-12-10:
I'm glad to know you solved the problem.

Do you have multiple keyboard layouts?
xdotool has the known issue that input delay occurs with multiple keyboard layouts.

titaniumjones, 2020-12-11:
And so it continues.
I did think I had fixed it and indeed for a few hours I didn't experience any strange behaviour. But then it came back again.
I tried to do some sleuthing of my own and I don't know if what I'm seeing is normal because I didn't see the output when I wasn't experiencing strange behaviour.

$ fusuma -v

The output continued to scroll up the terminal screen, when I pressed nothing I got a continual timeout as follows:

D, [2020-12-11T16:41:23.084113 #80803] DEBUG -- : {:input_event=>time: 2020-12-11 16:41:23 +0100, tag: libinput_command_input, record: LIBINPUT TIMEOUT}

When I do a 3 finger swipe I get the following:

D, [2020-12-11T16:46:28.875366 #80925] DEBUG -- : {:input_event=>time: 2020-12-11 16:46:28 +0100, tag: libinput_command_input, record: -event4   GESTURE_SWIPE_UPDATE +1.699s	3 -1.78/22.34 (-4.75/59.63 unaccelerated)}
D, [2020-12-11T16:46:28.883570 #80925] DEBUG -- : {:input_event=>time: 2020-12-11 16:46:28 +0100, tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE +1.707s	3 -2.48/29.13 (-6.56/77.03 unaccelerated)}
D, [2020-12-11T16:46:28.891943 #80925] DEBUG -- : {:input_event=>time: 2020-12-11 16:46:28 +0100, tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE +1.715s	3 -3.42/34.80 (-7.82/79.45 unaccelerated)}
D, [2020-12-11T16:46:28.899598 #80925] DEBUG -- : {:input_event=>time: 2020-12-11 16:46:28 +0100, tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE +1.723s	3 -3.74/35.38 (-8.38/79.31 unaccelerated)}
D, [2020-12-11T16:46:28.907644 #80925] DEBUG -- : {:input_event=>time: 2020-12-11 16:46:28 +0100, tag: libinput_command_input, record:  event4   GESTURE_SWIPE_UPDATE +1.731s	3 -3.90/34.87 (-8.80/78.60 unaccelerated)}

I'm not sure if what I'm seeing is correct or something wrong.

Thanks

titaniumjones, 2020-12-11:
Ok
More debugging, I ran 

$ ps -C fusuma

and fond I had (about) 10 instances of fusuma running. I suspect this occurs because when the swipe actions stop working I manually start another instance of fusuma -d. Clearly the previous instance hadn't stopped. I must do more investigation.

Cheers

closed 2022-01-09 by stale[bot]