iberianpig/fusuma #215: Doesn't run commands

sm177y23, 2021-01-01:
Doesn't execute anything. It sees my touchscreen and is configured to use it but no swipes, taps, holds or anything of any finger count actually work. 
Libinput is definitely working and spitting out events as usual. In fact, I haven't been able to get any gesture support on my tablet other than Firefox that does it's own. Touchegg and others like it don't work either. 
To this end, I have started reverse engineering the event codes from my touchscreen and handling them myself in Golang as it already has event handling via golang-evdev. So far I have single tap and a simple swipe able to launch commands without issue. 
I will post my work when I'm done as I'm guessing I'm not the only one who hasn't had any luck with these sort of applications.
It's quite a simple concept, I don't see what's so difficult about reading the input events and running system commands off them. This should be part of the linux desktop already and it's sad that it's not because nobody can agree on who should write the code. SMH

solar0037, 2021-01-29:
I use Manjaro LInux with KDE Plasma. I installed ruby from [asdf](https://github.com/asdf-vm/asdf), and installed fusuma by `gem install fusuma`.
It's not working on my system, either. I installed libinput, libinput-gestures, and ran `libinput-gestures-setup restart` according to the update notes. I was able to run fusuma and get logs printed out, but nothing happened. Normal gestures without fusuma runs fine.
Some of the logs:
```
W, [2021-01-29T11:38:09.835919 #4621]  WARN -- : config file: /home/solar0037/.config/fusuma/config.yml is NOT FOUND
I, [2021-01-29T11:38:09.835970 #4621]  INFO -- : reload config: /home/solar0037/.asdf/installs/ruby/3.0.0/lib/ruby/gems/3.0.0/gems/fusuma-1.11.1/lib/fusuma/config.yml
I, [2021-01-29T11:38:09.880549 #4621]  INFO -- : ---------------------------------------------
I, [2021-01-29T11:38:09.880586 #4621]  INFO -- : Fusuma: 1.11.1
I, [2021-01-29T11:38:09.881126 #4621]  INFO -- : libinput: 1.16.4
I, [2021-01-29T11:38:09.881585 #4621]  INFO -- : OS: Linux 5.9.16-1-MANJARO #1 SMP PREEMPT Mon Dec 21 22:00:46 UTC 2020
I, [2021-01-29T11:38:09.882031 #4621]  INFO -- : Distribution: Manjaro Linux \r  (\n) (\l)
I, [2021-01-29T11:38:09.884389 #4621]  INFO -- : Desktop session: /usr/share/xsessions/plasma x11
I, [2021-01-29T11:38:09.884413 #4621]  INFO -- : ---------------------------------------------
I, [2021-01-29T11:38:09.884427 #4621]  INFO -- : Enabled Plugins: 
I, [2021-01-29T11:38:09.884481 #4621]  INFO -- :   Fusuma::Plugin::Buffers::GestureBuffer
I, [2021-01-29T11:38:09.884494 #4621]  INFO -- :   Fusuma::Plugin::Detectors::PinchDetector
I, [2021-01-29T11:38:09.884503 #4621]  INFO -- :   Fusuma::Plugin::Detectors::RotateDetector
I, [2021-01-29T11:38:09.884511 #4621]  INFO -- :   Fusuma::Plugin::Detectors::SwipeDetector
I, [2021-01-29T11:38:09.884520 #4621]  INFO -- :   Fusuma::Plugin::Events::Records::GestureRecord
I, [2021-01-29T11:38:09.884529 #4621]  INFO -- :   Fusuma::Plugin::Events::Records::IndexRecord
I, [2021-01-29T11:38:09.884538 #4621]  INFO -- :   Fusuma::Plugin::Events::Records::TextRecord
I, [2021-01-29T11:38:09.884546 #4621]  INFO -- :   Fusuma::Plugin::Executors::CommandExecutor
I, [2021-01-29T11:38:09.884558 #4621]  INFO -- :   Fusuma::Plugin::Filters::LibinputDeviceFilter
I, [2021-01-29T11:38:09.884566 #4621]  INFO -- :   Fusuma::Plugin::Filters::LibinputTimeoutFilter
I, [2021-01-29T11:38:09.884575 #4621]  INFO -- :   Fusuma::Plugin::Inputs::LibinputCommandInput
I, [2021-01-29T11:38:09.884583 #4621]  INFO -- :   Fusuma::Plugin::Parsers::LibinputGestureParser
I, [2021-01-29T11:38:09.884593 #4621]  INFO -- : ---------------------------------------------
I, [2021-01-29T11:38:11.723886 #4621]  INFO -- : {:command=>"xdotool key super+s"}
I, [2021-01-29T11:38:12.498555 #4621]  INFO -- : {:command=>"xdotool key super+a"}
I, [2021-01-29T11:38:12.984767 #4621]  INFO -- : {:command=>"xdotool key alt+Left"}
I, [2021-01-29T11:38:13.521454 #4621]  INFO -- : {:command=>"xdotool key alt+Right"}
```

iberianpig, 2021-01-29:
@solar0037 could you run xdotool command on terminal?

```
xdotool key super+a
```

solar0037, 2021-01-30:
> @solar0037 could you run xdotool command on terminal?
> 
> ```
> xdotool key super+a
> ```

I use Linux Mint with Cinamon now. Sorry that I can't test in KDE again.
It was because I didn't install xdotool. `alt+Left`, `alt+Right`, `ctrl+w`, `ctrl+t`, and some commands work fine, but `super+a`, `super+s` doesn't work, which doesn't work manually as well (due to my desktop environment?). Fusuma didn't output any errors, so is  it intended to be used for other features? Or why is `xdotool` marked optional?

iberianpig, 2021-01-30:
Fusuma binds gestures to commands with `command:` properties in config.yml. 
You can set any command. xdotool is an option of them.

But I think errors by execution command should be display as stderr.
I will fix it😊

commit 929891e1 (2021-04-24): fix(command_executor): Output error in Execute

```
I, [2021-04-25T01:16:18.297902 #500330]  INFO -- : {:command=>"invalidcommand", :args=>{:move_x=>9.71, :move_y=>-16.18, :unaccelerated_x=>31.99, :unaccelerated_y=>-53.31, :zoom=>0.0, :rotate=>0.0}}
E, [2021-04-25T01:16:18.298733 #500330] ERROR -- : {:"[swipe, 4, up]"=>"No such file or directory - invalidcommand"}
```

Closes #215
closed 2021-04-24 by iberianpig