iberianpig/fusuma #218: Opening application with fusuma

Pinche-Dev, 2021-01-17:
Can't fusuma be configured such that it opens application like calculator 
using it on Hold can become handy

iberianpig, 2021-02-01:
Hi @Pinche-Dev 

> Can't fusuma be configured such that it opens application like calculator
> using it on Hold can become handy

Which one did you mean?

1. binding `hold gesture` to open calculator
1. Configure settings for specific applications (wm_class/window name)

iberianpig, 2021-02-01:
1 → Try hold gesture in https://github.com/iberianpig/fusuma-plugin-tap 

2 → Please wait to release https://github.com/iberianpig/fusuma-plugin-application_matcher (work in progress)

iberianpig, 2021-04-24:
I released  [fusuma-plugin-appmatcher](https://github.com/iberianpig/fusuma-plugin-appmatcher) depend on fusuma v2(need to update gem). Please try it.

closed 2021-04-24 by iberianpig