iberianpig/fusuma #222: Touchpad - Fusuma problem when changing the language

coctailpreperation, 2021-02-14:
Ubuntu 20.04

Fusuma is not working when i change the language to Arabic, it's working perfectly fine in English language.

it suddenly lags out the whole PC "if used a shortcut multiple times".

installation

sudo gpasswd -a $USER input
newgrp input
sudo apt-get install libinput-tools
sudo apt-get install ruby
sudo gem install fusuma
fusuma
and i added it as a startup command.

configuration file contents "~/.config/fusuma/config.yml" 

swipe:
  3:
    left:
      command: "xdotool key alt+Right" # History forward 
    right:
      command: "xdotool key alt+Left" # History back
    up:
      command: "xdotool key super" # Activity
    down:
      command: "xdotool key super+h" # Minimize
  4:
    right:
      command: "xdotool key alt+Escape"
    left:
      command: "xdotool key alt+Escape"  	
    up:
      command: "xdotool key ctrl+Tab" # Switch to next workspace
    down:
      command: "xdotool key ctrl+w" # Close Tab
      
pinch:
  in:
    command: 'xdotool key ctrl+plus'
  out:
    command: 'xdotool key ctrl+minus'
    
    
threshold:
  pinch: 0.1
  
interval:
  swipe: 0.75
  pinch: 1

iberianpig, 2021-02-14:
https://github.com/iberianpig/fusuma#alternatives-to-xdotool

Xdotool has some delay on multi keyboard layouts.
Please use fusuma-plugin-sendkey instead of xdotool.

coctailpreperation, 2021-02-15:
> https://github.com/iberianpig/fusuma#alternatives-to-xdotool
> 
> Xdotool has some delay on multi keyboard layouts.
> Please use fusuma-plugin-sendkey instead of xdotool.

It worked for me thanks a lot for the efforts

closed 2021-02-15 by coctailpreperation